OpenIPC/firmware #276: Error when changing settings in Majestic.

khseal, 2022-07-23:
Platform hi3516cv300
Sensor  ar0237_i2c
Here is the error log
Jul 23 14:33:52 openipc-hi3516cv300 kern.err kernel: [90823.849432] mmz_userdev:mmz_userdev_release: MMB LEAK(pid=837): 0x83380000, 77824 bytes, 'isp_stat_user'
Jul 23 14:33:52 openipc-hi3516cv300 kern.warn kernel: [90823.849475] mmz_userdev:mmz_userdev_release: mmz_userdev_release: mmb<0x83380000> mapped to userspace 0xb696d000 will be force unmaped!

widgetii, 2022-07-24:
It's warnings from vendor SDK. Nothing valuable is here.

closed 2022-07-24 by widgetii

khseal, 2022-07-25:
After the error, the video disappears.
Example
Stream video codec changes.
Any change that reloads the sensor configuration.
Sometimes changes are applied without error.