OpenIPC/firmware #284: Motion Alarm no work.

khseal, 2022-07-27:
Latest version firmware.No alarm in log. Visualize motion detection Ok.
Old version firmware work ok.
My config
```
root@openipc-hi3516cv300:~# cat /etc/majestic.yaml 
system:
  buffer: 1024
image:
  mirror: false
  flip: false
  rotate: none
  contrast: 50
  hue: 50
  saturation: 50
  luminance: 50
osd:
  enabled: false
  template: "%a %e %B %Y %H:%M:%S %Z"
nightMode:
  enabled: true
  irSensorPin: 64
  irSensorPinInvert: false
  irCutPin1: 54
  irCutPin2: 53
records:
  enabled: false
  path: /mnt/mmc/%Y/%m/%d/%H.mp4
  maxUsage: 95
video0:
  enabled: true
  codec: h264
video1:
  enabled: false
jpeg:
  enabled: true
mjpeg:
  size: 640x360
  fps: 5
  bitrate: 1024
audio:
  enabled: false
  volume: auto
  srate: 8000
rtsp:
  enabled: true
  port: 554
hls:
  enabled: true
youtube:
  enabled: false
motionDetect:
  enabled: true
  visualize: true
  debug: true
ipeye:
  enabled: false
watchdog:
  enabled: true
  timeout: 10
isp:
  blkCnt: 6
```

khseal, 2022-07-27:
Motion guard set sensitvity 5. 
Motion Alarm work.

khseal, 2022-07-27:
Works but not for long.
```
Jul 27 20:39:22 openipc-hi3516cv300 user.debug majestic[842]: Motion detection started on 3 ext channel
Jul 27 20:39:22 openipc-hi3516cv300 user.debug majestic[842]: 656,900	48,24
Jul 27 20:39:22 openipc-hi3516cv300 user.debug majestic[842]: Motion detected in 1 regions, 656x900 -> [704x924]
Jul 27 20:39:23 openipc-hi3516cv300 user.debug majestic[842]: take jpeg venc_chn(2)   1920x1080  q:70
Jul 27 20:39:23 openipc-hi3516cv300 user.debug majestic[842]: JPEG snapshot was received from hardware encoder
Jul 27 20:39:23 openipc-hi3516cv300 user.debug majestic[842]: JPEG transcoding was started
Jul 27 20:39:24 openipc-hi3516cv300 user.debug majestic[842]: 664,908	40,32
Jul 27 20:39:24 openipc-hi3516cv300 user.debug majestic[842]: Motion detected in 1 regions, 664x908 -> [704x940]
Jul 27 20:39:24 openipc-hi3516cv300 user.debug majestic[842]: 664,908	40,32
Jul 27 20:39:24 openipc-hi3516cv300 user.debug majestic[842]: Motion detected in 1 regions, 664x908 -> [704x940]
Jul 27 20:39:25 openipc-hi3516cv300 user.debug majestic[842]: 664,908	32,32
Jul 27 20:39:25 openipc-hi3516cv300 user.debug majestic[842]: Motion detected in 1 regions, 664x908 -> [696x940]
Jul 27 20:39:25 openipc-hi3516cv300 kern.err kernel: [   21.437263] mmz_userdev:mmz_userdev_release: MMB LEAK(pid=842): 0x8342E000, 77824 bytes, 'isp_stat_user'
Jul 27 20:39:25 openipc-hi3516cv300 kern.warn kernel: [   21.437312] mmz_userdev:mmz_userdev_release: mmz_userdev_release: mmb<0x8342E000> mapped to userspace 0xb68b7000 will be force unmaped!
Jul 27 20:39:25 openipc-hi3516cv300 kern.err kernel: [   21.437338] mmz_userdev:mmz_userdev_release: MMB LEAK(pid=842): 0x841AD000, 507904 bytes, ''
Jul 27 20:39:25 openipc-hi3516cv300 kern.warn kernel: [   21.437353] mmz_userdev:mmz_userdev_release: mmz_userdev_release: mmb<0x841AD000> mapped to userspace 0xb631b000 will be force unmaped!
Jul 27 20:39:25 openipc-hi3516cv300 kern.err kernel: [   21.437369] mmz_userdev:mmz_userdev_release: MMB LEAK(pid=842): 0x84229000, 507904 bytes, ''
Jul 27 20:39:25 openipc-hi3516cv300 kern.warn kernel: [   21.437382] mmz_userdev:mmz_userdev_release: mmz_userdev_release: mmb<0x84229000> mapped to userspace 0xb629f000 will be force unmaped!
Jul 27 20:39:25 openipc-hi3516cv300 kern.err kernel: [   21.437396] mmz_userdev:mmz_userdev_release: MMB LEAK(pid=842): 0x842A5000, 4096 bytes, ''
Jul 27 20:39:25 openipc-hi3516cv300 kern.warn kernel: [   21.437409] mmz_userdev:mmz_userdev_release: mmz_userdev_release: mmb<0x842A5000> mapped to userspace 0xb629e000 will be force unmaped!
Jul 27 20:39:25 openipc-hi3516cv300 kern.err kernel: [   21.437423] mmz_userdev:mmz_userdev_release: MMB LEAK(pid=842): 0x842A7000, 1048576 bytes, 'MD_ASSIST'
Jul 27 20:39:25 openipc-hi3516cv300 kern.warn kernel: [   21.437436] mmz_userdev:mmz_userdev_release: mmz_userdev_release: mmb<0x842A7000> mapped to userspace 0xb619d000 will be force unmaped!
```

themactep, 2022-07-27:
Это не имеет отношения ни к веб интерфейсу ни к motion detection. Нет сигнала в логах - аларм сидит и не отсвечивает. А по поводу поему его там нет -- открой тикет в репо маджестика.

closed 2022-07-27 by themactep