OpenIPC/firmware #293: Device ID 00014911 network card is not detected by the bootloader

khseal, 2022-07-30:
Log
[image]
Board photo
[image]

dimerr, 2022-07-30:
Please provide output of
```ipctool reginfo```
```ipctool reginfo --script```
```ipctool gpio scan```

from stock firmware and from openipc

khseal, 2022-07-30:
Near stock. The network is working.

```
# /utils/ipctool
---
board:
  vendor: Xiongmai
  model: HI3516D_83H40
  cloudId: 444272731d04222f
chip:
  vendor: HiSilicon
  model: 3516DV100
ethernet:
  mac: "00:00:46:34:45:66"
  phy-mode: rmii
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x80000
        sha1: eef9a868
        contains:
          - name: uboot-env
            offset: 0x40000
      - name: romfs
        size: 0x500000
        path: /,cramfs
        sha1: 4148beb1
      - name: user
        size: 0x640000
        path: /mnt/custom/data/Fonts,squashfs
        sha1: 36c3ac5a
      - name: web
        size: 0x180000
        path: /mnt/web,squashfs
        sha1: d938e3c4
      - name: custom
        size: 0x140000
        path: /mnt/custom,cramfs
        sha1: 8289599d
      - name: logo
        size: 0x40000
        path: /mnt/logo,cramfs
        sha1: 87a20d9c
      - name: mtd
        size: 0x140000
        path: /mnt/mtd,jffs2,rw
    size: 16M
    addr-mode: 3-byte
ram:
  total: 128M
  media: 78M
firmware:
  u-boot: "2010.06-svn (Oct 30 2014 - 17:22:51)"
  kernel: "3.4.35 (Thu Apr 16 13:57:42 CST 2015)"
  toolchain: gcc version 4.8.3 20131202 (prerelease) (Hisilicon_v300) 
  libc: uClibc 0.9.33.2
  sdk: "Hi3516A_MPP_V1.0.3.0 B040 Debug (Mar 13 2015, 19:09:49)"
sensors:
- vendor: ON Semiconductor
  model: AR0237
  control:
    bus: 0
    type: i2c
    addr: 0x20
  vicap-state: down
# ipctool reginfo 
-sh: ipctool: not found
# /utils/ipctool reginfo
[... 241 lines cut ...]
devmem 0x200f01d4 32 0
devmem 0x200f01d8 32 0
devmem 0x200f01dc 32 0
devmem 0x200f01e0 32 0
devmem 0x200f01e4 32 0
devmem 0x200f01e8 32 0
# /utils/ipctool gpio scan
Gr: 0, Addr:0x20140078, Data:0x02 = 0bxxx0001x, Addr:0x20140400, Dir:0x02 = 0bxxx0001x
Gr: 1, Addr:0x20150000, Data:0x00 = 0bxxxxxxxx, Addr:0x20150400, Dir:0x00 = 0bxxxxxxxx
Gr: 2, Addr:0x20160010, Data:0x00 = 0bxxxxx0xx, Addr:0x20160400, Dir:0x04 = 0bxxxxx1xx
Gr: 3, Addr:0x20170000, Data:0x00 = 0bxxxxxxxx, Addr:0x20170400, Dir:0x00 = 0bxxxxxxxx
Gr: 4, Addr:0x20180000, Data:0x00 = 0bxxxxxxxx, Addr:0x20180400, Dir:0x00 = 0bxxxxxxxx
Gr: 5, Addr:0x20190000, Data:0x00 = 0bxxxxxxxx, Addr:0x20190400, Dir:0x00 = 0bxxxxxxxx
Gr: 6, Addr:0x201A0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201A0400, Dir:0x00 = 0bxxxxxxxx
Gr: 7, Addr:0x201B0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201B0400, Dir:0x00 = 0bxxxxxxxx
Gr: 8, Addr:0x201C03F0, Data:0x08 = 0b000010xx, Addr:0x201C0400, Dir:0x08 = 0b000010xx
Gr: 9, Addr:0x201D03FC, Data:0xDF = 0b11011111, Addr:0x201D0400, Dir:0x03 = 0b00000011
Gr:10, Addr:0x201E00FC, Data:0x3C = 0bxx111100, Addr:0x201E0400, Dir:0x00 = 0bxx000000
Gr:11, Addr:0x201F0010, Data:0x00 = 0bxxxxx0xx, Addr:0x201F0400, Dir:0x00 = 0bxxxxx0xx
Gr:12, Addr:0x202003FC, Data:0x00 = 0b00000000, Addr:0x20200400, Dir:0x00 = 0b00000000
Gr:13, Addr:0x202103FC, Data:0x00 = 0b00000000, Addr:0x20210400, Dir:0x00 = 0b00000000
Gr:14, Addr:0x20220000, Data:0x00 = 0bxxxxxxxx, Addr:0x20220400, Dir:0x00 = 0bxxxxxxxx
======================================================================================

```

carbofos, 2022-07-31:
Скидываю данные по одной из бесперспективных камер на DV100 с каким то левым сенсором, который врядли будет поддерживаться. Но данные для сетевой платы могут пригодиться.

```
# 
# /mnt/logo/ipctool 
---
board:
  vendor: Xiongmai
  model: HI3516D_83H30
  cloudId: dbe3b3e226b009ac
chip:
  vendor: HiSilicon
  model: 3516DV100
ethernet:
  mac: "00:12:16:d6:1a:fc"
  phy-mode: rmii
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x80000
        sha1: 602d1f7f
        contains:
          - name: xmcrypto
            offset: 0x3fc00
          - name: uboot-env
            offset: 0x40000
      - name: romfs
        size: 0x500000
        path: /,cramfs
        sha1: 32352cad
      - name: user
        size: 0x640000
        path: /mnt/custom/data/Fonts,squashfs
        sha1: 7e7a7172
      - name: web
        size: 0x180000
        path: /mnt/web,squashfs
        sha1: e20acc0c
      - name: custom
        size: 0x140000
        path: /mnt/custom,cramfs
        sha1: b492dd29
      - name: logo
        size: 0x40000
        path: /mnt/logo,cramfs
        sha1: 08c6b473
      - name: mtd
        size: 0x140000
        path: /mnt/mtd,jffs2,rw
    size: 16M
    addr-mode: 3-byte
ram:
  total: 128M
  media: 78M
firmware:
  u-boot: "2010.06-svn (Oct 30 2014 - 17:22:51)"
  kernel: "3.4.35 (Thu Feb 18 11:05:33 CST 2016)"
  toolchain: gcc version 4.8.3 20131202 (prerelease) (Hisilicon_v300) 
  libc: uClibc 0.9.33.2
  sdk: "Hi3516A_MPP_V1.0.5.0 B030 Release (Oct 29 2015, 20:50:54)"
  main-app: dogtest

# /mnt/logo/ipctool reginfo
muxctrl_reg0 0x200f0000 0x1 GPIO0_5 [SENSOR_CLK]
muxctrl_reg1 0x200f0004 0x1 GPIO0_6 [FLASH_TRIG] SPI1_CSN1
muxctrl_reg2 0x200f0008 0x1 GPIO0_7 [SHUTTER_TRIG] SPI1_CSN2
muxctrl_reg3 0x200f000c 0x1 GPIO1_0 [VO_CLK]
muxctrl_reg4 0x200f0010 0x1 GPIO6_0 [VO_DAT15]
[... 238 lines cut ...]
devmem 0x200f01d8 32 0
devmem 0x200f01dc 32 0
devmem 0x200f01e0 32 0
devmem 0x200f01e4 32 0
devmem 0x200f01e8 32 0

# /mnt/logo/ipctool gpio scan
Gr: 0, Addr:0x20140078, Data:0x0A = 0bxxx0101x, Addr:0x20140400, Dir:0x02 = 0bxxx0001x
Gr: 1, Addr:0x20150000, Data:0x00 = 0bxxxxxxxx, Addr:0x20150400, Dir:0x00 = 0bxxxxxxxx
Gr: 2, Addr:0x20160018, Data:0x00 = 0bxxxxx00x, Addr:0x20160400, Dir:0x06 = 0bxxxxx11x
Gr: 3, Addr:0x20170000, Data:0x00 = 0bxxxxxxxx, Addr:0x20170400, Dir:0x00 = 0bxxxxxxxx
Gr: 4, Addr:0x20180000, Data:0x00 = 0bxxxxxxxx, Addr:0x20180400, Dir:0x00 = 0bxxxxxxxx
Gr: 5, Addr:0x20190000, Data:0x00 = 0bxxxxxxxx, Addr:0x20190400, Dir:0x00 = 0bxxxxxxxx
Gr: 6, Addr:0x201A0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201A0400, Dir:0x00 = 0bxxxxxxxx
Gr: 7, Addr:0x201B0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201B0400, Dir:0x00 = 0bxxxxxxxx
Gr: 8, Addr:0x201C03F0, Data:0x08 = 0b000010xx, Addr:0x201C0400, Dir:0x08 = 0b000010xx
Gr: 9, Addr:0x201D035C, Data:0x93 = 0b10x1x011, Addr:0x201D0400, Dir:0x47 = 0b01x0x111
Gr:10, Addr:0x201E00FC, Data:0x3F = 0bxx111111, Addr:0x201E0400, Dir:0x00 = 0bxx000000
Gr:11, Addr:0x201F0010, Data:0x00 = 0bxxxxx0xx, Addr:0x201F0400, Dir:0x00 = 0bxxxxx0xx
Gr:12, Addr:0x202003FC, Data:0x00 = 0b00000000, Addr:0x20200400, Dir:0x00 = 0b00000000
Gr:13, Addr:0x202103FC, Data:0x00 = 0b00000000, Addr:0x20210400, Dir:0x00 = 0b00000000
Gr:14, Addr:0x20220000, Data:0x00 = 0bxxxxxxxx, Addr:0x20220400, Dir:0x00 = 0bxxxxxxxx

 
```

carbofos, 2022-07-31:
На подобной камере HI3516D_83H40 у меня возникла проблема с сетью при установке OpenIPC

```
# /mnt/logo/ipctool 
---
board:
  vendor: Xiongmai
  model: HI3516D_83H40
  cloudId: xxxxxxxxxxxxxxxx
chip:
  vendor: HiSilicon
  model: 3516DV100
ethernet:
  mac: "00:12:17:8f:98:07"
  phy-mode: rmii
rom:
  - type: nor
    block: 64K
    chip:
      name: "MX25L128XX"
      id: 0xc22018
    partitions:
      - name: boot
        size: 0x80000
        sha1: 517cbc2d
        contains:
          - name: uboot-env
            offset: 0x40000
      - name: romfs
        size: 0x500000
        path: /,cramfs
        sha1: 5d93f3c9
      - name: user
        size: 0x640000
        path: /mnt/custom/data/Fonts,squashfs
        sha1: c8a358ca
      - name: web
        size: 0x180000
        path: /mnt/web,squashfs
        sha1: 61f76abb
      - name: custom
        size: 0x140000
        path: /mnt/custom,cramfs
        sha1: b7601686
      - name: logo
        size: 0x40000
        path: /mnt/logo,cramfs
        sha1: 6d3dd0f9
      - name: mtd
        size: 0x140000
        path: /mnt/mtd,jffs2,rw
    size: 16M
    addr-mode: 3-byte
ram:
  total: 128M
  media: 71M
firmware:
  u-boot: "2010.06-svn969 (Mar 29 2018 - 11:31:22)"
  kernel: "3.4.35 (Thu Mar 8 23:05:26 CST 2018)"
  toolchain: gcc version 4.8.3 20131202 (prerelease) (Hisilicon_v300) 
  libc: uClibc 0.9.33.2
  sdk: "Hi3516A_MPP_V1.0.7.0 B050 Release (Nov 25 2016, 21:36:12)"
  main-app: /usr/bin/Sofia
Error: unexpected value for Aptina == 0x2000
sensors:
- vendor: OmniVision
  model: OV4689
  control:
    bus: 0
    type: i2c
[... 249 lines cut ...]
devmem 0x200f01d4 32 0
devmem 0x200f01d8 32 0
devmem 0x200f01dc 32 0
devmem 0x200f01e0 32 0
devmem 0x200f01e4 32 0
devmem 0x200f01e8 32 0

# /mnt/logo/ipctool gpio scan
Gr: 0, Addr:0x20140078, Data:0x0A = 0bxxx0101x, Addr:0x20140400, Dir:0x02 = 0bxxx0001x
Gr: 1, Addr:0x20150000, Data:0x00 = 0bxxxxxxxx, Addr:0x20150400, Dir:0x00 = 0bxxxxxxxx
Gr: 2, Addr:0x20160018, Data:0x00 = 0bxxxxx00x, Addr:0x20160400, Dir:0x06 = 0bxxxxx11x
Gr: 3, Addr:0x20170000, Data:0x00 = 0bxxxxxxxx, Addr:0x20170400, Dir:0x00 = 0bxxxxxxxx
Gr: 4, Addr:0x20180000, Data:0x00 = 0bxxxxxxxx, Addr:0x20180400, Dir:0x00 = 0bxxxxxxxx
Gr: 5, Addr:0x20190000, Data:0x00 = 0bxxxxxxxx, Addr:0x20190400, Dir:0x00 = 0bxxxxxxxx
Gr: 6, Addr:0x201A0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201A0400, Dir:0x00 = 0bxxxxxxxx
Gr: 7, Addr:0x201B0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201B0400, Dir:0x00 = 0bxxxxxxxx
Gr: 8, Addr:0x201C03F0, Data:0x08 = 0b000010xx, Addr:0x201C0400, Dir:0x08 = 0b000010xx
Gr: 9, Addr:0x201D035C, Data:0x93 = 0b10x1x011, Addr:0x201D0400, Dir:0x47 = 0b01x0x111
Gr:10, Addr:0x201E02FC, Data:0xBC = 0b1x111100, Addr:0x201E0400, Dir:0x80 = 0b1x000000
Gr:11, Addr:0x201F0010, Data:0x00 = 0bxxxxx0xx, Addr:0x201F0400, Dir:0x00 = 0bxxxxx0xx
Gr:12, Addr:0x202003FC, Data:0x00 = 0b00000000, Addr:0x20200400, Dir:0x00 = 0b00000000
Gr:13, Addr:0x202103FC, Data:0x00 = 0b00000000, Addr:0x20210400, Dir:0x00 = 0b00000000
Gr:14, Addr:0x20220000, Data:0x00 = 0bxxxxxxxx, Addr:0x20220400, Dir:0x00 = 0bxxxxxxxx

```

ZigFisher, 2022-07-31:
**Временное решение**
* прошиваем тестовый загрузчик для XM
* устанавливаем адреса и интерфейс
    * `setenv serverip 192.168.1.17; setenv ipaddr 192.168.1.55; setenv mdio_intf rmii; saveenv; run setnor8m`
* прошиваем свежий релиз через tftp
    * ` run uknor8m; run urnor8m; reset`

carbofos, 2022-07-31:
Это помогло, сеть появилась.

khseal, 2022-08-01:
Подверждаю сеть заработала.

closed 2026-09-04 by openipc-ai