OpenIPC/firmware #294: Device ID 00002520 sensor not working.

khseal, 2022-07-30:
root@openipc-hi3516cv100:~# ipctool

The ipctool installed as remote GitHub plugin

---
board:
  vendor: OpenIPC
  version: 2.2.06.09
chip:
  vendor: HiSilicon
  model: 3516CV100
ethernet:
  mac: "00:12:12:73:10:04"
  u-mdio-phyaddr: 1
  phy-id: 0x02430c54
  d-mdio-phyaddr: 0
  phy-mode: mii
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 45f68564
        contains:
          - name: uboot-env
            offset: 0x30000
      - name: wtf
        size: 0x10000
        sha1: a4784464
      - name: kernel
        size: 0x200000
        sha1: 0a2b585d
      - name: rootfs
        size: 0x500000
        path: /,squashfs
        sha1: 65c685ca
      - name: rootfs_data
        size: 0xb0000
        path: /overlay,jffs2,rw
    size: 8M
    addr-mode: 3-byte
ram:
  total: 128M
  media: 96M
firmware:
  kernel: "3.0.8 (Thu Jun 9 04:21:39 UTC 2022)"
  toolchain: gcc version 7.5.0 (Buildroot -gf34305f) 
  sdk: "Hi3518_MPP_V1.0.B.0  (Nov 19 2015, 16:37:04)"
sensors:
- vendor: SmartSens
  model: SC2135
  control:
    bus: 0
    type: i2c
    addr: 0x60
  vicap-state: down

dimerr, 2022-07-30:
Post output of:
```killall majestic; sleep 2; majestic```

khseal, 2022-07-30:
`root@openipc-hi3516cv100:~# killall majestic; sleep 2; majestic
killall: majestic: no process killed
13:53:03 [    main] main@128                      Majestic Lite for HiSilicon, version master+f1a54f7, built on 2022-07-29
13:53:03 [app_conf] load_config@103               Using /etc/majestic.yaml as main configuration
13:53:04 [     sdk] try_sensor_config@3311        SENSOR=sc2135
13:53:04 [     sdk] find_sensor_config@3274       matched sensor config: sc2135_i2c_dc_1080p.ini
13:53:04 [     sdk] start_sdk@399                 Using /etc/sensors/sc2135_i2c_dc_1080p.ini sensor configuration file
13:53:04 [     sdk] start_sdk@408                 App was built with MPP version: Hi3518_MPP_V1.0.B.0.
13:53:04 [     sdk] start_sdk@411                 Current MPP version: HI_VERSION=Hi3518_MPP_V1.0.B.0 
13:53:04 [     sdk] start_sdk@436                 sensor sc2135
13:53:04 [     sdk] start_sdk@448                   dev [1920x1080]@0x0 30fps, BGGR
13:53:04 [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_sc2135.so
13:53:04 [     hal] hisi_free_mem_ex@714          Free MMZ mem before allocation: 97404KB
13:53:04 [     sdk] dump_vb_configuration@2253    VB configuration:
13:53:04 [     sdk] dump_vb_configuration@2263      [0]: 3110400 x 9
13:53:04 [     sdk] dump_vb_configuration@2263      [1]: 4147200 x 1
13:53:04 [     hal] hisi_free_mem_ex@714          Free MMZ mem after allocation: 65996KB
Segmentation fault
`

dimerr, 2022-07-30:
Unfortunately we dont have a driver for this sensor.

khseal, 2022-08-01:
It is sad =(

khseal, 2022-08-03:
Most likely this camera uses IMX307.

closed 2022-08-24 by dimerr