OpenIPC/firmware #318: Deprecate Xiongmai chips support

widgetii, 2022-08-05:
Due to lack of testers for XM platform (and nobody is willing to fix osmem issue discussed in Majestic Telegram group) I suggest dropping XM support from the project by end of 2022

dimerr, 2022-08-05:
I would extend this to anyka and mstar chips
mstar needs userspace tools from android and uclibc - highly unlikely to happen
anyka is a living antipattern of how to do things, tonns of hardcode in kernel, including hardcoded gpios ann muxctls, lack of uboot sources, non working sdk's...

ZigFisher, 2022-08-05:
Gentlemen, you forgot to mention in the removal list the GrandMedia processors that are no longer produced, Ambarella which is too expensive, Rockchip which is too complicated to understand.

widgetii, 2022-08-05:
I'd take Rockchip support using freed time from outdated platforms

schneemaier, 2022-08-16:
This will be a sad day for my xm510 camera... What do you need on testing side? I can look into functionalities on my camera and fill issues like https://github.com/OpenIPC/firmware/issues/248 (where the issue is somewhere between majestic and the overlay fs). 

I would be also happy to test the GPIO for the XM510, but was unable to find any documentation, how to.

Thanks

widgetii, 2022-08-16:
There is no documentation for XM chips, everything should be obtained from reverse engineering (like digging a lot in IDA)

For XM510, fixing overlayfs would be a good starting point

schneemaier, 2022-09-05:
I played around with the overlayfs settings and I think this is a bug in the overlayfs implementation with 3.0.101 kernel. Based on the patch for xm510 the overlayfs version is v11, but what is the origin code for that patch?

HakGit, 2023-01-02:
Sad to see this, just got 40 XM530_80X30T_8M camera's and found out about this project. What can I do to help testing? I'm no programmer, just a little handy.

widgetii, 2023-01-02:
The best thing you can do - find developers who would be interested in Xiongmai platform

schneemaier, 2023-01-17:
Created a pull request to fix an issue with overlayFS https://github.com/OpenIPC/microbe-web/pull/116

closed 2026-08-16 by widgetii