OpenIPC/firmware #325: libevent build fail

cronyx, 2022-08-09:
libevent build fails after pull #320 

local build t31 firmware:

```/firmware/output/build/libevent-openipc-mmap64/CMakeFiles/CMakeTmp/src.c: В функции «main»:
/firmware/output/build/libevent-openipc-mmap64/CMakeFiles/CMakeTmp/src.c:8:48: ошибка: «struct sockaddr_storage» не содержит элемент с именем «__ss_family»; имелось в виду «ss_family»?
   (void)sizeof(((struct sockaddr_storage *)0)->__ss_family);
                                                ^~~~~~~~~~~
                                                ss_family
make[4]: *** [CMakeFiles/cmTC_49d51.dir/build.make:66: CMakeFiles/cmTC_49d51.dir/src.c.o] Ошибка 1```

cronyx, 2022-08-09:
cmake version issue:
```
v3.16.x - not ok
v3.18.x - not ok
v3.24 - ok
```

widgetii, 2022-08-09:
Opened [issue](https://github.com/libevent/libevent/issues/1321) in libevent repo.
As for temporary workaround `pip install cmake` will mitigate issue on local build machine

leonelhs, 2022-08-12:
upgrading cmake from sources to version 3.24.0 has worked for me.

commit e7fca9c6 (2022-08-14): Bump libevent to version with fixed builds with old CMake, fixes #325 (#348)
closed 2022-08-14 by widgetii