OpenIPC/firmware #377: OpenIPC reboots on  Fullhan FH8852 + GC4653

carbofos, 2022-08-19:
After update on Fulhan the system is rebooting after 10-30 seconds on start.
I did try without success:
- change mem=32M +/-2M
- disabled all majestic services

In previous version disabled JPEG did solve the problem (I can't remember majestic was crashing or the system was rebooting). Now the system does not work at all. Sometimes I could catch messages from majestic like "killed, out of memory" but in most cases it reboots silently.

The problem was solved by stopping httpd.
Also I did remove /etc/init.d/S50httpd to avoid the problem after reboot.
After this step I can use all required features like ONVIF and RTSP.
It did not reboot even after enabling JPEG feature in majestic.yaml but /image.jpg still is not availble:
```
08:27:40 [     sdk] capture_jpeg@1471             Error: create jpeg channel failed,ret=-1593687936!
```

My settings:

```
root@openipc-fh8852v200:~# 
root@openipc-fh8852v200:~# fw_printenv
baudrate=115200
phymode=RMII
ethact=FH EMAC
bootupdate=netupdate PA 0x7a0000 0x50000
bootcmd=${bootupdate};sf probe 0;sf read 0xa1000000 0x50000 0x280000;bootm 0xa1000000
soc=fh8852v200
bootdelay=3
ipaddr=192.168.1.28
serverip=192.168.1.1
stdin=serial
stdout=serial
stderr=serial
sensor=gc4653
ethaddr=00:24:b8:f3:3d:15
bootargs=mem=31M console=ttyS0,115200 root=/dev/mtdblock3 init=/init rootfstype=squashfs mtdparts=spi0.0:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)

root@openipc-fh8852v200:~# diff /rom//etc/majestic.yaml  /etc/majestic.yaml
--- /rom//etc/majestic.yaml
+++ /etc/majestic.yaml
@@ -21,9 +21,10 @@
 video0:
   enabled: true
 jpeg:
   enabled: true
+  toProgressive: false
 mjpeg:
   size: 640x360
   fps: 5
@@ -47,4 +48,4 @@
   enabled: false
 watchdog:
   enabled: true
-  timeout: 10
+  timeout: 600

```

widgetii, 2022-08-19:
Looks like low memory pressure?

What do you have in `free` output?

dimerr, 2022-08-19:
You cant just change osmem on fullhan,

https://github.com/OpenIPC/firmware/blob/master/general/package/fullhan-osdrv-fh8852v100/files/script/load_fullhan#L15

if you change osmem, then you have to recalculate all that

carbofos, 2022-08-19:
> Looks like low memory pressure?
> What do you have in `free` output?

I did run watch -n 1 free.
Here is the last data in the moment of reboot/freeze:

```
Every 1.0s: free                                                                                                                                                                                                        2022-08-19 18:45:37

              total        used        free      shared  buff/cache   available
Mem:          28444        7140       16304         104        5000       19704
Swap:             0           0           0
```