OpenIPC/firmware #390: Sensor Request: SC2035

carnei-ro, 2022-08-20:
Hey guys, I just flashed my 50H20L_S39 using the coupler file. It is now booting the OpenIPC and it is recognizing the sensor “sc2035” by default…. 
This sensor is not supported yet, right? 😩

I’d like to request your love to this sensor 😄 

Until it works, is there any workaround I can do to make my camera stream?

dimerr, 2022-08-20:
There are 2 ways:
1. Find libsns_sc2035.so for this platform.
2. Write driver yourself.

ZigFisher, 2022-09-02:
Please probe in Linux, step by step: 

1. Run `firstboot`  (board clean config and reboot)
2. Run `fw_setenv sensor sc2135; reboot`  (black hack, may be work)

carnei-ro, 2022-09-05:
Hello @ZigFisher, here is the outputs:

```
root@openipc-hi3516cv100:~# firstboot 
OpenIPC System Updater

Vendor  hisilicon
SoC     hi3516cv100
Kernel  0x62ed3b5e
RootFS  master+66c44fa, 2022-08-05

Offline upgrade or Overlay clean, skip sysupgrade version check.

Stop services, sync and fresh memory
Killing majestic . OK
Killing crond .. OK
Killing klogd ...... OK
Killing ntpd ...... OK
Killing rngd ...... OK
Killing syslogd ...... OK

OverlayFS
Erase overlay partition
Erasing 64 Kibyte @ b0000 - 100% complete.Cleanmarker written at a0000.

Unconditional reboot
client_loop: send disconnect: Broken pipe
```

Then

```
root@openipc-hi3516cv100:~# fw_setenv sensor sc2135; reboot
root@openipc-hi3516cv100:~# client_loop: send disconnect: Broken pipe
```

and

```
root@openipc-hi3516cv100:~# majestic 
16:33:00 [    main] main@128                      Majestic Lite for HiSilicon, version master+f51e488, built on 2022-08-04
16:33:00 [app_conf] load_config@103               Using /etc/majestic.yaml as main configuration
16:33:01 [watchdog] watchdog_start@101            Detected Hisilicon Watchdog
16:33:01 [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
16:33:01 [     sdk] try_sensor_config@3313        SENSOR=sc2135
16:33:01 [     sdk] find_sensor_config@3276       matched sensor config: sc2135_i2c_dc_1080p.ini
16:33:01 [     sdk] start_sdk@399                 Using /etc/sensors/sc2135_i2c_dc_1080p.ini sensor configuration file
16:33:01 [     sdk] start_sdk@408                 App was built with MPP version: Hi3518_MPP_V1.0.B.0.
16:33:01 [     sdk] start_sdk@411                 Current MPP version: HI_VERSION=Hi3518_MPP_V1.0.B.0 
16:33:01 [     sdk] start_sdk@436                 sensor sc2135
16:33:01 [     sdk] start_sdk@448                   dev [1920x1080]@0x0 30fps, BGGR
16:33:01 [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_sc2135.so
16:33:01 [  sensor] try_to_load@22                dlopen "/usr/lib/sensors/libsns_sc2135.so" error: Error loading shared library /usr/lib/sensors/libsns_sc2135.so: No such file or directory
16:33:01 [  sensor] try_to_load@19                trying to load .//usr/lib/sensors/libsns_sc2135.so
16:33:01 [  sensor] try_to_load@22                dlopen ".//usr/lib/sensors/libsns_sc2135.so" error: Error loading shared library .//usr/lib/sensors/libsns_sc2135.so: No such file or directory
16:33:01 [  sensor] try_to_load@19                trying to load /usr/lib//usr/lib/sensors/libsns_sc2135.so
16:33:01 [  sensor] try_to_load@22                dlopen "/usr/lib//usr/lib/sensors/libsns_sc2135.so" error: Error loading shared library /usr/lib//usr/lib/sensors/libsns_sc2135.so: No such file or directory
16:33:01 [    main] main@209                      Shutdown main thread
```

closed 2026-11-16 by openipc-ai