OpenIPC/firmware #400: HI3518EV300 flash bootlader failed

danielk117, 2022-08-23:
Hi,

according to https://openipc.org/cameras/vendors/hisilicon/socs/hi3518ev300?cip=192.168.1.10&sip=192.168.1.254&net=wifi&rom=nand&sd=sd I tried to flash the bootloader of my Xiaomi MJSXJ02HL.

Because I have no ethernet and only wifi , I used `fatload` and `usb` (and a OTG usb adapter with a usb stick) instead of `tftp`.

```
mw.b 0x42000000 ff 0x50000; fatload usb 0 0x42000000 u-boot-hi3518ev300-universal.bin
sf probe 0; sf erase 0x0 0x50000; sf write 0x42000000 0x0 0x50000
reset
```

It gaves me that...

```
System startup

Uncompress Ok!

U-Boot 2016.11-g3db078b-dirty (May 27 2022 - 23:20:13 +0300)hi3518ev300

Relocation Offset is: 0374b000
Relocating to 43f4b000, new gd at 43b3aef0, sp at 43b3aed0
SPI Nor:  hifmc_ip_ver_check(44): Check Flash Memory Controller v100 ...hifmc_ip_ver_check(50):  Found
hifmc_spi_nor_probe(1732): SPI Nor ID Table Version 1.0
SPI Nor(cs 0) ID: 0x1c 0x70 0x18 can't find in the ID table !!!
NAND:  SPI Nand ID Table Version 2.7
0 MiB
MMC:   Card did not respond to voltage select!
No SD device found !
hisi-sdhci: 0
hifmc_spi_nor_probe(1732): SPI Nor ID Table Version 1.0
SPI Nor(cs 0) ID: 0x1c 0x70 0x18 can't find in the ID table !!!
*** Warning - spi_flash_probe() failed, using default environment

hifmc_spi_nor_probe(1732): SPI Nor ID Table Version 1.0
SPI Nor(cs 0) ID: 0x1c 0x70 0x18 can't find in the ID table !!!
*** Warning - spi_flash_probe() failed, using default environment
```

The last 3 lines are repeated a couple of times. I can't unterupt.

I recovered succesfully using HiTool. 

I also tried flashing the `u-boot-hi3518ev300-universal.bin` with HiTool, but I got the same result as above. 

[image]

Any idea what I'm doing wrong?

dimerr, 2022-08-23:
Added your spi flash id, try reflashing bootloader

danielk117, 2022-08-24:
With the new bootloader I got a boot loop...

```
System startup

Uncompress Ok!

U-Boot 2016.11-gbc4964c-dirty (Aug 24 2022 - 00:29:14 +0300)hi3518ev300

Relocation Offset is: 0374a000
Relocating to 43f4a000, new gd at 43b39ef0, sp at 43b39ed0
SPI Nor:  hifmc_ip_ver_check(44): Check Flash Memory Controller v100 ...hifmc_ip_ver_check(50):  Found
hifmc_spi_nor_probe(1759): SPI Nor ID Table Version 1.0
hifmc_spi_nor_probe(1784): SPI Nor(cs 0) ID: 0x1c 0x70 0x18
hifmc_spi_nor_probe(1849): Block:64KB hifmc_spi_nor_probe(1850): Chip:16MB hifmc_spi_nor_probe(1851): Name:"EN25QH128A"
hifmc100_spi_nor_probe(147): SPI Nor total size: 16MB
NAND:  0 MiB
MMC:   Card did not respond to voltage select!
No SD device found !
hisi-sdhci: 0
boot from spi flashNOR Size: 353584384
Saving Environment to SPI Flash...
Erasing SPI flash...Writing to SPI flash...done
resetting ...

System startup

Uncompress Ok!

U-Boot 2016.11-gbc4964c-dirty (Aug 24 2022 - 00:29:14 +0300)hi3518ev300

Relocation Offset is: 0374a000
Relocating to 43f4a000, new gd at 43b39ef0, sp at 43b39ed0
SPI Nor:  hifmc_ip_ver_check(44): Check Flash Memory Controller v100 ...hifmc_ip_ver_check(50):  Found
hifmc_spi_nor_probe(1759): SPI Nor ID Table Version 1.0
hifmc_spi_nor_probe(1784): SPI Nor(cs 0) ID: 0x1c 0x70 0x18
hifmc_spi_nor_probe(1849): Block:64KB hifmc_spi_nor_probe(1850): Chip:16MB hifmc_spi_nor_probe(1851): Name:"EN25QH128A"
hifmc100_spi_nor_probe(147): SPI Nor total size: 16MB
NAND:  0 MiB
MMC:   Card did not respond to voltage select!
No SD device found !
hisi-sdhci: 0
boot from spi flashNOR Size: 353584384
Saving Environment to SPI Flash...
Erasing SPI flash...Writing to SPI flash...done
resetting ...
```

dimerr, 2022-08-24:
updated,try again

danielk117, 2022-08-24:
now it is working. thank you!

closed 2022-08-24 by danielk117