OpenIPC/firmware #419: add RTL8189FTV driver to HI3518EV300

danielk117, 2022-08-29:
Hi,
I tried adding the RTL8189FTV driver driver to OpenIPC firmware for HI3518EV300, but the built images does not contain any RTL8189FTV related files. 

branch: https://github.com/OpenIPC/firmware/compare/master...danielk117:firmware:master
firmware: https://github.com/danielk117/firmware/releases/download/latest/openipc.hi3518ev300-br.tgz
build log: https://github.com/danielk117/firmware/commit/2d15250e3cf7e5f75bf3f428f75cd63e21cdb531/checks/8034925614/logs

I missed something or did something wrong. Has someone a hint for me, how to fix this? Thank you!

dimerr, 2022-08-29:
Have you tried local build? it doesn't look like driver is built, probably there's something in it's makefile that is preventing it from build and gives graceful exit though

dimerr, 2022-08-29:
https://github.com/jwrdegoede/rtl8189ES_linux/blob/master/Makefile#L38 i'd start from here, there is no CONFIG_RTL8189FTV there

danielk117, 2022-08-29:
Yes, `CONFIG_RTL8189FTV` makes no sense. I have no local enviroment for testing the build.

I just followed https://linux-sunxi.org/Wifi#RTL8189FTV and the Makefile in that branch https://github.com/jwrdegoede/rtl8189ES_linux/blob/rtl8189fs/Makefile has 
`CONFIG_RTL8188F = y` and `CONFIG_SDIO_HCI = y` as default. So the outputted files could be named something like `8189fs`.

danielk117, 2022-09-13:
I tried a couple of ENV combinations, but I wasn't able to get the driver build. It always results in something like [this](https://github.com/danielk117/firmware/commit/51b1423340a80afd9113057bcf486cbe40e7c136/checks/8154942031/logs).

This whole buildroot and make things are really new to me. So someone with more make and driver knowledge will probably have to do it.

niudvb, 2022-10-20:
I use openipc source code ,add "BR2_PACKAGE_RTL8189FS=y" in hi3518ev200 defconfig,and add "CONFIG_RTL8189FS=m" in hi3518ev200.generic.config, then "make BOARD=unknown_unknown_hi3518ev200_openipc all".Eventually 8189fs can used .
The problem is that when I run "load_hisilicon -i -sensor jxh62 -total 64", dmesg gets "himci: card disconnected!" SDIO WIFI
become unavailable.

danielk117, 2022-11-01:
At the telegram channel for the alternative micam mjsxj02hl firmware, the maintainer of the firmware tells @ZigFisher how he build the driver for openipc. Hopefully he will add the driver soon. 😉

https://t.me/mjsxj02hl_firmware/4920