OpenIPC/firmware #424: No network connection on Hi3516CV100

carlsa, 2022-08-30:
I have flashed my `IPG5020A-H-V1.0` board from stock firmware to `OpenIPC master+e4d73334, 2022-08-30` but can't access the network after flashing.

The network works in U-BOOT, but not when I have booted in to Linux.

I looked at issue #156 and have tried this without success:
- Boot to U-BOOT, ping and run `run bootcmd`
- Run `devmem 120100ec 32 9f`

The Ethernet transceiver chip is an [IP101GR](http://www.bdtic.com/DataSheet/ICplus/IP101G_DS_R01_20121224.pdf).

The chip marking is `HI3516 CRBCV100 CD6191716 1716-CHINA`

Here are some logs:
[2022-08-30_dmsg.log](https://github.com/OpenIPC/firmware/files/9452942/2022-08-30_dmsg.log)
[2022-08-30_serial_boot_log.log](https://github.com/OpenIPC/firmware/files/9452944/2022-08-30_serial_boot_log.log)
[2022-08-30_proc_cpuinfo.txt](https://github.com/OpenIPC/firmware/files/9452961/2022-08-30_proc_cpuinfo.txt)

ZigFisher, 2022-09-02:
There is some chance to try the "extras" options for your board to write into the U-Boot ENV.

carlsa, 2022-09-05:
Thank you for the suggestion!

I got it working with these extras that I found [here](https://github.com/hi35xx/hi35xx-buildroot/issues/4#issuecomment-312980395):
`setenv extras hieth.mdioifu=0 hieth.mdioifd=0 hieth.phyaddru=1 hieth.phyaddrd=2`

closed 2022-09-05 by carlsa