OpenIPC/firmware #427: open_sys Segmentation fault on hi3518ev300

widgetii, 2022-09-01:
```
hisilicon: Loading of kernel modules and initialization of the video system has started
+ board=demo
+ ipcinfo --chip-name
+ chipid=hi3518ev300
+ mem_start=0x40000000
+ tr -d M
+ fw_printenv -n totalmem
+ mem_total=64
+ mem_total=64
+ tr -d M
+ fw_printenv -n osmem
+ os_mem_size=32
+ os_mem_size=32
+ YUV_TYPE0=0
+ b_arg_sensor0=0
+ b_arg_sensor1=0
+ b_arg_yuv_type0=0
+ b_arg_insmod=0
+ b_arg_remove=0
+ b_arg_os_mem=0
+ b_arg_board=0
+ '[' 0 -eq 1 ]
+ '[' 0 -eq 1 ]
+ '[' 0 -eq 1 ]
+ '[' 0 -eq 1 ]
+ b_arg_insmod=1
+ '[' 32 -ge 64 ]
+ calc_mmz_info
+ awk 'BEGIN { temp = 0; }
        {
                temp = $1/1024/1024 + $2;
        }
        END { printf("0x%x00000\n", temp); }'
+ echo '0x40000000 32'
+ mmz_start=0x42000000
+ awk 'BEGIN { temp = 0; }
        {
                temp = $1 - $2;
        }
        END { printf("%dM\n", temp); }'
+ echo '64 32'
+ mmz_size=32M
+ echo 'mmz_start: 0x42000000, mmz_size: 32M'
mmz_start: 0x42000000, mmz_size: 32M
+ '[' 1 -lt 1 ]
+ '[' -n  ]
+ fw_printenv -n sensor
+ fw_printenv -n sensor
+ SENSOR_ENV=jxf23
+ export 'SENSOR=jxf23'
+ logger -s -p daemon.info -t hisilicon 'Get data from environment and set SENSOR as jxf23'
hisilicon: Get data from environment and set SENSOR as jxf23
+ '[' 0 -eq 1 ]
+ '[' jxf23 '=' unknown ]
+ '[' 1 -eq 1 ]
+ uname -r
+ cd /lib/modules/4.9.37/hisilicon
+ insert_ko
+ modprobe open_sys_config 'chip=hi3518ev300' 'sensors=jxf23' 'g_cmos_yuv_flag=0' 'board=demo'
+ insert_osal
+ check_allocator
+ grep mmz_allocator /proc/cmdline
+ allocator='mem=32M console=ttyAMA0,115200 panic=20 rootfstype=squashfs root=/dev/mtdblock3 init=/init mtdparts=hi_sfc:256k(boot),64k(env),3072k(kernel),10240k(rootfs),-(rootfs_data) mmz_allocator=hisi'
+ '[' -z 'mem=32M console=ttyAMA0,115200 panic=20 rootfstype=squashfs root=/dev/mtdblock3 init=/init mtdparts=hi_sfc:256k(boot),64k(env),3072k(kernel),10240k(rootfs),-(rootfs_data) mmz_allocator=hisi' ]
+ awk -F '=' '$1=="mmz"{print $2}' 'RS= ' /proc/cmdline
+ MMZ=
+ '[' -z  ]
+ modprobe open_osal 'anony=1' 'mmz_allocator=hisi' 'mmz=anonymous,0,0x42000000,32M'
+ modprobe open_base
+ modprobe open_sys
Segmentation fault
+ modprobe open_rgn
+ modprobe open_vgs
```

HugasRadvilenas, 2022-09-12:
looks like problem solved , works ok now after latest firmware update

widgetii, 2022-09-12:
Thank you for report!

closed 2022-09-12 by widgetii