OpenIPC/firmware #442: ONVIF button in disabled state in web interface does not disable ONVIF

carbofos, 2022-09-08:
В вебинтерфейсе кнопка отключения onvif не дает отключения этого протокола по факту.
Рег все равно получает свои ответы. Но если включить и выключить, чтобы в majestic.yaml появилась запись onvif.enabled=false, тогда уже отключает.

themactep, 2022-09-09:
Нет, не так.

Вот дефолный конфиг, изменений нет:

[image]

При открытии этой формы веб запрашивает у yaml-cli состояние onvif в конфиге; yaml-cli отвечает, что у него нет данных:
```
root@openipc-hi3518ev100-web:~# yaml-cli -g .onvif.enabled
root@openipc-hi3518ev100-web:~# 
```
Т.е. значение onvif.enabled находится в квантовой суперпозиции. Но веб считает все не однозначно включеное выключеным и рисует переключатеть в положении "выключено".

[image]

Здесь я нажимаю кнопку "Save Changes" даже не меняя положение переключателя.

[image]

В конфиг записан параметр для onvif - не включать.

[image]

После чего дается команда на перечитывание конфига маджестиком.

https://github.com/OpenIPC/microbe-web/blob/d6ab350f55220920667eda958d04e4afe62a8d81/files/var/www/cgi-bin/majestic-settings.cgi#L74

Если что-то работает -- давайте разбираться по шагам.

longcat99, 2022-11-04:
Maybe you should reset the config file.  I have encountered this situation, because I flashed other firmware and caused the configuration file to be wrong, some settings did not take effect.

closed 2024-10-26 by flyrouter