OpenIPC/firmware #455: NC-HBR2200B / HI3516CV300 boot kernel panic

DSchndr, 2022-09-12:
Hi there, device gets kernel panic while booting (probably because of spi flash)
Flash is gd 25q127csig, flashed full image on it

Kernel log: 
```
                         

System startup

Uncompress........Ok

System startup

U-Boot 2010.06 (Sep 07 2022 - 08:23:51)

Check Flash Memory Controller v100 ... Found
SPI Nor(cs 0) ID: 0xc8 0x40 0x18
Block:64KB Chip:16MB Name:"GD25Q128"
SPI Nor total size: 16MB
MMC:   
EMMC/MMC/SD controller initialization.
Card did not respond to voltage select!
No EMMC/MMC/SD device found !
In:    serial
Out:   serial
Err:   serial
RAM size: 256MB
Hit any key to stop autoboot:  1  0 
16384 KiB hi_fmc at 0:0 is now current device

## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   Linux-3.18.20-hi3516cv300
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1935195 Bytes = 1.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Linux version 3.18.20 (runner@fv-az460-541) (gcc version 8.4.0 (Buildroot -gb3c48f37) ) #1 Sun Sep 11 12:05:31 UTC 2022
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine model: Hisilicon Hi3516CV300 DEMO Board
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
[    0.000000] Kernel command line: mem=32M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
[    0.000000] PID hash table entries: 128 (order: -3, 512 bytes)
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Memory: 27056K/32768K available (3664K kernel code, 171K rwdata, 1044K rodata, 160K init, 239K bss, 5712K reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xc2800000 - 0xff000000   ( 968 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc04a176c   (4710 kB)
[    0.000000]       .init : 0xc04a2000 - 0xc04ca000   ( 160 kB)
[    0.000000]       .data : 0xc04ca000 - 0xc04f4fe0   ( 172 kB)
[    0.000000]        .bss : 0xc04f4fe0 - 0xc0530de0   ( 240 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] VIC @c2800000: id 0x00641190, vendor 0x41
[    0.000032] sched_clock: 32 bits at 3000kHz, resolution 333ns, wraps every 1431655765682ns
[    0.000769] Console: colour dummy device 80x30
[... 132 lines cut ...]
[    2.601608] bridge: automatic filtering via arp/ip/ip6tables has been deprecated. Update your scripts to load br_netfilter if you need this.
[    2.629219] List of all partitions:
[    2.632845] 1f00             256 mtdblock0  (driver?)
[    2.637910] 1f01              64 mtdblock1  (driver?)
[    2.643041] 1f02            2048 mtdblock2  (driver?)
[    2.648093] 1f03            5120 mtdblock3  (driver?)
[    2.653169] 1f04            8896 mtdblock4  (driver?)
[    2.658216] No filesystem could mount root, tried:  squashfs
[    2.663902] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
[    2.672255] CPU: 0 PID: 1 Comm: swapper Not tainted 3.18.20 #1
[    2.678136] [<c0012f1c>] (unwind_backtrace) from [<c00112d4>] (show_stack+0x10/0x14)
[    2.685903] [<c00112d4>] (show_stack) from [<c03751f8>] (panic+0x80/0x1dc)
[    2.692792] [<c03751f8>] (panic) from [<c04a31e8>] (mount_block_root+0x1d8/0x204)
[    2.700280] [<c04a31e8>] (mount_block_root) from [<c04a33d8>] (mount_root+0xc8/0xec)
[    2.708024] [<c04a33d8>] (mount_root) from [<c04a3554>] (prepare_namespace+0x158/0x1b8)
[    2.716030] [<c04a3554>] (prepare_namespace) from [<c04a2d94>] (kernel_init_freeable+0x164/0x1a8)
[    2.724906] [<c04a2d94>] (kernel_init_freeable) from [<c0374350>] (kernel_init+0x8/0xe0)
[    2.733000] [<c0374350>] (kernel_init) from [<c000e448>] (ret_from_fork+0x14/0x2c)
[    2.740567] Rebooting in 20 seconds..                         

System startup

Uncompress........Ok

```

themactep, 2022-09-12:
SInce it's a 16MB flash chip you need to change partitioning to match the 16MB image. Please run `setnor16m` in bootloader console.

DSchndr, 2022-09-12:
Thanks, it worked :D

But why do I still need to do it after flashing the 16m ultimate image over a programmer? seems like an issue to me :P

closed 2022-09-12 by DSchndr

themactep, 2022-09-12:
The environment that comes with our _universal_ bootloader is pre-set for 8MB flash. Although it has all the macro, they still need to be run once after flashing. If you have an idea how to amend it feel free to share :)

DSchndr, 2022-09-12:
Would have to look into it, would be nice if i could flash my cameras without disassembly xD
(Or i just compile myselff with 16mb preset)
Tought of using flashrom to directly flash the spi flash, i know the root password so it should be possible or is there another way?
(Did not dissasemble the hunter binary but it has symbols iirc so i could reverse engineer how the manufacturer updates over webif work)

themactep, 2022-09-12:
Have you seen https://github.com/OpenIPC/coupler ? If your camera is supported then you could reflash it without disassembling.

DSchndr, 2022-09-18:
@themactep `The environment that comes with our universal bootloader is pre-set for 8MB flash. Although it has all the macro, they still need to be run once after flashing. If you have an idea how to amend it feel free to share :)`

In uboot: 
check flashed env -> continue normal booting
else
check firstboot8m variable, if it exists, delete it and try to boot with 8m mtd (reset)
If rebooted and firstboot8m does not exist, check firstboot16m, set 16m mtd and reset
If both variables are missing start tftp server and reset both variables after successfull flash

If boot succeeded set "flashed" env from userspace

(Or use bootcounter which would be better since tftp recovery could be started on x failed boots)

I kinda did a firstboot check then tftp for those cameras I can now flash really easy over telnet
example:
```
setenv boot "setenv setargs setenv bootargs ${bootargs}; run setargs; sf probe 0; sf read ${baseaddr} 0x50000 0x200000; bootm ${baseaddr}"
setenv download "setenv ipaddr 192.168.1.16; setenv serverip 10.0.2.67; setenv ethaddr e2:08:4f:72:b3:f5; saveenv; run uknor16m; run urnor16m; run setnor16m; setenv firstboot 1; saveenv; reset"
setenv bootcmd "if printenv firstboot; then run boot; else run download; fi"
```