OpenIPC/firmware #476: Can I help with Ingenic T21?

felipecrs, 2022-09-19:
I have 4 cameras with this hardware here, and they are basically useless with the original firmware they have.

Is there anything I can do to help accelerate the development for supporting Ingenic T21? Maybe performing some tests, or providing information?

Regardless, thanks a lot for the great initiative. This project is for sure changing the world for the good.

ZigFisher, 2022-09-19:
Greetings
You can still install the OpenIPC firmware, but you will need to make some effort to figure out the original U-Boot. In any case you can get more detailed OnLine consultations in our chat groups. https://openipc.org/our-channels

dimerr, 2022-09-20:
T21 is supported and even tested on a couple of boards, do you have some specific problem?

felipecrs, 2022-09-20:
> T21 is supported and even tested on a couple of boards, do you have some specific problem?

Since the documentation (https://openipc.org/cameras/vendors/ingenic) lists T21 with missing docs and U-Boot, I was afraid to get started.

firedevel, 2022-11-05:
i have 4 cameras too, （Mi 2K Pro SSC323DE). What can I do?

felipecrs, 2022-11-25:
Any chance we can have precompiled u-boot for T21 as well?

[image]

I'm just waiting for it for trying this out. :)

dimerr, 2022-11-25:
We dont have SDK for T21, so no u-boot

felipecrs, 2022-11-25:
I see. In regards to the installation docs, do you think I can use the same commands as T20 or some other model?

dimerr, 2022-11-25:
t10-t31, they are all the same

dimerr, 2022-11-25:
https://bbs.16rd.com/thread-487003-1-1.html it could be a sdk for t21, if you have access to this site..

CapnRon, 2022-12-27:
I have hacked it onto my camera. 
[image]

It's MIT license so I will post my firmware here and contribute my improvements back to openIPC and the community. 

It has a couple of issues but does boot and run. It is for 16mb NOR flash and the original test camera is a "Wansview K5"

It has a couple of issues but does boot and run. Unfortunately I haven't been able to get the mt7601U usb wifi working yet on my camera. also I haven't been able to get the env partition quite right so that may need some work. Anyway you are welcome to try it. I seriously don't recommend trying it unless you have a way to dump your original flash chip and write directly onto the chip via a hardware flasher. I'm not responsible if you brick your cam or any thing else etc...

Below I have posted a link to the bin and my serial console boot log

https://drive.google.com/file/d/1dsWOYe07lSxktahhOqAS1VZcpL8RDP1Q/view?usp=sharing

`U-Boot SPL 2013.07 (Aug 26 2019 - 16:11:21)
Timer init
CLK stop
PLL init
pll_init:399
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
fbdiv = 36 , refdiv = 1 , fdivq = 2 ,pllod = 1 range = 3
cppcr is 02300860
CPM_CPAPCR 0470484d
fbdiv = 38 , refdiv = 1 , fdivq = 2 ,pllod = 1 range = 3
cppcr is 02500860
CPM_CPMPCR 04a0484d
fbdiv = 50 , refdiv = 1 , fdivq = 2 ,pllod = 1 range = 3
cppcr is 03100860
CPM_CPVPCR 0310086d
cppcr 0x9a773310
apll_freq 864000000
mpll_freq 900000000
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 450000000
cclk  864000000
l2clk 432000000
h0clk 180000000
h2clk 180000000
pclk  90000000
CLK init
SDRAM init
sdram init start
ddr_inno_phy_init ..!
phy reg = 0x00000007, CL = 0x00000007
ddr_inno_phy_init ..! 11:  00000004
ddr_inno_phy_init ..! 22:  00000006
ddr_inno_phy_init ..! 33:  00000006
REG_DDR_LMR: 00000210
REG_DDR_LMR: 00000310
REG_DDR_LMR: 00000110
REG_DDR_LMR, MR0: 00d73011
T30_0x5: 00000007
T30_0x15: 0000000c
T30_0x4: 00000000
T30_0x14: 00000002
INNO_TRAINING_CTRL 1: 00000000
INNO_TRAINING_CTRL 2: 000000a1
T30_cc: 00000003
INNO_TRAINING_CTRL 3: 000000a0
T30_118: 0000003c
T30_158: 0000003c
T30_190: 00000020
T30_194: 0000001d
jz-04 :  0x00000051
jz-08 :  0x000000a0
jz-28 :  0x00000024
DDR PHY init OK
INNO_DQ_WIDTH   :00000003
INNO_PLL_FBDIV  :00000014
INNO_PLL_PDIV   :00000005
[... 236 lines cut ...]
Motion detection is disabled in config!
2022-12-26 01:56:13 [665:motion] Remove lock file.
2022-12-26 01:56:13 [665:motion] Exiting
FAIL
ingenic: Check MAC for Ingenic devices
Redundant environments have unequal size
Redundant environments have unequal size
ingenic: The eth0 interface has a correct MAC -
ingenic: Loading of kernel modules and initialization of the video system has started
Redundant environments have unequal size
ingenic: Get data from sinfo and set SENSOR as ov2735b
Redundant environments have unequal size
Error: environment not initialized
--------------------
ISP_PARAM: isp_clk=125000000
SENSOR: ov2735b
SENSOR_PARAM:
--------------------
Starting majestic: OK

Welcome to OpenIPC
openipc-t21 login: root
Redundant environments have unequal size
Redundant environments have unequal size
`

CapnRon, 2022-12-29:
I'm unable to find the '/lib/modules/3.10.14/kernel/fs/overlayfs/'  directory in the t21 bin file, I also looked in the t20 and t10 bins?

root@openipc-t21:/overlay# mount -t overlay overlay -o lowerdir=/,upperdir=/overlay/root,workdir=/overlay/work /mnt
mount: mounting overlay on /mnt failed: No such device

root@openipc-t21:/# lsmod
Module                  Size  Used by    Tainted: P
sensor_ov2735b_t21      9640  0
audio                  73090  0
tx_isp_t21            659258  1 sensor_ov2735b_t21
sinfo                  16508  0
mt7601Usta           1952725  1
cfg80211              257984  1 mt7601Usta
mtprealloc              2033  1 mt7601Usta
vfat                    9477  0
fat                    56289  1 vfat
f2fs                  132539  0

root@openipc-t21:/# modprobe -l
kernel/fs/fat/fat.ko
kernel/fs/fat/vfat.ko
kernel/fs/f2fs/f2fs.ko
kernel/drivers/net/wireless/rtlwifi/rtlwifi.ko
kernel/drivers/net/wireless/rtlwifi/rtl8192c/rtl8192c-common.ko
kernel/drivers/net/wireless/rtlwifi/rtl8192cu/rtl8192cu.ko
kernel/drivers/net/tun.ko
kernel/drivers/usb/serial/usbserial.ko
kernel/net/wireless/cfg80211.ko
kernel/net/mac80211/mac80211.ko
ingenic/reset.ko
ingenic/audioout.ko
ingenic/gpio.ko
extra/wireguard.ko
ingenic/sensor_jxf23_t21.ko
ingenic/sinfo.ko
ingenic/sensor_sc2232_t21.ko
ingenic/audio.ko
ingenic/ircut.ko
ingenic/rled.ko
ingenic/motor.ko
ingenic/sensor_sc2335_t21.ko
ingenic/tx-isp-t21.ko
ingenic/wifien.ko

it is listed in '/proc/filesystems'  and in '/lib/modules/3.10.14/modules.builtin'

root@openipc-t21:/# cat /proc/filesystems
nodev   sysfs
nodev   rootfs
nodev   bdev
nodev   proc
nodev   cgroup
nodev   tmpfs
nodev   devtmpfs
nodev   debugfs
nodev   sockfs
nodev   pipefs
nodev   anon_inodefs
nodev   rpc_pipefs
nodev   configfs
nodev   devpts
        squashfs
nodev   ramfs
nodev   nfs
nodev   jffs2
nodev   overlayfs
nodev   mtd_inodefs
        f2fs
        vfat

[... 79 lines cut ...]
./modules.builtin: line 79: kernel/sound/soundcore.ko: not found
./modules.builtin: line 80: kernel/sound/sound_firmware.ko: not found
./modules.builtin: line 81: kernel/net/802/p8022.ko: not found
./modules.builtin: line 82: kernel/net/802/psnap.ko: not found
./modules.builtin: line 83: kernel/net/802/stp.ko: not found
./modules.builtin: line 84: kernel/net/bridge/bridge.ko: not found
./modules.builtin: line 85: kernel/net/ipv4/ip_tunnel.ko: not found
./modules.builtin: line 86: kernel/net/ipv4/ipip.ko: not found
./modules.builtin: line 87: kernel/net/ipv4/esp4.ko: not found
./modules.builtin: line 88: kernel/net/ipv4/tunnel4.ko: not found
./modules.builtin: line 89: kernel/net/ipv4/inet_diag.ko: not found
./modules.builtin: line 90: kernel/net/ipv4/tcp_diag.ko: not found
./modules.builtin: line 91: kernel/net/ipv4/tcp_cubic.ko: not found
./modules.builtin: line 92: kernel/net/ipv6/ipv6.ko: not found
./modules.builtin: line 93: kernel/net/ipv6/inet6_hashtables.ko: not found
./modules.builtin: line 94: kernel/net/llc/llc.ko: not found
./modules.builtin: line 95: kernel/net/packet/af_packet.ko: not found
./modules.builtin: line 96: kernel/net/sunrpc/sunrpc.ko: not found
./modules.builtin: line 97: kernel/net/unix/unix.ko: not found
./modules.builtin: line 98: kernel/net/xfrm/xfrm_algo.ko: not found
./modules.builtin: line 99: kernel/lib/bitrev.ko: not found
./modules.builtin: line 100: kernel/lib/crc32.ko: not found
./modules.builtin: line 101: kernel/lib/xz/xz_dec.ko: not found
./modules.builtin: line 102: kernel/lib/zlib_deflate/zlib_deflate.ko: not found
./modules.builtin: line 103: kernel/lib/zlib_inflate/zlib_inflate.ko: not found

dimerr, 2022-12-29:
Try searching here https://github.com/OpenIPC/firmware/blob/master/br-ext-chip-ingenic/board/t21/kernel/t21.generic.config#L1607

CapnRon, 2022-12-29:
Thank you, I see its compiled in the kernel. I have sorted my networking issue and resized my partitions in u-boot such that rootfs_apps is the largest partition on the flash and cleared all the errors in dmesg aside from the random number generator.

My webUI loads, but pretty much everything complains about a readonly filesystem. 

I have tried running this which I found in /init

mount -t jffs2 /dev/mtdblock4 /overlay
mount -t overlayfs overlayfs -o lowerdir=/,upperdir=/overlay,ro /mnt
pivot_root /mnt /mnt/rom
mount -o noatime,move /rom/proc /proc
mount -o noatime,move /rom/dev /dev
mount -o noatime,move /rom/overlay /overlay

which executes with no issues however it brings down the webUI and other strange issues in the serial console

I tried the alternative 

root@openipc-t21:/overlay# mount -t overlay overlay -o lowerdir=/,upperdir=/overlay/root,workdir=/overlay/work /mnt

but get this error

mount: mounting overlay on /mnt failed: No such device

It seems like I'm so close to having a working openIPC but this whole overlayfs thing is new to me. 

Is there a doc that will guide me to the finish line here, or am I missing something?

Included here is my current firmware bin 

https://drive.google.com/file/d/1KHwFtL2RGQedch1NqD6XFvYkXfzFFE9i/view?usp=sharing

Thanks again for your help

dimerr, 2022-12-29:
It't not clear from your messages, what are you trying to do or achieve? OpenIPC for t21 just works. There's no need to check kernel configs or mount overlays or fix scripts.

CapnRon, 2022-12-29:
I would just like it to work as intended. From stock it mounts the root filesystem as read only?

Due to the read only filesystem I needed to add modules for my wifi card and wpa_supplicant.conf before programming the bin onto my flash chip with an spi programmer, which got wireless networking working. 

I was then able to use the webUI, but again because of the readonly filesystem none of the settings were able to be adjusted and saved.  It can't even format the SDcard

[image]
[image]

Unfortunately in my case OpenIPC doesnt "just work", it may be a problem with my boot partition in the first 256k that I copied from the original camera firmware but its the only one I have to work with. 

My toolset I have been using so far is dd, cat, binwalk, firmware mod toolkit, a hex editor, and a TL866II plus programmer. I socketed the flash chip on my camera.  

Would you be so kind as to post a link to a working chip dump bin of the 16mb nor flash so I can write it directly to my chip with my programmer to test with? 

Thank you again

CapnRon, 2022-12-31:
Ok I have converted the root Squashfs to JFFS2, this has given me RW on root filesystem. Things are starting to work now I can successfully use the webUI to format an SDcard or save changes made in the UI. Now it seems the last hurdle may be the camera sensor.

Included here is my current firmware bin file :  https://drive.google.com/file/d/1FR1N6k29crgPHd3ugBNjIVKP0VihJUHi/view?usp=sharing

[image]
 
 below is a piece of my dmesg showing errors about the sensor which may explain why I only get the test pattern on the Preview on it webUI. Also I noticed after running Nmap only my SSH and HTTP ports are open, no RTSP port 554, I just wanted to mention this now but until we get the sensor working, RTSP is useless.
 
 Does anyone know how I might fix this issue and get the camera sensor working properly?
 
 Thanks again for all your help! Hopefully together we can get this working soon
 
 Take care

 
 [   20.381865] RTMP_TimerListAdd: add timer obj c07925d4!
[   20.557491] RTMP_TimerListAdd: add timer obj c07905b0!
[   20.773110] RTMP_TimerListAdd: add timer obj c07905f0!
[   20.865780] name : i2c0 nr : 0
[   20.950480] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   20.950508] i2c i2c-0: --I2C txabrt:
[   20.950524] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   20.950549] error: sensor_read,285 ret = -5
[   20.950559]  sensor_read: addr=0xa value = 0x0
[   20.950569] err sensor read addr = 0xa, value = 0x0
[   21.040496] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   21.040524] i2c i2c-0: --I2C txabrt:
[   21.040540] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   21.040565] error: sensor_read,285 ret = -5
[   21.040576]  sensor_read: addr=0x300a value = 0x0
[   21.040586] err sensor read addr = 0x300a, value = 0x0
[   21.130486] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   21.130513] i2c i2c-0: --I2C txabrt:
[   21.130529] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   21.130554] error: sensor_read,285 ret = -5
[   21.130565]  sensor_read: addr=0x300b value = 0x0
[   21.130575] err sensor read addr = 0x300b, value = 0x0
[   21.220517] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   21.220545] i2c i2c-0: --I2C txabrt:
[   21.220561] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   21.220586] error: sensor_read,285 ret = -5
[   21.220596]  sensor_read: addr=0xa value = 0x0
[   21.220606] err sensor read addr = 0xa, value = 0x0
[   21.310479] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   21.310507] i2c i2c-0: --I2C txabrt:
[   21.310523] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   21.310549] error: sensor_read,285 ret = -5
[   21.310559]  sensor_read: addr=0x580b value = 0x0
[   21.310569] err sensor read addr = 0x580b, value = 0x0
[   21.400507] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   21.400535] i2c i2c-0: --I2C txabrt:
[   21.400551] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   21.400576] error: sensor_read,285 ret = -5
[   21.400587]  sensor_read: addr=0x580b value = 0x0
[   21.400597] err sensor read addr = 0x580b, value = 0x0
[   21.490497] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   21.490525] i2c i2c-0: --I2C txabrt:
[   21.490541] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   21.490566] error: sensor_read,285 ret = -5
[   21.490577]  sensor_read: addr=0x3107 value = 0x0
[   21.490587] err sensor read addr = 0x3107, value = 0x0
[   21.580484] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   21.580513] i2c i2c-0: --I2C txabrt:
[   21.580529] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   21.580553] error: sensor_read,285 ret = -5
[   21.580564]  sensor_read: addr=0x3107 value = 0x0
[   21.580574] err sensor read addr = 0x3107, value = 0x0
[   21.670486] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   21.670513] i2c i2c-0: --I2C txabrt:
[   21.670530] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[... 278 lines cut ...]
[   50.033356] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   50.033373] i2c i2c-0: --I2C txabrt:
[   50.033387] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   50.033572] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   50.033589] i2c i2c-0: --I2C txabrt:
[   50.033604] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   50.033789] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   50.033805] i2c i2c-0: --I2C txabrt:
[   50.033820] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   50.034007] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   50.034024] i2c i2c-0: --I2C txabrt:
[   50.034039] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   50.034224] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   50.034241] i2c i2c-0: --I2C txabrt:
[   50.034255] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   50.034440] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   50.034457] i2c i2c-0: --I2C txabrt:
[   50.034471] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   50.034656] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[   50.034673] i2c i2c-0: --I2C txabrt:
[   50.034687] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[   55.282328] Rcv Wcid(1) AddBAReq
[   55.282350] Start Seq = 00000000
[   55.282371] RTMP_TimerListAdd: add timer obj c079267c!
root@openipc-t21:~#

dimerr, 2023-01-01:
You have to take 2735b ko and bin files from stock firmware, we dont have them

@felipecrs I've added our own uboot for t21n, you can try and see if it works https://github.com/OpenIPC/firmware/releases/download/latest/u-boot-t21n-universal.bin

felipecrs, 2023-01-01:
Hey @dimerr, I found out that the description of my camera in its website was wrong. My camera isn't a T21 but a T20. Therefore I can't test it. But thanks a lot anyway.

felipecrs, 2023-01-09:
U-boot and docs seems to be available now in the OpenIPC website. Thus closing this issue.

closed 2023-01-09 by felipecrs

CapnRon, 2023-01-09:
@dimerr I have tested the t21n u-boot you posted as well as the newly updated bin on the main openIPC page. Neither allow my t21 to run at all. Perhaps the u-boot for t21n is different than for the t21?

dimerr, 2023-01-10:
Maybe you have t21l?