OpenIPC/firmware #481: Majestic's default openipc.local should just point to /etc/hostname

brainstorm, 2022-09-23:
That way folks can easily change the default mDNS/Zeroconf advertised domain name without having to reverse the `majestic` binary...

[image]

widgetii, 2022-09-23:
We need a second opinion from other project developers about changing behavior for this feature. From my point of view, end-user doesn't care about specific name of hardware platform, e.g. it better to ping/ssh to `openipc.local` than `openipc-t10-cam.lan`, but for experienced users for sure it's better to have everything they want.

BTW, some good routers have same feature with autodiscovery option and as for me, it's better to implement this as well.

brainstorm, 2022-09-23:
Fair enough ;)

Well, one big problem with the hardcoded openipc.local in majestic is when there's more that one OpenIPC camera on the same LAN.

Another less disruptive way to implement this would be to add a section on majestic.yaml that allows to specify a camera name or switch to /etc/hostname explicitly from that yaml.

Sigh, it's such a pity that Majestic is closed source, IMHO it's perhaps the only downside of **Open**IPC at this point in time ;)

widgetii, 2022-09-23:
Here's its small brother [mini](https://github.com/openipc/mini), but there are no PRs since 2019

We're still looking for more developers to our small team

themactep, 2022-09-28:
My 1.17 ruble: if hostname is the default one then advertise as openipc.local, otherweise use <hostname>.local

widgetii, 2022-09-28:
It's hard to determine, is it "default" because for each platform currently we have different name

themactep, 2022-09-28:
check /etc/hostname against /rom/etc/hostname. 
`cmp -s /etc/hostname /rom/etc/hostname && echo "same"`

p0i5k, 2022-09-28:
Check /etc/hostname in overlay is better. What if i want to use the same name?

cronyx, 2022-09-28:
Just add the last 3 bytes of the mac address to the name from /etc/hostname: openipc-t31-xxxxxx

brainstorm, 2022-11-21:
First step I'd take with this issue is to remove this particular functionality from Majestic and put it on a separate service, i.e:

https://github.com/sudomesh/mdnssd-min/

Or instead of removing, at least make it on/off switchable via `majestic.yaml` so that future mDNS solutions do not clash with Majestic.

cdg123, 2023-05-26:
Has there been any more thoughts on this as I've got three cameras on the same LAN and would ideally like to use their .local names to address them. I'm not sure many people would mind where it is implemented but I would prefer the hostname is used as set in /etc/hostname?