OpenIPC/firmware #491: No network access with u-boot for hi3518cv100

PinkFreud, 2022-09-26:
I decided to try installing openipc on my old Foscam (FI9821W V2.1, HI3518CV100).  While following through the instructions, I made it as far as flashing the u-boot image (https://github.com/OpenIPC/firmware/releases/download/latest/u-boot-hi3518cv100-universal.bin) and repartitioning flash for openipc.  The next step has me install openipc firmware from tftp (where I successfully saved a backup of my original flash contents in the first step!).  Unfortunately, I can no longer reach the tftp server at this point - both tftp and ping fails.

I can see the network interface reset when I try to start tftp or ping - the serial console tells me the interface is down, followed by another line informing me the interface is back up.  If I look at the switch when this happens, I see the link light turn off briefly, followed by it turning back on.  Unfortunately, that's the only thing I get - if I watch for traffic from the camera board with wireshark, I see nothing.  Network functionality in the above u-boot image seems to simply be broken.

Is this a known issue?  Is there anything I can do to recover at this point without having to unsolder the flash IC from the board and connecting to it directly?

DSchndr, 2022-09-27:
I don't know if xmodem upload works but "ubootwrite" should be able to upload kernel and squashfs over serial if you have time...

CKlymko, 2023-11-22:
I think this may be a foscam issue because I'm seeing very similar results with the F19826w with the hi3518cv100.
I was able to downgrade the foscam firmware to a version where tftp was enabled in uboot. I made a backup and did all the required checking for which gpios did what and where and all that stuff. When I flashed the full image (lite was the only one available for this chip) the network was dead.
There's a physical link light that flashes but that's it. There's no network in uboot, and no network in the openipc system. I even checked mii in uboot and there are no devices showing up but there were 0 and 1 showing up in the default firmware.
I thought maybe some gpio needed to pop up to enable the network, but I couldn't get a reply out of toggling any of the gpios. As in it wouldn't even respond to toggling the known IR CUT or any other gpio settings.
Oddly though when I pull the ethernet cable I do see activity on the serial console and I see that it is trying to find the network adapter when I plug the cable back in. It seems like there's a bunch of stuff missing for this chip with regards to it being a foscam.

closed 2026-11-16 by openipc-ai