OpenIPC/firmware #502: HI3516AV100 отсутствует eth0

mixatronik, 2022-10-06:
В новых сборках HI3516AV100 отсутствует eth0. Куда копать?

widgetii, 2022-10-06:
Если в старых все работало, но нужно искать когда и чем сломалось

mixatronik, 2022-10-07:
К сожалению, при попытке прошить православный бут камера окирпичилась и было это в конце мая. Последняя из рабочих сборок, что у меня сохранились - от 27.05.22 9:30. Другая аналогичная камера появилась только теперь и проблема проявилась. Вот такая вот вилка.
Если есть какой-то архив, то готов оттестировать все, что между маем и октябрём, чтобы понять, на какой сломалось.

widgetii, 2022-10-07:
Все архивы хранятся в [dev канале](https://t.me/s/openipc_dev). Чтобы сократить число попыток, воспользуйтесь бинарным поиском

ZigFisher, 2022-10-07:
> В новых сборках HI3516AV100 отсутствует eth0 на камере DS-2CD2345F-IS. Куда копать?

Необходимо в переменных extras и bootargs U-boot выставить правильные значения.

mixatronik, 2022-10-07:
> Все архивы хранятся в [dev канале](https://t.me/s/openipc_dev). Чтобы сократить число попыток, воспользуйтесь бинарным поиском

про дихотомию в курсе :)

mixatronik, 2022-10-07:
> > В новых сборках HI3516AV100 отсутствует eth0 на камере DS-2CD2345F-IS. Куда копать?
> 
> Необходимо в переменных extras и bootargs U-boot выставить правильные значения.

как узнать какие? и extras у меня вообще нет.

mixatronik, 2022-10-07:
Нашёл - в сборке от 31.07.22 интерфейс eth0 присутствует, а в сборке от 1.08.22 его уже нет. Что-то ещё нужно? Лог загрузки может быть?

widgetii, 2022-10-07:
Под дату и описание подходит [этот коммит](https://github.com/OpenIPC/firmware/commit/2a757c3a1901f425c8e9f22c204872b2b95cc670)

ZigFisher, 2022-10-07:
На части плат автодетект не работает, по-этому отключено.

mixatronik, 2022-10-08:
что и где прописать для инициализации eth0?

mixatronik, 2022-10-26:
Проблема локализована. Есть рабочая сборка и идёт работа над тем, как сделать так, чтобы работало у всех.

closed 2022-10-26 by mixatronik