OpenIPC/firmware #517: IMX385 on Hi3516AV200

gitgayhub, 2022-10-19:
Hi, thanks for this great project! I'm trying to run OpenIPC on a camera based on Hi3516AV200 and IMX385. I've installed the newest OpenIPC firmware without any problem, but can't get majestic to show any image.

### Camera ino
SoC: Hi3516AV200
Sensor: IMX385
Memory Size: 256MB
Flash Size: 128MB Nand
```bash
root@openipc-hi3516av200:~# ipctool
The ipctool installed as remote GitHub plugin---
board:
  vendor: OpenIPC
  version: 2.2.10.17
chip:
  vendor: HiSilicon
  model: 3516AV200
ethernet:
  mac: "00:00:23:34:45:66"
  phy-mode: rmii
rom:
  - type: nand
    block: 128K
    partitions:
      - name: uboot
        size: 0xc0000
        sha1: 2b1faaf9
        contains:
          - name: uboot-env
            offset: 0x60000
      - name: wtf
        size: 0x40000
        sha1: 1e4903cd
      - name: kernel
        size: 0x300000
        sha1: 9d0cc1b0
      - name: ubi
        size: 0x7c00000
        sha1: 53fda73a
    size: 128M
ram:
  total: 256M
  media: 64M
firmware:
  u-boot: "2010.06 (Mar 21 2019 - 16:49:26)"
  kernel: "3.18.20 (SMP Mon Oct 17 14:10:31 UTC 2022)"
  toolchain: gcc version 8.4.0 (Buildroot -gbc3606dc-dirty)
  sdk: "Hi3519V101_MPP_V1.0.5.0 B040 Release (Jul  5 2018, 18:13:06)"
sensors:
- vendor: Sony
  model: IMX385
  control:
    bus: 0
    type: i2c
    addr: 0x34
  vicap0-state: down
  vicap1-state: down
```
### ```fw_printenv```
```
bootdelay=1
baudrate=115200
mdio_intf=rmii
ethaddr=00:00:23:34:45:66
bootfile="uImage"
filesize=BA0000
fileaddr=82000000
netmask=255.255.255.0
ipaddr=192.168.2.37
serverip=192.168.2.52
bootcmd=nand read 0x82000000 0x100000 0x300000;bootm 0x82000000
[... 223 lines cut ...]
     0             0              0             0              0

-----ALING Err info--------------------------------------
 Devno FIFO_FullErr  Lane0Err  Lane1Err  Lane2Err  Lane3Err  Lane4Err  Lane5Err  Lane6Err  Lane7Err  Lane8Err  Lane9Err Lane10Err Lane11Err
     0            0         0         0         0         0         0         0         0         0         0         0         0         0
```
### And this is the libsns_imx385.so file from stock firmware
[libsns_imx385.so.zip](https://github.com/OpenIPC/firmware/files/9819652/libsns_imx385.so.zip)

I tried to replace /usr/lib/sensors/libsns_imx385.so with the one from stock firmware, but majestic gave this message:
```
05:16:11  <       majestic> [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx385.so
05:16:11  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem before allocation: 65420KB
05:16:11  <       majestic> [     sdk] dump_vb_configuration@2135    VB configuration:
05:16:11  <       majestic> [     sdk] dump_vb_configuration@2145      [0]: 3110400 x 3
05:16:11  <       majestic> [     sdk] dump_vb_configuration@2145      [1]: 3110400 x 1
05:16:11  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem after allocation: 53256KB
05:16:11  <       majestic> [  sensor] sensor_getsnsobj@103          Cannot load symbol 'stSnsObj' from driver
05:16:11  <       majestic> [  sensor] sensor_getsnsobj@104          Run: strings /usr/lib/sensors/libsns_imx385.so | grep -e ^stSns
05:16:11  <       majestic> [  sensor] sensor_getsnsobj@105          to demermine proper Sensor_type parameter
05:16:11  <       majestic> [     sdk] start_sdk@477                 Cannot init sensor
05:16:11  <       majestic> [    main] main@209                      Shutdown main thread
```

Please let me know if you need any other information and thank you again for this great project

widgetii, 2022-10-19:
We don't have such hardware in our lab, so nothing is guaranteed
Feel free to donate hardware if you want to have it support in the future

gitgayhub, 2022-10-19:
> Feel free to donate hardware if you want to have it support in the future

Thanks, how can I donate hardware, where should I ship it to?

gitgayhub, 2022-10-19:
I can also arrange remote ssh access if you need

gitgayhub, 2022-10-19:
Got into the shell of stock firmware

### ```/nfsroot # ./ipctool```
```
---
chip:
  vendor: HiSilicon
  model: 3516AV200
ethernet:
  mac: "marked"
  phy-mode: rmii
rom:
  - type: nand
    block: 128K
    partitions:
      - name: boot
        size: 0x100000
        sha1: b77adba9
        contains:
          - name: uboot-env
            offset: 0x60000
      - name: kernel
        size: 0x400000
        sha1: a6af50e1
      - name: rootfs
        size: 0x800000
        sha1: b6241040
      - name: data
        size: 0x1000000
        sha1: be646d15
      - name: upgradefs
        size: 0x1000000
        sha1: 1f050995
      - name: appfs
        size: 0x4000000
        sha1: 9217002e
      - name: exdata
        size: 0x1300000
        sha1: b35f9c75
    size: 128M
ram:
  total: 512M
  media: 256M
firmware:
  u-boot: "2010.06 (Mar 21 2019 - 16:49:26)"
  kernel: "3.18.20 (SMP Tue May 7 13:37:26 CST 2019)"
  toolchain: gcc version 4.9.4 20150629 (prerelease) (Hisilicon_v500_20160223)
  libc: uClibc 0.9.33.2
  sdk: "Hi3519V101_MPP_V1.0.4.0 B040 Release (Nov 10 2017, 14:08:45)"
  main-app: /app/bin/robcat
sensors:
- vendor: Sony
  model: IMX385
  control:
    bus: 0
    type: i2c
    addr: 0x34
  vicap0-state: up
  vicap1-state: down
  data:
    vicap0-input: MIPI0
    mipi0-type: MIPI
```
### ```cat /proc/umap/vi```
```
[VIU] Version: [Hi3519V101_MPP_V1.0.4.0 B040 Release], Build Time: [Nov 10 2017, 14:08:58]

-----MODULE PARAM---------------------------------------------------------------
detect_err_frame  drop_err_frame   stop_int_level    yuv_skip
              10               0                0           0
[... 170 lines cut ...]
     0  2       0       0
     0  3       0       0

-----fsm timeout and escape info---------------------------------------------
 link clkTOutCnt d0TOutCnt   d1TOutCnt   d2TOutCnt   d3TOutCnt clkEscCnt  d0EscCnt  d1EscCnt  d2EscCnt  d3EscCnt
    0     0           0           0           0           0         0         0         0         0         0
    1     0           0           0           0           0         0         0         0         0         0
    2     0           0           0           0           0         0         0         0         0         0
-----MIPI int error info-----------------------------------------------------------

 Devno vc0CRC vc1CRC vc2CRC vc3CRC vc0OrderErr vc1OrderErr vc2OrderErr vc3OrderErr vc0NMatCnt vc1NMatCnt vc2NMatCnt vc3NMatCnt
     0      0      0      0      0           0           0           0           0          0          0          0          0

 Devno HCntErr vc0HECC vc1HECC vc2HECC vc3HECC vc0DtErr vc1DtErr vc2DtErr vc3DtErr
     0       0       0       0       0       0        0        0        0        0

 Devno CMD_FIFO_RERR DATA_FIFO_RERR CMD_FIFO_WERR DATA_FIFO_WERR
     0             0              0             0              0

-----ALING Err info--------------------------------------
 Devno FIFO_FullErr  Lane0Err  Lane1Err  Lane2Err  Lane3Err  Lane4Err  Lane5Err  Lane6Err  Lane7Err  Lane8Err  Lane9Err Lane10Err Lane11Err
     0            0         0         0         0         0         0         0         0         0         0         0         0         0
```

Will wait for the camera to arrive your place