OpenIPC/firmware #520: "Invalid sensor type imx291" & "open sys: No such file or directory" on Hi3516CV300

gitgayhub, 2022-10-21:
Tried OpenIPC Lite and Ultimate for Hi3516CV300, but both with the same error message

```
hisilicon: Loading of kernel modules and initialization of the video system has started
mmz_start: 0x82000000, mmz_size: 96M
hisilicon: Get data from environment and set SENSOR as imx291
xxxx Invalid sensor type imx291 xxxx
******* Error: There's something wrong, please check! *****
Starting majestic: OK
```

### ```SENSOR=imx291 majestic```
```
04:15:12  <       majestic> [    main] main@125                      Majestic ultimate for HISILICON, version HEAD+6f69f02, built on 2022-10-20
04:15:12  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
04:15:12  <       majestic> [watchdog] watchdog_start@82             Watchdog device (/dev/watchdog) not found
04:15:12  <       majestic> [     sdk] try_sensor_config@3202        SENSOR=imx291
04:15:12  <       majestic> [     sdk] find_sensor_config@3165       matched sensor config: imx291_i2c_lvds_1080p_line.ini
04:15:12  <       majestic> [     sdk] start_sdk@402                 Using /etc/sensors/imx291_i2c_lvds_1080p_line.ini sensor configuration file
04:15:12  <       majestic> [     hal] HiSi_HAL_start_logger@619     Cannot open /dev/logmpp
04:15:12  <       majestic> [     sdk] start_sdk@411                 App was built with MPP version: Hi3516CV300_MPP_V1.0.4.0.B050 Release
04:15:12  <       majestic> [     sdk] start_sdk@414                 Current MPP version: HI_VERSION=Hi3516CV300_MPP_V1.0.4.0 B050 Release
open sys: No such file or directory
04:15:12  <       majestic> [     sdk] start_sdk@423                 Cannot get chip id
04:15:12  <       majestic> [     sdk] start_sdk@423                 Unknown error code 2
04:15:13  <       majestic> [    main] main@209                      Shutdown main thread

```

### Board Info
SoC: Hi3516CV300
Flash: 16M NOR
Memory: 128M
Sensor: IMX291

### ipctool inside stock firmware
```
../../ipctool
---
chip:
  vendor: HiSilicon
  model: 3516CV300
ethernet:
  mac: "marked"
  u-mdio-phyaddr: 1
  phy-id: 0xffffffff
  d-mdio-phyaddr: 0
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 05329b26
      - name: kernel
        size: 0x200000
        sha1: 4d85ad2e
      - name: userdata
        size: 0x80000
        path: /mnt/nand,jffs2,rw
      - name: rootfs
        size: 0xcc0000
        sha1: c465590a
      - name: userdata_bk
        size: 0x80000
        path: /mnt/data_backup,jffs2,rw
    size: 16M
    addr-mode: 3-byte
ram:
  total: 128M
  media: 61M
firmware:
  kernel: "3.18.20 (Wed Sep 20 10:29:01 CST 2017)"
  toolchain: gcc version 4.9.4 20150629 (prerelease) (Hisilicon_v500_20170615)
  libc: uClibc 0.9.33.2
  sdk: "Hi3516CV300_MPP_V1.0.3.0 B020 Release (Nov 26 2017, 12:44:11)"
  main-app: /opt/topsee/encode_stream
sensors:
- vendor: Sony
  model: IMX291
  control:
    bus: 0
    type: i2c
    addr: 0x34
  params:
    bitness: 12
    databus: LVDS 4 ch
    fps: 30
  data:
    type: LVDS
    lane-id:
    - 0
    - 1
    - 2
    - 3
    lvds-wdr-en: 0
    lvds-wdr-mode: 0
    lvds-wdr-num: 0
    raw-data-type: RAW_DATA_12BIT
    sync-mode: LVDS_SYNC_MODE_SAV
    data-endian: LVDS_ENDIAN_BIG
    sync-code-endian: LVDS_ENDIAN_BIG
    sync-code:
    -
      - 0xab0, 0xb60, 0x800, 0x9d0
      - 0xab0, 0xb60, 0x800, 0x9d0
      - 0xab0, 0xb60, 0x800, 0x9d0
      - 0xab0, 0xb60, 0x800, 0x9d0
    -
      - 0xab0, 0xb60, 0x800, 0x9d0
      - 0xab0, 0xb60, 0x800, 0x9d0
      - 0xab0, 0xb60, 0x800, 0x9d0
      - 0xab0, 0xb60, 0x800, 0x9d0
    -
      - 0xab0, 0xb60, 0x800, 0x9d0
      - 0xab0, 0xb60, 0x800, 0x9d0
      - 0xab0, 0xb60, 0x800, 0x9d0
      - 0xab0, 0xb60, 0x800, 0x9d0
    -
      - 0xab0, 0xb60, 0x800, 0x9d0
      - 0xab0, 0xb60, 0x800, 0x9d0
      - 0xab0, 0xb60, 0x800, 0x9d0
      - 0xab0, 0xb60, 0x800, 0x9d0
  clock: 37.125MHz
```
### ipctool inside OpenIPC
```
The ipctool installed as remote GitHub plugin---
board:
  vendor: OpenIPC
  version: 2.2.10.20
chip:
  vendor: HiSilicon
  model: 3516CV300
ethernet:
  mac: "00:00:23:34:45:66"
  u-mdio-phyaddr: 1
  phy-id: 0xffffffff
  d-mdio-phyaddr: 0
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 17cef7fc
        contains:
          - name: uboot-env
            offset: 0x30000
      - name: env
        size: 0x10000
        sha1: 472a55b0
      - name: kernel
        size: 0x200000
        sha1: 6b2e847c
      - name: rootfs
        size: 0x800000
        sha1: ca8d64cf
      - name: rootfs_data
        size: 0x5b0000
        path: /overlay,jffs2,rw
    size: 16M
    addr-mode: 3-byte
ram:
  total: 32M
firmware:
  u-boot: "2010.06 (Sep 20 2017 - 09:48:18)"
  kernel: "3.18.20 (Thu Oct 20 20:43:29 UTC 2022)"
  toolchain: gcc version 8.4.0 (Buildroot -gbc3606dc-dirty)
sensors:
- vendor: Sony
  model: IMX291
  control:
    bus: 0
    type: i2c
    addr: 0x34
  params:
    bitness: 12
    databus: Parallel CMOS SDR
    fps: 30
  vicap-state: down
```
### ```fw_printenv```
```
bootdelay=1
baudrate=115200
ethaddr=00:00:23:34:45:66
bootfile="uImage"
bootcmd=sf probe 0;sf read 0x81000000 0x50000 0x200000;bootm 0x81000000
soc=hi3516cv300
totalmem=128M
filesize=5F9000
fileaddr=82000000
netmask=255.255.255.0
ipaddr=192.168.2.37
serverip=192.168.2.88
bootargs=mem=32M totalmem=128M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 init=/init rootfstype=squashfs mtdparts=hi_sfc:256k(boot),64k(env),2048k(kernel),8192k(rootfs),-(rootfs_data)
osmem=32M
stdin=serial
stdout=serial
stderr=serial
verify=n
ver=U-Boot 2010.06 (Sep 20 2017 - 09:48:18)
sensor=imx291
```

Please let me know if you need any other info for debug

widgetii, 2022-10-21:
It seems that kernel modules weren't loaded (could be checked by `lsmod`)

Try to run `fw_setenv sensor imx291_i2c_lvds` and then reboot

gitgayhub, 2022-10-21:
> 

Thanks for the quick support, and the problem is quickly solved

```lsmod``` showed only 3 modules loaded
```
Module                  Size  Used by    Not tainted
vfat                    9127  0
fat                    54439  1 vfat
f2fs                  138066  0
```
After setting sensor as ```imx291_i2c_lvds``` and rebooting, majestic is running without problem

closed 2022-10-21 by gitgayhub