OpenIPC/firmware #525: hi3516dv100  не работает встроенный микрофон

mixatronik, 2022-10-26:
Камера  DS-2CD2345F-IS. Работает линейный аудиовход, но е работает встроенный микрофон. Переключение источника в конфиге Маджестика не даёт никакого результата. Звук как шёл с линейного входа, так и идёт, а с микрофона как не было, так и нет.

mixatronik, 2022-10-30:
https://disk.yandex.ru/d/Kibcz9SEOnKhMQ - оригинал
https://disk.yandex.ru/d/VEoi8d6N8Y-u2A - OpenIPC

правда, оригинал не до конца получилось запустить, но может быть, того что есть будет достаточно. или нет?

widgetii, 2022-10-30:
```
--- 2345.txt	2022-10-30 17:39:13.000000000 +0300
+++ OpenIPC.txt	2022-10-30 17:39:20.000000000 +0300
@@ -1,15 +1,15 @@ 
 -----ACODEC PARAM--------------------------------------------------------------
-   LGain(dB)   RGain(dB) DacLVol(dB) DacRVol(dB) AdcLVol(dB) AdcRVol(dB) MicLMut MicRMut DacLMut DacRMut   Boost
-      28.5      28.5        -1          -1           0           0           0       0       1       1       0
+   LGain(dB)   RGain(dB) DacLVol(dB) DacRVol(dB) AdcLVol(dB) AdcRVol(dB) MicLMut MicRMut DacLMut DacRMut  BoostL  BoostR
+       0.0       0.0        -1          -1           0           0           0       0       1       1       0       0
   
 -----AI DEV ATTR---------------------------------------------------------------------------------------------------------
 AiDev WorkMod   SampR  BitWid  ChnCnt  ClkSel SoundMod  PoiNum  ExFlag  FrmNum
-    0 i2s_mas   16kHZ   16bit       2       0     mono     320       1      30
 
 -----AI DEV STATUS-------------------------------------------------------------------------------------------------------
 AiDev      IntCnt     fifoCnt     buffInt FrmTime  MaxFrmTime TranLen IsrTime  MaxIsrTime     CBPhy  CBSize   ROffSet   WOffSet
-    0       26779           0           0   19984       20109    1280     110         134  83c19000    2560       500         0
 
 -----AI DEV EXTEND STATUS------------------------------------------------------------------------------------------------
 AiDev enTrack   bMute  Volume
-    0       2       0       0
 
 ```

mixatronik, 2022-10-30:
это я заметил, но о чём это говорит - хз. подозреваю, что показывает уровень сигнала... что с этой информацией делать?

widgetii, 2022-10-30:
Нужно собрать `mpp/sample/audio` и подобрать такие же параметры для аудио, как на оригинальной прошивке. Там же есть примеры как с помощью регистров настраивать I2S параметры внешних аудиокодеков

mixatronik, 2022-10-30:
а поподробнее?

widgetii, 2026-11-10:
Проверили на актуальной прошивке на hi3516dv100 — сейчас ситуация обратная той, что была в 2022 году.

Встроенный кодек Hi3516A/DV100 имеет два аналоговых входа перед одним усилителем: линейный и микрофонный. Драйвер кодека при каждой установке громкости захвата заодно переключает вход на микрофонный. Поэтому все текущие сборки majestic с ненулевой `audio.volume` фактически пишут с **микрофонного** входа (в `/proc/umap/acodec` это видно как `BoostL/BoostR = 1`), а линейный вход был недоступен. В вашем дампе 2022 года громкость ещё не выставлялась (`LGain 0.0`, `Boost 0`), поэтому тогда работал линейный вход, а микрофон молчал.

В majestic добавлен параметр `audio.input` (только для hi3516av100/dv100), он выбирает вход и сохраняет выбор при изменении громкости:

```yaml
audio:
  enabled: true
  input: mic   # mic (по умолчанию) или line
```

В веб-интерфейсе это выпадающий список «Microphone input» в разделе Audio. Изменение появится в ближайшей ночной сборке.

Замер на dv100 при `audio.volume: 80`, без источника сигнала на входах: `mic` — около −56 dBFS (шум предусилителя микрофонного входа), `line` — около −75 dBFS. Чувствительность настраивается через `audio.volume`. Заводская прошивка в вашем дампе держала усиление +28.5 dB, это примерно `audio.volume: 75`.

Если камера ещё у вас, проверьте, пожалуйста, на свежей ночной сборке оба значения `audio.input` с реальным звуком, работают ли и встроенный микрофон, и линейный вход.