OpenIPC/firmware #541: Can webrtc customize ice?

longcat99, 2022-11-04:
I checked the source and it seems to be ice provided by Amazon.  This is very unfriendly to people from China or other distant places.  If I build stun and turn with coturn myself.  The camera connection will be very fast.  Mainly intended for low-latency viewing.  For 4g remote control vehicles or drones.

widgetii, 2022-11-06:
New configuration option was added:

```json
webrtc:
  # https://www.w3.org/TR/webrtc/#rtciceserver-dictionary with optional
  # '?transport=udp' or '?transport=tcp'
  iceServers: stun:stun.kinesisvideo.eu-north-1.amazonaws.com:443
```

closed 2022-11-06 by widgetii

longcat99, 2022-11-07:
First of all, thank you for your positive contribution. 
Is it OK to write directly in the configuration file? 
I'm a little confused.