OpenIPC/firmware #563: No video output from HI3516EV300_IMX335

gnif, 2022-11-17:
Hi, I have just installed OpenIPC onto a camera here and I am getting no image out of the device.

```
root@openipc-hi3516ev300:~# ipctool
---
board:
  vendor: OpenIPC
  version: 2.2.11.17
chip:
  vendor: HiSilicon
  model: 3516EV300
  id: 002836877e0038f94a3d0fac66894d1320e5c200111525e3
ethernet:
  mac: "8a:31:1e:76:bf:49"
  u-mdio-phyaddr: 1
  phy-id: 0x20669903
  d-mdio-phyaddr: 0
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 3f34dd7a
      - name: env
        size: 0x10000
        sha1: 55b73783
        contains:
          - name: uboot-env
            offset: 0x0
      - name: kernel
        size: 0x200000
        sha1: 61d6a081
      - name: rootfs
        size: 0x500000
        sha1: 977d52a9
      - name: rootfs_data
        size: 0xb0000
        path: /overlay,jffs2,rw
    size: 8M
    addr-mode: 3-byte
ram:
  total: 128M
  media: 96M
firmware:
  kernel: "4.9.37 (Thu Nov 17 10:40:29 UTC 2022)"
  toolchain: gcc version 8.4.0 (Buildroot -gff5aa3b6-dirty) 
  sdk: "Hi3516EV200_MPP_V1.0.1.2 B030 Release (Oct 18 2019, 18:21:00)"
  main-app: /usr/bin/majestic
sensors:
- vendor: Sony
  model: IMX335
  control:
    bus: 0
    type: i2c
    addr: 0x34
  data:
    type: MIPI
    input-data-type: DATA_TYPE_RAW_10BIT
    lane-id:
    - 0
    - 1
    - 2
    - 3
    image: 2592x1520
  clock: 37.125MHz
```

```
root@openipc-hi3516ev300:~# fw_printenv
[... 87 lines cut ...]
14:02:27  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
14:02:29  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
14:02:29  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
14:02:31  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
14:02:31  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
14:02:33  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
14:02:33  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
^C14:02:34  <       majestic> [     log] printf@268                    Interrupt signal received
14:02:36  <  thread-pool-1> [     hal] take_yuv420sp_video_fram@1156 Cannot get chn 0 frame for vpssGrp=0
14:02:36  <  thread-pool-1> [     hal] take_yuv420sp_video_fram@1156 ERR_VPSS_BUF_EMPTY: The picture queue is empty
14:02:36  <  thread-pool-1> [image_co] image_control_compute_hist@99 Cannot take yuv420sp VI frame
14:02:36  <  thread-pool-1> [image_co] image_control_compute_hist@99 ERR_VPSS_BUF_EMPTY: The picture queue is empty
14:02:36  <  thread-pool-1> [image_tu] tune_image_params@186         get_histogram errored
14:02:36  <      hi_logger> [    vpss] vpss_get_chnframe_check_w@401 vpss(0,0) user get frame time out, s32MilliSec=10000
14:02:39  <  thread-pool-0> [     hal] take_yuv420sp_video_fram@1156 Cannot get chn 0 frame for vpssGrp=0
14:02:39  <  thread-pool-0> [     hal] take_yuv420sp_video_fram@1156 ERR_VPSS_BUF_EMPTY: The picture queue is empty
14:02:39  <  thread-pool-0> [image_co] image_control_compute_hist@99 Cannot take yuv420sp VI frame
14:02:39  <  thread-pool-0> [image_co] image_control_compute_hist@99 ERR_VPSS_BUF_EMPTY: The picture queue is empty
14:02:39  <  thread-pool-0> [image_tu] tune_image_params@186         get_histogram errored
14:02:39  <      hi_logger> [    vpss] vpss_get_chnframe_check_w@401 vpss(0,0) user get frame time out, s32MilliSec=10000
14:02:39  <       majestic> [     sdk] stop_chn@1942                 Stopped 0 channel
14:02:39  <      hi_logger> [      vi] VI_WaitCaptureFrameStop@2280  Wait Pipe 0 Capture stop timeout 120ms!!!
14:02:39  <       majestic> [     sdk] stop_sdk@3149                 Stop sdk Ok!
14:02:39  <       majestic> [    main] main@209                      Shutdown main thread
```

dimerr, 2022-11-17:
was it working before flashing?

gnif, 2022-11-17:
Unfortunately I do not know, I was unable to figure out how to obtain a stream from the device as it came with no documentation.

gnif, 2022-11-17:
I have another device which I have just verified I can get the camera stream from with the factory firmware. If it's of any use I have a programmer and can dump the factory image. Uboot can't be used to download it due to missing `tftpput` support.

FlyRouter, 2022-11-17:
Please provide a photo of the board, preferably several and from different sides

gnif, 2022-11-17:
Sure thing, in the meantime here is the boot log from the factory firmware
```
~ # dmesg
Booting Linux on physical CPU 0x0
Linux version 4.9.37 (root@localhost6.1) (gcc version 6.3.0 (HC&C V100R002C00B032_20190114) ) #5 Wed Feb 27 16:38:20 CST 2019
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Hisilicon HI3516EV200 DEMO Board
Memory policy: Data cache writeback
On node 0 totalpages: 14336
free_area_init_node: node 0, pgdat c040f81c, node_mem_map c3789000
  Normal zone: 112 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 14336 pages, LIFO batch:3
CPU: All CPU(s) started in SVC mode.
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0 
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 14224
Kernel command line: mem=56M console=ttyAMA0,115200 root=/dev/mtdblock3 noinitrd rootfstype=squashfs mtdparts=hi_sfc:256K(boot),1536K(kernel),768K(userdata)
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 52372K/57344K available (3142K kernel code, 119K rwdata, 716K rodata, 140K init, 209K bss, 4972K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4000000 - 0xff800000   ( 952 MB)
    lowmem  : 0xc0000000 - 0xc3800000   (  56 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc0319b48   (3143 kB)
      .init : 0xc03cf000 - 0xc03f2000   ( 140 kB)
      .data : 0xc03f2000 - 0xc040fe60   ( 120 kB)
       .bss : 0xc0411000 - 0xc044573c   ( 210 kB)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:16 nr_irqs:16 16
Gic dist init...
arm_arch_timer: Architected cp15 timer(s) running at 50.00MHz (phys).
clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb8812736b, max_idle_ns: 440795202655 ns
sched_clock: 56 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
Switching to timer-based delay loop, resolution 20ns
route: SIOCADDRT: File exists
dmesg
Booting Linux on physical CPU 0x0
Linux version 4.9.37 (root@localhost6.1) (gcc version 6.3.0 (HC&C V100R002C00B032_20190114) ) #5 Wed Feb 27 16:38:20 CST 2019
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Hisilicon HI3516EV200 DEMO Board
Memory policy: Data cache writeback
On node 0 totalpages: 14336
free_area_init_node: node 0, pgdat c040f81c, node_mem_map c3789000
  Normal zone: 112 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 14336 pages, LIFO batch:3
CPU: All CPU(s) started in SVC mode.
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0 
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 14224
Kernel command line: mem=56M console=ttyAMA0,115200 root=/dev/mtdblock3 noinitrd rootfstype=squashfs mtdparts=hi_sfc:256K(boot),1536K(kernel),768K(userdata),5504K(rootfs),64K(userdata_bk),64K(log)
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 52372K/57344K available (3142K kernel code, 119K rwdata, 716K rodata, 140K init, 209K bss, 4972K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4000000 - 0xff800000   ( 952 MB)
    lowmem  : 0xc0000000 - 0xc3800000   (  56 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
[... 92 lines cut ...]
hi_osal 1.0 init success!
hi3516ev200_base: module license 'Proprietary' taints kernel.
Disabling lock debugging due to kernel taint
load sys.ko for Hi3516EV200...OK!
load region.ko for Hi3516EV200...OK!
load vgs.ko for Hi3516EV200...OK!
load vi.ko for Hi3516EV200...OK !
ISP Mod init!
load vpss.ko for Hi3516EV200...OK!
load rc.ko for Hi3516EV200...OK!
load venc.ko for Hi3516EV200...OK!
load chnl.ko for Hi3516EV200...OK!
load vedu.ko for Hi3516EV200...OK!
load h264e.ko for Hi3516EV200...OK!
load h265e.ko for Hi3516EV200...OK!
load ive.ko for Hi3516EV200...OK!
load mipi_rx driver successful!
register dev
Hisilicon Watchdog Timer: 0.01 initialized. default_margin=30 sec (nodeamon= 1)
hiwtdg init ok. ver=Mar 11 2019, 09:05:33.
jffs2: notice: (760) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
hisi-femac 10040000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
random: fast init done
random: crng init done
```

gnif, 2022-11-17:
[image]
[image]

dimerr, 2022-11-17:
Its a topsee board. I think you need to backup /proc/umap/* from stock firmware

gnif, 2022-11-17:
```
/proc/umap # for F in *; do echo -e "$F\n========="; cat $F; done
acodec
=========

[ACODEC] Version: [Hi3516EV200_MPP_V1.0.0.3 B040 Release], Build Time: [Mar 11 2019, 09:05:33]

-----ACODEC PARAM--------------------------------------------------------------
   LGain(dB)   RGain(dB) DacLVol(dB) DacRVol(dB) AdcLVol(dB) AdcRVol(dB) MicLMut MicRMut DacLMut DacRMut  BoostL  BoostR
      26.0      -1.5         6           0           0           0           0       0       0       0       1       1
adec
=========

[ADEC] Version: [Hi3516EV200_MPP_V1.0.0.3 B040 Release], Build Time[Mar 11 2019, 09:05:34]

-----ADEC CHN ATTR--------------------------------------------------------------
ChnId  PlType  ADPCMType BufSize        G726Rate    Mode OriSendCnt    SendCnt     GetCnt     PutCnt
    0   g711u       NULL      20            NULL  stream          0          0          0          0
aenc
=========

[AENC] Version: [Hi3516EV200_MPP_V1.0.0.3 B040 Release], Build Time[Mar 11 2019, 09:05:36]

-----AENC CHN ATTR--------------------------------------------------------------
ChnId  PlType  ADPCMType  PoiNum BufSize        G726Rate
    0   g711u       NULL     320      30            NULL

-----AENC CHN STATUS------------------------------------------------------------
ChnId     RcvFrm  AiQueLost      EncOk     FrmErr    BufFull    GetStrm    RlsStrm
    0       1399          0       1399          0          0       1399       1399
ai
=========

[AI] Version: [Hi3516EV200_MPP_V1.0.0.3 B040 Release], Build Time: [Mar 11 2019, 09:05:33]

-----AI DEV ATTR----------------------------------------------------------------
AiDev WorkMod   SampR  BitWid  ChnCnt  ClkSel SoundMod  PoiNum  ExFlag  FrmNum
    0 i2s_mas    8kHZ   16bit       1       0     mono     320       0      30

-----AI DEV STATUS0-------------------------------------------------------------
AiDev     IntCnt    fifoCnt    buffInt FrmTime MaxFrmTime TranLen IsrTime 
    0       1399          0          0   40003      40101     640      65

-----AI DEV STATUS1-------------------------------------------------------------
AiDev MaxIsrTime      CBPhy  CBSize    ROffSet    WOffSet
    0         86   4789f000    1920        280          0

-----AI DEV EXTEND STATUS-------------------------------------------------------
AiDev enTrack   bMute  Volume
    0       0       N       0

-----AI CHN STATUS--------------------------------------------------------------
AiDev   AiChn   State     BufFul UsrQueLost UsrFrmDepth   u32Data0   u32Data1    UserGet    UserRls
    0       0  enable          0          0          30          0          0       1399          0

-----AI CHN RESAMPLE STATUS-----------------------------------------------------
AiDev AiChn   State  bResmp  PoiNum   InSampR  OutSampR
    0     0  enable       N       0    (null)    (null)

-----AI CHN VQE STATUS0---------------------------------------------------------
AiDev AiChn   State bVqe workmod    RATE  PoiNum bAgc  bEq bHpf bRnr bHdr bDrc  WrFile
    0     0  enable    Y   noisy    8kHZ     320    N    N    Y    N    N    N       N

-----AI CHN VQE STATUS1---------------------------------------------------------
AiDev AiChn   State    bAnr bUsrmod NrIntensity NoiseDbThr SpProSwi
    0     0  enable       Y       Y          15         40        0

-----AI CHN VQE STATUS2---------------------------------------------------------
AiDev AiChn   State    bHpf bUsrmod HpfFreq
    0     0  enable       Y       Y      80
[... 1047 lines cut ...]
   GrpID   ChnID  Enable  viewType   XOffset   YOffset       DistortionRatio       MinRatio

-------------------------------VPSS CHN LOWDELAY ATTR-------------------------------------------------------
   GrpID   ChnID  Enable   LineCnt   OneBufEnable   OneBufAddr

-------------------------------VPSS CHN BUF WRAP ATTR-------------------------------------------------------
   GrpID   ChnID  Enable   BufLine   WrapBufSize
       0       0       N         0             0

-------------------------------FRAME INTERRUPT ATTR---------------------------------------------------------
   GrpID        IntType      EarlyLine
       0              -              0

-------------------------------DRV WORK STATUS--------------------------------------------------------------
       StartSuc0       StartSuc1         LinkInt   StartErr0  NodeIdErr0   StartErr1  NodeIdErr1      BusErr
             924               0             923           0           0           0           0           0

-------------------------------DRV NODE QUEUE---------------------------------------------------------------
   FreeNum   WaitNum  Busy00  Busy01    Sel0  Busy10  Busy11    Sel1    Proc
         8         0       0       1       1       0       0       0       1

-------------------------------INT WORK STATUS--------------------------------------------------------------
     CntPerSec  MaxCntPerSec        CostTm    MostCostTm  CostTmPerSec MCostTmPerSec
             0             0             0             0             0             0
```

dimerr, 2022-11-17:
https://github.com/OpenIPC/firmware/blob/master/general/package/hisilicon-osdrv-hi3516ev200/files/sensor/config/imx335_i2c_4M.ini#L11 change to 1|0|-1|-1|-1|-1|-1|-1|

gnif, 2022-11-17:
Done, still no output but the output for magestic seem to have changed
```
16:38:15  <       majestic> [    main] main@125                      Majestic Lite for HiSilicon, version HEAD+710e284, built on 2022-11-17
16:38:15  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
16:38:15  <       majestic> [watchdog] watchdog_start@101            Detected Watchdog
16:38:15  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
16:38:16  <       majestic> [     sdk] try_sensor_config@3234        SENSOR=imx335
16:38:16  <       majestic> [     sdk] find_sensor_config@3197       matched sensor config: imx335_i2c_4M.ini
16:38:16  <       majestic> [     sdk] start_sdk@403                 Using /etc/sensors/imx335_i2c_4M.ini sensor configuration file
16:38:16  <       majestic> [     sdk] start_sdk@412                 App was built with MPP version: Hi3516EV200_MPP_V1.0.1.2.B030 Release
16:38:16  <       majestic> [     sdk] start_sdk@415                 Current MPP version: HI_VERSION=Hi3516EV200_MPP_V1.0.1.2 B030 Release
16:38:16  <      hi_logger> [    vpss] vpss_get_chnframe_check_w@401 vpss(0,0) user get frame time out, s32MilliSec=10000
16:38:16  <      hi_logger> [    vpss] vpss_get_chnframe_check_w@401 vpss(0,0) user get frame time out, s32MilliSec=10000
16:38:16  <      hi_logger> [    vpss] vpss_get_chnframe_check_w@401 vpss(0,0) user get frame time out, s32MilliSec=10000
16:38:16  <      hi_logger> [    vpss] vpss_get_chnframe_check_w@401 vpss(0,0) user get frame time out, s32MilliSec=10000
16:38:16  <      hi_logger> [      vi] VI_WaitCaptureFrameStop@2280  Wait Pipe 0 Capture stop timeout 120ms!!!
16:38:16  <       majestic> [     sdk] start_sdk@436                 sensor stSnsImx335Obj
16:38:16  <       majestic> [     sdk] start_sdk@441                   input_mode MIPI, WDR NONE
16:38:16  <       majestic> [     sdk] start_sdk@448                   dev [2592x1520]@200x20 25fps, RGGB
16:38:16  <       majestic> [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx335.so
16:38:16  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem before allocation: 98272KB
16:38:16  <       majestic> [     sdk] dump_vb_configuration@2151    VB configuration:
16:38:16  <       majestic> [     sdk] dump_vb_configuration@2161      [0]: 5909760 x 5
16:38:16  <       majestic> [     sdk] dump_vb_configuration@2161      [1]: 4924800 x 1
16:38:16  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem after allocation: 64592KB
16:38:16  <       majestic> [     sdk] init_sensor@2347              Sensor driver loaded
16:38:16  <       majestic> [     sdk] init_vi@2765                  SDK is in 'VI_OFFLINE_VPSS_ONLINE' mode
16:38:16  <       majestic> [    puts]                               linear mode
16:38:16  <       majestic> [    puts]                               binit false IMX335 i2c init
16:38:16  <       majestic> [    puts]                               -------Sony IMX335_init_5M_2592x1944_12bit_linear30 Initial OK!-------
16:38:16  <     iq_monitor> [     sdk] monitor_thread@332            Loading IQ profile /etc/sensors/iq/default.ini
16:38:16  <       majestic> [     sdk] log_venc_chn@1245             H.265 2592x1520 25fps 4096Kbit  25 GOP
16:38:16  <       majestic> [     sdk] create_vpss_chn@992             new venc: 0   vpss_grp: 0,   vpss_chn: 0
16:38:17  <      hi_logger> [    venc] MkpVencCheckChnCreated@97     venc check create:venc 0 chn not create!
16:38:17  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:38:17  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:38:17  <       majestic> [     hal] show_blocks@686                 allocated Ct x 1=40, Str x 1=2896, PInfo x 2=616, Rcn x 2=11752
16:38:17  <       majestic> [     sdk] start_video@1901                venc took 15304KB of MMZ [2885]
16:38:17  <       majestic> [image_tu] start_image_params_tuning@63  Image tuning task started
16:38:17  <       majestic> [     sdk] start_sdk@548                 HiSilicon SDK started
16:38:17  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem finally: 42988KB
16:38:17  <       majestic> [   httpd] new_http_server@370           HTTP server started on :::80
16:38:17  <      hi_logger> [      vi] VI_DRV_CapIsWrongInt@7785     Pipe:0 err intterupt status:0x1a !! EOF:2 BUFOVF:0 DATATHROW:8 .
16:38:17  <       majestic> [    rtsp] rtsp_init@31                  RTSP server started on port 554
16:38:17  <       majestic> [   netip] netip_start@2051              NETIP server started on port 34567
16:38:17  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:38:17  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:38:17  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:38:17  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:38:18  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:38:18  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
[... cut ...]
16:38:19  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
16:38:19  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
16:38:20  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:38:21  <      hi_logger> [      vi] VI_DRV_CapIsWrongInt@7785     Pipe:0 err intterupt status:0x1a !! EOF:2 BUFOVF:0 DATATHROW:8 .
16:38:21  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
```

dimerr, 2022-11-17:
[libsns_imx335_2l.zip](https://github.com/OpenIPC/firmware/files/10033696/libsns_imx335_2l.zip)

gnif, 2022-11-17:
`/rom` is read only and loading this with `LD_PRELOAD` produces the following output, and the web interface is broken.

```
root@openipc-hi3516ev300:~# LD_PRELOAD=/root/libsns_imx335_2l.so majestic
16:50:04  <       majestic> [    main] main@125                      Majestic Lite for HiSilicon, version HEAD+710e284, built on 2022-11-17
16:50:04  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
16:50:04  <       majestic> [watchdog] watchdog_start@101            Detected Watchdog
16:50:04  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
16:50:04  <       majestic> [     sdk] try_sensor_config@3234        SENSOR=imx335
16:50:04  <       majestic> [     sdk] find_sensor_config@3197       matched sensor config: imx335_i2c_4M.ini
16:50:04  <       majestic> [     sdk] start_sdk@403                 Using /etc/sensors/imx335_i2c_4M.ini sensor configuration file
16:50:04  <       majestic> [     sdk] start_sdk@412                 App was built with MPP version: Hi3516EV200_MPP_V1.0.1.2.B030 Release
16:50:04  <       majestic> [     sdk] start_sdk@415                 Current MPP version: HI_VERSION=Hi3516EV200_MPP_V1.0.1.2 B030 Release
16:50:04  <       majestic> [     sdk] start_sdk@436                 sensor stSnsImx335Obj
16:50:04  <       majestic> [     sdk] start_sdk@441                   input_mode MIPI, WDR NONE
16:50:04  <       majestic> [     sdk] start_sdk@448                   dev [2592x1520]@200x20 25fps, RGGB
16:50:04  <       majestic> [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx335.so
16:50:04  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem before allocation: 98272KB
16:50:04  <       majestic> [     sdk] dump_vb_configuration@2151    VB configuration:
16:50:04  <       majestic> [     sdk] dump_vb_configuration@2161      [0]: 5909760 x 5
16:50:04  <       majestic> [     sdk] dump_vb_configuration@2161      [1]: 4924800 x 1
16:50:04  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem after allocation: 64592KB
16:50:04  <       majestic> [     sdk] init_sensor@2347              Sensor driver loaded
16:50:04  <       majestic> [     sdk] init_vi@2765                  SDK is in 'VI_OFFLINE_VPSS_ONLINE' mode
16:50:04  <       majestic> [    puts]                               linear mode
16:50:04  <       majestic> [    puts]                               binit false IMX335 i2c init
16:50:04  <       majestic> [    puts]                               -------Sony IMX335_init_5M_2592x1944_12bit_linear30 Initial OK!-------
16:50:04  <     iq_monitor> [     sdk] monitor_thread@332            Loading IQ profile /etc/sensors/iq/default.ini
16:50:04  <       majestic> [     sdk] log_venc_chn@1245             H.265 2592x1520 25fps 4096Kbit  25 GOP
16:50:04  <       majestic> [     sdk] create_vpss_chn@992             new venc: 0   vpss_grp: 0,   vpss_chn: 0
16:50:04  <       majestic> [     hal] show_blocks@686                 allocated Ct x 1=40, Str x 1=2896, PInfo x 2=616, Rcn x 2=11752
16:50:04  <       majestic> [     sdk] start_video@1901                venc took 15304KB of MMZ [2885]
16:50:04  <       majestic> [image_tu] start_image_params_tuning@63  Image tuning task started
16:50:04  <       majestic> [     sdk] start_sdk@548                 HiSilicon SDK started
16:50:04  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem finally: 42988KB
16:50:04  <       majestic> [   httpd] new_http_server@370           HTTP server started on :::80
16:50:04  <       majestic> [    rtsp] rtsp_init@31                  RTSP server started on port 554
16:50:04  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:50:04  <       majestic> [   netip] netip_start@2051              NETIP server started on port 34567
16:50:04  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:50:05  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:50:05  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:50:05  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:50:05  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:50:05  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:50:05  <      hi_logger> [      vi] VI_DRV_CapIsWrongInt@7785     Pipe:0 err intterupt status:0x1a !! EOF:2 BUFOVF:0 DATATHROW:8 .
16:50:05  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:50:05  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:50:06  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
16:50:09  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
16:50:09  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
```

Seems you are linking against a different version, loading the web interface generates the following errors:
```Error relocating /root/libsns_imx335_2l.so: HI_MPI_ISP_SensorUnRegCallBack: symbol not found
Error relocating /root/libsns_imx335_2l.so: HI_MPI_AE_SensorRegCallBack: symbol not found
Error relocating /root/libsns_imx335_2l.so: HI_MPI_AE_SensorUnRegCallBack: symbol not found
Error relocating /root/libsns_imx335_2l.so: HI_MPI_AWB_SensorUnRegCallBack: symbol not found
Error relocating /root/libsns_imx335_2l.so: HI_MPI_ISP_SensorRegCallBack: symbol not found
Error relocating /root/libsns_imx335_2l.so: HI_MPI_AWB_SensorRegCallBack: symbol not found
```

gnif, 2022-11-17:
Switching DllFile to the `_2l.so` variant in the config also has no effect.

gnif, 2022-11-17:
> Well, ofc specify new DllFile path, dont preload it or something
Yeah, sorry, it's 4AM here, was a brain fart :smile: 

> also change bitness from 12 to 10 in cfg

It was already 10, here is how I have the config now
```
[sensor]
Sensor_type=stSnsImx335Obj
Mode=WDR_MODE_NONE
DllFile=/root/libsns_imx335_2l.so

[mode]
input_mode=INPUT_MODE_MIPI
raw_bitness=10

[mipi]
lane_id = 1|0|-1|-1|-1|-1|-1|-1|          ;lane_id: -1 - disable
```

And the result:
```
root@openipc-hi3516ev300:~# majestic 
17:04:08  <       majestic> [    main] main@125                      Majestic Lite for HiSilicon, version HEAD+710e284, built on 2022-11-17
17:04:08  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
17:04:08  <       majestic> [watchdog] watchdog_start@101            Detected Watchdog
17:04:08  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
17:04:08  <       majestic> [     sdk] try_sensor_config@3234        SENSOR=imx335
17:04:08  <       majestic> [     sdk] find_sensor_config@3197       matched sensor config: imx335_i2c_4M.ini
17:04:08  <       majestic> [     sdk] start_sdk@403                 Using /etc/sensors/imx335_i2c_4M.ini sensor configuration file
17:04:08  <       majestic> [     sdk] start_sdk@412                 App was built with MPP version: Hi3516EV200_MPP_V1.0.1.2.B030 Release
17:04:08  <       majestic> [     sdk] start_sdk@415                 Current MPP version: HI_VERSION=Hi3516EV200_MPP_V1.0.1.2 B030 Release
17:04:08  <       majestic> [     sdk] start_sdk@436                 sensor stSnsImx335Obj
17:04:08  <       majestic> [     sdk] start_sdk@441                   input_mode MIPI, WDR NONE
17:04:08  <       majestic> [     sdk] start_sdk@448                   dev [2592x1520]@200x20 25fps, RGGB
17:04:08  <       majestic> [  sensor] try_to_load@19                trying to load /root/libsns_imx335_2l.so
17:04:08  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem before allocation: 98272KB
17:04:08  <       majestic> [     sdk] dump_vb_configuration@2151    VB configuration:
17:04:08  <       majestic> [     sdk] dump_vb_configuration@2161      [0]: 5909760 x 5
17:04:08  <       majestic> [     sdk] dump_vb_configuration@2161      [1]: 4924800 x 1
17:04:08  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem after allocation: 64592KB
17:04:08  <       majestic> [     sdk] init_sensor@2347              Sensor driver loaded
17:04:08  <       majestic> [     sdk] init_vi@2765                  SDK is in 'VI_OFFLINE_VPSS_ONLINE' mode
17:04:08  <       majestic> [    puts]                               linear mode
17:04:08  <       majestic> [    puts]                               binit false IMX335 i2c init
17:04:09  <       majestic> [    puts]                               -------Sony IMX335_init_5M_2592x1944_12bit_linear30 Initial OK!-------
17:04:09  <     iq_monitor> [     sdk] monitor_thread@332            Loading IQ profile /etc/sensors/iq/default.ini
17:04:09  <       majestic> [     sdk] log_venc_chn@1245             H.265 2592x1520 25fps 4096Kbit  25 GOP
17:04:09  <       majestic> [     sdk] create_vpss_chn@992             new venc: 0   vpss_grp: 0,   vpss_chn: 0
17:04:09  <       majestic> [     hal] show_blocks@686                 allocated Ct x 1=40, Str x 1=2896, PInfo x 2=616, Rcn x 2=11752
17:04:09  <       majestic> [     sdk] start_video@1901                venc took 15304KB of MMZ [2885]
17:04:09  <       majestic> [image_tu] start_image_params_tuning@63  Image tuning task started
17:04:09  <       majestic> [     sdk] start_sdk@548                 HiSilicon SDK started
17:04:09  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem finally: 42988KB
17:04:09  <       majestic> [   httpd] new_http_server@370           HTTP server started on :::80
17:04:09  <       majestic> [    rtsp] rtsp_init@31                  RTSP server started on port 554
17:04:09  <       majestic> [   netip] netip_start@2051              NETIP server started on port 34567
17:04:09  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
17:04:09  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
17:04:09  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
17:04:09  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
17:04:09  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
17:04:09  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
17:04:10  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
17:04:10  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
17:04:10  <      hi_logger> [      vi] VI_DRV_CapIsWrongInt@7785     Pipe:0 err intterupt status:0x1a !! EOF:2 BUFOVF:0 DATATHROW:8 .
17:04:11  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
17:04:11  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
17:04:13  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
[... cut ...]
17:04:23  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
^C17:04:24  <       majestic> [     log] printf@268                    Interrupt signal received
17:04:24  <       majestic> [     sdk] stop_chn@1942                 Stopped 0 channel
17:04:24  <      hi_logger> [      vi] VI_WaitCaptureFrameStop@2280  Wait Pipe 0 Capture stop timeout 120ms!!!
17:04:25  <       majestic> [     sdk] stop_sdk@3149                 Stop sdk Ok!
17:04:25  <       majestic> [    main] main@209                      Shutdown main thread
```

dimerr, 2022-11-17:
Please show /proc/umap/vi and mipi_rx from openipc while majestic is running

gnif, 2022-11-17:
```root@openipc-hi3516ev300:~# cat /proc/umap/vi 

[VI] Version: [Hi3516EV200_MPP_V1.0.1.2 B030 Release], Build Time[Oct 18 2019, 18:21:00]

-------------------------------MODULE PARAM ---------------------------------------------------------------------------
      DetectErrFrame        DropErrFrame            VbSource
                   0                   0              Common

-------------------------------VI MODE -------------------------------------------------------------
  Pipe0Mode   Pipe1Mode
    offline     offline
-------------------------------VPSS MODE -------------------------------------------------------------
  Pipe0Mode   Pipe1Mode
     online     offline

-------------------------------VI DEV ATTR1----------------------------------------------------------------------------
  DevID   DevEn  BindPipe     Width    Height               IntfM     WkM     ComMsk0     ComMsk1   ScanM
      0       Y         Y      2592      1520                MIPI    1Mux    fff00000           0       P

-------------------------------VI DEV ATTR2----------------------------------------------------------------------------
  DevID   AD0   AD1   AD2   AD3     Seq  DataType   DataRev    BasW    BasH   HReph   VReph   WDRMode  CacheLine  DataRate
      0    -1    -1    -1    -1     N/A       RGB         N    2592    1520    NONE    NONE      None       1520        X1

-------------------------------VI BIND ATTR----------------------------------------------------------------------------
   DevID PipeNum              PipeId
       0       1                   0

-------------------------------VI DEV TIMING ATTR----------------------------------------------------------------------
  DevID DevTimingEn  DevFrmRate  DevWidth   DevHeight

-------------------------------VI PIPE ATTR1---------------------------------------------------------------------------
  PipeID  BypassMode YuvSkip IspBypass     Width    Height    PixFmt  BitWidth    NrEn SharpenEn  CompressMode
       0  BypassNone       N         N      2592      1520     RAW10        10       N         N          None

-------------------------------VI PIPE ATTR2---------------------------------------------------------------------------
  DiscProPic    SrcFRate    DstFRate FrameSource  RepeatMode   VCNum           IntType EarlyLine  VbPoolId
           N          -1          -1         DEV        NONE       0         EARLY_END       128        -1

-------------------------------VI INTERRUPT ONEBUF INFO----------------------------------------------------------------
      ProcCostTime    CapturedCostTime LowerLine UpperLine WdrExposureInterval

-------------------------------VI PIPE CROP ATTR-----------------------------------------------------------------------
  PipeID CropEn    CoorX   CoorY   Width  Height

-------------------------------VI PIPE USER PIC ATTR-------------------------------------------------------------------
  PipeID  Enable   ChnID    Mode BgColor   PicID   Width  Height  Stride  PixFmt  PoolID         PhyAddr

-------------------------------VI PIPE DUMP ATTR-----------------------------------------------------------------------
  PipeID    Enable     Depth  DumpType

-------------------------------VI CHN ATTR1----------------------------------------------------------------------------
  PipeID   ChnID   Width    Height    Mirror    Flip    SrcFRate    DstFRate    PixFmt      VideoFmt  DynamicRange

-------------------------------VI CHN ATTR2----------------------------------------------------------------------------
  CompressMode     Depth     Align  VbPoolId

-------------------------------VI EXTCHN ATTR1-------------------------------------------------------------------------
  PipeID   ChnID  Source  SrcChn   Width    Height    SrcFRate    DstFRate    PixFmt  DynamicRange  CompressMode     Depth

-------------------------------VI EXTCHN ATTR2-------------------------------------------------------------------------
  Align  VbPoolId

-------------------------------VI CHN CROP INFO------------------------------------------------------------------------
  PipeID   ChnID  CropEn  CoorType   CoorX   CoorY   Width  Height   TrimX   TrimY TrimWid TrimHgt

-------------------------------VI CHN ROTATION INFO--------------------------------------------------------------------
  PipeID   ChnID    Rotation

-------------------------------VI CHN LDCV3 INFO-------------------------------------------------------------------------

[... 70 lines cut ...]

-----PHY CIL ERR INT INFO---------------------------------------------
   PhyId  Clk2TmOut  ClkTmOut  Lane0TmOut  Lane1TmOut  Lane2TmOut  Lane3TmOut  Clk2Esc  ClkEsc  Lane0Esc  Lane1Esc  Lane2Esc  Lane3Esc
       0          0         0           0           0           0           0        0       0         0         0         0         0

-----MIPI ERROR INT INFO 1-----------------------------------------------------------
   Devno  Ecc2  Vc0CRC  Vc1CRC  Vc2CRC  Vc3CRC  Vc0EccCorrct  Vc1EccCorrct  Vc2EccCorrct  Vc3EccCorrct
       0324821    4044    1962    2125    2223        102935         41598         46432         49191

-----MIPI ERROR INT INFO 2-----------------------------------------------------------
   Devno  Vc0Dt  Vc1Dt  Vc2Dt  Vc3Dt  Vc0FrmCrc  Vc1FrmCrc  Vc2FrmCrc  Vc3FrmCrc
       0  13352   9254  11037  16907          0          0          0          0

-----MIPI ERROR INT INFO 3-----------------------------------------------------------
[... cut ...]
       0        398        507        951       1406         457         364         420         521

-----MIPI ERROR INT INFO 4-----------------------------------------------------------
   Devno  DataFifoRdErr  CmdFifoRdErr  CmdFifoWrErr  DataFifoWrErr
       0             41          3119            50              0

-----ALIGN ERROR INT INFO--------------------------------------
   Devno  FIFO_FullErr  Lane0Err  Lane1Err  Lane2Err  Lane3Err
       0        345057    179381    175523         0         0
```

dimerr, 2022-11-17:
[libsns_imx335_2l.zip](https://github.com/OpenIPC/firmware/files/10034216/libsns_imx335_2l.zip)

gnif, 2022-11-17:
```
root@openipc-hi3516ev300:~# majestic 
17:45:37  <       majestic> [    main] main@125                      Majestic Lite for HiSilicon, version HEAD+710e284, built on 2022-11-17
17:45:37  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
17:45:37  <       majestic> [watchdog] watchdog_start@101            Detected Watchdog
17:45:37  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
17:45:37  <       majestic> [     sdk] try_sensor_config@3234        SENSOR=imx335
17:45:37  <       majestic> [     sdk] find_sensor_config@3197       matched sensor config: imx335_i2c_4M.ini
17:45:37  <       majestic> [     sdk] start_sdk@403                 Using /etc/sensors/imx335_i2c_4M.ini sensor configuration file
17:45:37  <       majestic> [     sdk] start_sdk@412                 App was built with MPP version: Hi3516EV200_MPP_V1.0.1.2.B030 Release
17:45:37  <       majestic> [     sdk] start_sdk@415                 Current MPP version: HI_VERSION=Hi3516EV200_MPP_V1.0.1.2 B030 Release
17:45:37  <       majestic> [     sdk] start_sdk@436                 sensor stSnsImx335Obj
17:45:37  <       majestic> [     sdk] start_sdk@441                   input_mode MIPI, WDR NONE
17:45:37  <       majestic> [     sdk] start_sdk@448                   dev [2592x1520]@200x20 25fps, RGGB
17:45:37  <       majestic> [  sensor] try_to_load@19                trying to load /root/libsns_imx335_2l.so
17:45:37  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem before allocation: 98272KB
17:45:37  <       majestic> [     sdk] dump_vb_configuration@2151    VB configuration:
17:45:37  <       majestic> [     sdk] dump_vb_configuration@2161      [0]: 5909760 x 5
17:45:37  <       majestic> [     sdk] dump_vb_configuration@2161      [1]: 4924800 x 1
17:45:37  <       majestic> [     sdk] dump_vb_configuration@2161      [2]: 608256 x 1
17:45:37  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem after allocation: 63992KB
17:45:37  <       majestic> [     sdk] init_sensor@2347              Sensor driver loaded
17:45:37  <       majestic> [     sdk] init_vi@2765                  SDK is in 'VI_OFFLINE_VPSS_ONLINE' mode
17:45:37  <       majestic> [    puts]                               linear mode
17:45:37  <       majestic> [    puts]                               binit false IMX335 i2c init
17:45:37  <       majestic> [    puts]                               -------Sony IMX335_2l_init_5M_2592x1944_10bit_linear Initial OK!-------
17:45:37  <     iq_monitor> [     sdk] monitor_thread@332            Loading IQ profile /etc/sensors/iq/default.ini
17:45:37  <       majestic> [     sdk] log_venc_chn@1245             H.265 2592x1520 25fps 4096Kbit  25 GOP
17:45:37  <       majestic> [     sdk] create_vpss_chn@992             new venc: 0   vpss_grp: 0,   vpss_chn: 0
17:45:37  <       majestic> [     hal] show_blocks@686                 allocated Ct x 1=40, Str x 1=2896, PInfo x 2=616, Rcn x 2=11752
17:45:37  <       majestic> [     sdk] start_video@1901                venc took 15304KB of MMZ [2885]
17:45:37  <       majestic> [     sdk] log_venc_chn@1245             H.265 704x576 15fps 1024Kbit  15 GOP
17:45:37  <       majestic> [     sdk] create_vpss_chn@992             new venc: 1   vpss_grp: 0,   vpss_chn: 1
17:45:37  <       majestic> [     hal] show_blocks@686                 allocated Ct x 1=12, PInfo x 2=64, Str x 1=308, Rcn x 2=1200
17:45:37  <       majestic> [     sdk] start_video@1901                venc took 1584KB of MMZ [297]
17:45:37  <       majestic> [image_tu] start_image_params_tuning@63  Image tuning task started
17:45:37  <       majestic> [     sdk] start_sdk@548                 HiSilicon SDK started
17:45:37  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem finally: 40804KB
17:45:37  <       majestic> [   httpd] new_http_server@370           HTTP server started on :::80
17:45:37  <       majestic> [    rtsp] rtsp_init@31                  RTSP server started on port 554
17:45:37  <       majestic> [   netip] netip_start@2051              NETIP server started on port 34567
17:45:37  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
17:45:39  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
17:45:39  <      hi_logger> [      vi] VI_DRV_CapIsWrongInt@7785     Pipe:0 err intterupt status:0x1a !! EOF:2 BUFOVF:0 DATATHROW:8 .
17:45:39  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
17:45:39  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
17:45:39  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 1
17:45:39  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
17:45:39  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
17:45:39  <      hi_logger> [      vi] VI_DRV_CapIsWrongInt@7785     Pipe:0 err intterupt status:0x1a !! EOF:2 BUFOVF:0 DATATHROW:8 .
17:45:41  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
17:45:41  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
[... cut ...]
17:45:54  <       majestic> [     sdk] stop_chn@1942                 Stopped 1 channel
17:45:54  <       majestic> [     sdk] stop_chn@1942                 Stopped 0 channel
17:45:54  <      hi_logger> [    vpss] vpss_get_chnframe_check_w@401 vpss(0,0) user get frame time out, s32MilliSec=10000
17:45:54  <      hi_logger> [      vi] VI_WaitCaptureFrameStop@2280  Wait Pipe 0 Capture stop timeout 120ms!!!
17:45:54  <       majestic> [     sdk] stop_sdk@3149                 Stop sdk Ok!
17:45:54  <       majestic> [    main] main@209                      Shutdown main thread
```

dimerr, 2022-11-17:
[libsns_imx335_2l.zip](https://github.com/OpenIPC/firmware/files/10034309/libsns_imx335_2l.zip)

gnif, 2022-11-17:
That one works somewhat, the output is all garbled, but it actually does something now!
No errors reported in majestic output 

[image]

dimerr, 2022-11-17:
[libsns_imx335_2l.zip](https://github.com/OpenIPC/firmware/files/10034366/libsns_imx335_2l.zip)
try powering off board for a while after replacing with this file

gnif, 2022-11-17:
Still the same output

dimerr, 2022-11-17:
[libsns_imx335_2l.zip](https://github.com/OpenIPC/firmware/files/10034520/libsns_imx335_2l.zip)

gnif, 2022-11-17:
[image]

dimerr, 2022-11-17:
Run on stock fw:
ipctool i2cdump 0x34 0x3000 0x3B00 --script
then do the same on openipc while majestic is running.

Also you could try to kill main app ("APP") and run it via ipctool
ipctool trace /opt/topsee/APP

and save the output

gnif, 2022-11-17:
I tried earlier to run `ipctool` on the stock firmware and it crashes with a bus error.

dimerr, 2022-11-17:
try 5MP mode, 2592x1944 in cfg

gnif, 2022-11-17:
Changed `DevRect_h` to `1944`, no improvement.

dimerr, 2022-11-17:
https://github.com/OpenIPC/firmware/blob/master/general/package/hisilicon-osdrv-hi3516ev200/files/sensor/config/5M_imx335.ini use this cfg for reference, there are few more differences

gnif, 2022-11-17:
[image]
Solid pink edge down the bottom now also

dimerr, 2022-11-17:
can you provide ssh access to cam?

gnif, 2022-11-17:
Sure, please provide a public SSH key and I will set it up. How can I send you IP/Port details privately?

gnif, 2022-11-18:
Not sure if this helps any but I pulled the raw YUV420 frame to give a clearer indication of the output, avoiding JPEG compression artifacts.

[image]

gnif, 2022-11-18:
I found that if I poke these values into these registers I can get a somewhat sane output...

ipctool i2cset 0x34 0x314c 0xaa
ipctool i2cset 0x34 0x315a 0x02

[image]

dimerr, 2022-11-18:
thats sensor clock

gnif, 2022-11-18:
I have no datasheets so I am not sure at all what I am poking at.

dimerr, 2022-11-18:
Add Clock=74,25

refer to https://github.com/OpenIPC/firmware/blob/master/general/package/hisilicon-osdrv-hi3516ev200/files/sensor/config/sc2315e_i2c_1080p.ini#L8

gnif, 2022-11-18:
This causes lots of errors and no output:

```
20:08:53  <       majestic> [    puts]                               -------Sony IMX335_2l_init_5M_2592x1944_10bit_linear Initial OK!-------
20:08:53  <     iq_monitor> [     sdk] monitor_thread@332            Loading IQ profile /etc/sensors/iq/default.ini
20:08:53  <       majestic> [     sdk] log_venc_chn@1245             H.265 2592x1944 15fps 4096Kbit  15 GOP
20:08:53  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
20:08:53  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
20:08:53  <       majestic> [     sdk] create_vpss_chn@992             new venc: 0   vpss_grp: 0,   vpss_chn: 0
20:08:53  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
20:08:53  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
20:08:53  <      hi_logger> [      vi] VI_COMM_CapPrepareFrame@2894  Pipe 0 lost frame because of wrong int order!
20:08:53  <       majestic> [     hal] show_blocks@686                 allocated PInfo x 2=784, Ct x 1=40, Str x 1=3700, Rcn x 2=15088
```

Is there any other way I can dump the registers from the factory firmware?

gnif, 2022-11-18:
I managed to get a register dump using `i2c_read` on the factory firmware while `APP` is running:

```
0x3000: 0x0
0x3001: 0x0
0x3002: 0x0
0x3003: 0x0
0x3004: 0x0
0x3005: 0x0
0x3006: 0x0
0x3007: 0x0
0x3008: 0x0
0x3009: 0x0
0x300a: 0x0
0x300b: 0x0
0x300c: 0x5b
0x300d: 0x40
0x300e: 0x0
0x300f: 0x0
0x3010: 0x6
0x3011: 0x0
0x3012: 0x0
0x3013: 0x1
0x3014: 0x2
0x3015: 0x0
0x3016: 0x0
0x3017: 0x0
0x3018: 0x0
0x3019: 0x0
0x301a: 0x0
0x301b: 0x0
0x301c: 0x8
0x301d: 0x0
0x301e: 0x20
0x301f: 0x0
0x3020: 0x0
0x3021: 0x0
0x3022: 0x98
0x3023: 0xa
0x3024: 0x38
0x3025: 0xa
0x3026: 0x38
0x3027: 0xa
0x3028: 0x38
0x3029: 0xa
0x302a: 0x0
0x302b: 0x0
0x302c: 0x30
0x302d: 0x0
0x302e: 0x38
0x302f: 0xa
0x3030: 0xf2
0x3031: 0x2b
0x3032: 0x0
0x3033: 0x0
0x3034: 0x26
0x3035: 0x2
0x3036: 0x0
0x3037: 0x0
0x3038: 0x0
0x3039: 0x2
0x303a: 0x8
0x303b: 0x0
0x303c: 0xf0
0x303d: 0xff
0x303e: 0x0
0x303f: 0x0
0x3040: 0xff
0x3041: 0xff
0x3042: 0x3
[... 2726 lines cut ...]
0x3ae9: 0x0
0x3aea: 0x0
0x3aeb: 0x0
0x3aec: 0x0
0x3aed: 0x0
0x3aee: 0x0
0x3aef: 0x0
0x3af0: 0x0
0x3af1: 0x0
0x3af2: 0x0
0x3af3: 0x0
0x3af4: 0x0
0x3af5: 0x0
0x3af6: 0x0
0x3af7: 0x0
0x3af8: 0x0
0x3af9: 0x0
0x3afa: 0x0
0x3afb: 0x0
0x3afc: 0x0
0x3afd: 0x0
0x3afe: 0x0
0x3aff: 0x0
0x3b00: 0x0
```

gnif, 2022-11-18:
I can confirm that if I just set these registers the picture is now perfect

gnif, 2022-11-18:
The following three changes fix the output completely:
```
ipctool i2cset 0x34 0x314c 0x80
ipctool i2cset 0x34 0x315a 0x2
ipctool i2cset 0x34 0x319e 0x1
```

dimerr, 2022-11-18:
run ipctool now, what clock speed it reports?

gnif, 2022-11-18:
It is reporting 24MHz, but I did try this in the `ini` which had no effect.

dimerr, 2022-11-18:
[libsns_imx335_2l.zip](https://github.com/OpenIPC/firmware/files/10041254/libsns_imx335_2l.zip)

gnif, 2022-11-18:
Beautiful! Works perfect!
Thank you so much for your time and help debugging this.

gnif, 2022-11-19:
Sorry I am not sure why but today when I checked again ipctool is now reporting 37.125MHz

widgetii, 2022-11-19:
New driver's source code is placed [here](https://github.com/OpenIPC/openhisilicon/tree/main/libraries/sensor/hi3516ev200/sony_imx335_2L). Feel free to modify init sequence to test any problem you still have

gnif, 2022-11-20:
Thanks but this is now working correctly, will close as fixed.

closed 2022-11-20 by gnif