OpenIPC/firmware #573: BUF_MALLOC_ERROR if camera started up in the dark

gnif, 2022-11-20:
This is a Topsee HI3516EV300_IMX335 (same as #563). I have three of these devices and they are exhibit the same issue.

If the device is powered off and on again in the dark (outdoor camera, night), majestic fails with the following:

```
22:01:03  <       majestic> [     log] printf@268                    Error buf: BUF_MALLOC_ERROR     put(...)    /home/runner/work/firmware/firmware/output/build/majestic-master/majestic/src/file_format/mp4/bitbuf.c:70
22:01:03  <       majestic> [     log] printf@268                    Error buf: BUF_MALLOC_ERROR     write_mdat(...)    /home/runner/work/firmware/firmware/output/build/majestic-master/majestic/src/file_format/mp4/moof.c:44
22:01:03  <       majestic> [     log] printf@268                    Error buf: BUF_MALLOC_ERROR     flush_video(...)    /home/runner/work/firmware/firmware/output/build/majestic-master/majestic/src/file_format/mp4/mp4.c:339
22:01:06  <  thread-pool-1> [image_co] set_image_luminance@58        Autoadjust image param cscAttrLumaField from 50 to 70
22:01:08  <       majestic> [     log] printf@268                    Error buf: BUF_MALLOC_ERROR     put(...)    /home/runner/work/firmware/firmware/output/build/majestic-master/majestic/src/file_format/mp4/bitbuf.c:70
22:01:08  <       majestic> [     log] printf@268                    Error buf: BUF_MALLOC_ERROR     write_mdat(...)    /home/runner/work/firmware/firmware/output/build/majestic-master/majestic/src/file_format/mp4/moof.c:44
22:01:08  <       majestic> [     log] printf@268                    Error buf: BUF_MALLOC_ERROR     flush_video(...)    /home/runner/work/firmware/firmware/output/build/majestic-master/majestic/src/file_format/mp4/mp4.c:339
22:01:08  <       majestic> [     log] printf@268                    Error buf: BUF_MALLOC_ERROR     put(...)    /home/runner/work/firmware/firmware/output/build/majestic-master/majestic/src/file_format/mp4/bitbuf.c:70
22:01:08  <       majestic> [     log] printf@268                    Error buf: BUF_MALLOC_ERROR     write_mdat(...)    /home/runner/work/firmware/firmware/output/build/majestic-master/majestic/src/file_format/mp4/moof.c:44
22:01:08  <       majestic> [     log] printf@268                    Error buf: BUF_MALLOC_ERROR     flush_video(...)    /home/runner/work/firmware/firmware/output/build/majestic-master/majestic/src/file_format/mp4/mp4.c:339
```

At which point the watchdog kicks in a resets the device. Simply turning on a light and then powering up the camera is enough to get it to start properly.

widgetii, 2022-11-20:
What's output of `free` on this board?

gnif, 2022-11-20:
```
root@cam-front-door:~# free
              total        used        free      shared  buff/cache   available
Mem:         124172      109916        1600         128       12656       11576
Swap:             0           0           0
```

widgetii, 2022-11-20:
Please check camera's behaviour with `/etc/sensors/iq/imx335.ini` iq profile (by default imx307 is used).

Issue can be mitigated disabling hls encoder

gnif, 2022-11-20:
No change using imx335, same issue. Disabling hls does free up some RAM and allow the device to start, however this breaks the `video.mp4` endpoint.

gnif, 2022-11-20:
It is certainly an out of memory error condition, I have changed the output resolution of the main video stream to 2160x1620 which works around the issue. Closing as fixed.

closed 2022-11-20 by gnif

widgetii, 2022-11-20:
The issue is that the video encoder produces enormous bandwidth in low light conditions, causing out of memory problem while in internal buffer, I think this probably can be fixed adjusting pq profile in [PQTools](https://github.com/OpenIPC/wiki/blob/master/en/image-quality-tuning.md#ev300-family)