OpenIPC/firmware #596: Ingenic T20 missing libssl.so.1.0.0 and libcrypto.so.1.0.0

moosehadley, 2022-12-22:
Attempting to run wpa_supplicant on the Ingenic T20 build produces the error:

 `Error loading shared library libssl.so.1.0.0: No such file or directory (needed by /bin/wpa_supplicant)
Error loading shared library libcrypto.so.1.0.0: No such file or directory (needed by /bin/wpa_supplicant)`

Neither library appears to be present in /usr/lib/

widgetii, 2022-12-22:
We don't use OpenSSL in our project, so wpa_supplicant should be linked against MbedTLS

moosehadley, 2022-12-22:
> We don't use OpenSSL in our project, so wpa_supplicant should be linked against MbedTLS

Perhaps this issue should be renamed then.

I'm using the precompiled build and this error occurs, wpa_supplicant will not run.

widgetii, 2022-12-22:
I don't have such hardware, so feel free to find root cause and make a PR

FlyRouter, 2022-12-22:
We need more details on your incident.

The board in the booth has no problem using mbedtls

```
openipc-t20 login: root
Password: 

    /######                                    /######  /#######    /######
   /##__  ##                                  |_  ##_/ | ##__  ##  /##__  ##
  | ##  \ ##   /######    /######   /#######    | ##   | ##  \ ## | ##  \__/
  | ##  | ##  /##__  ##  /##__  ## | ##__  ##   | ##   | #######/ | ##
  | ##  | ## | ##  \ ## | ######## | ##  \ ##   | ##   | ##____/  | ##
  | ##  | ## | ##  | ## | ##_____/ | ##  | ##   | ##   | ##       | ##    ##
  |  ######/ | #######/ |  ####### | ##  | ##  /###### | ##       |  ######/
   \______/  | ##____/   \_______/ |__/  |__/ |______/ |__/        \______/
             | ##
             | ##                              build
             |__/                             master+741151a4, 2022-11-22

 Please help the OpenIPC Project to cover the cost of development and
 long-term maintenance of what we believe is going to become a stable,
 flexible Open IP Network Camera Framework for users worldwide.

 Your contributions could help us to advance the development and keep
 you updated on improvements and new features more regularly.

 Please visit https://openipc.org/sponsor/ to learn more. Thank you.

root@openipc-t20:~# 
root@openipc-t20:~# wpa_supplicant 
Successfully initialized wpa_supplicant
wpa_supplicant v2.9
Copyright (c) 2003-2019, Jouni Malinen <j@w1.fi> and contributors

This software may be distributed under the terms of the BSD license.
See README for more details.

usage:
  wpa_supplicant [-BddhKLqqstvW] [-P<pid file>] [-g<global ctrl>] \
        [-G<group>] \
        -i<ifname> -c<config file> [-C<ctrl>] [-D<driver>] [-p<driver_param>] \
        [-b<br_ifname>] [-e<entropy file>] [-f<debug file>] \
        [-o<override driver>] [-O<override ctrl>] \
        [-N -i<ifname> -c<conf> [-C<ctrl>] [-D<driver>] \
        [-p<driver_param>] [-b<br_ifname>] [-I<config file>] ...]
        
and more..
```

moosehadley, 2022-12-23:
This is my output:

[fulldump.log](https://github.com/OpenIPC/firmware/files/10291782/fulldump.log)

> root@openipc-t20:~# wpa_supplicant
> Error loading shared library libssl.so.1.0.0: No such file or directory (needed by /bin/wpa_supplicant)
> Error loading shared library libcrypto.so.1.0.0: No such file or directory (needed by /bin/wpa_supplicant)
> Error relocating /bin/wpa_supplicant: SSL_CTX_set_client_CA_list: symbol not found
> Error relocating /bin/wpa_supplicant: EVP_DecryptFinal_ex: symbol not found
> Error relocating /bin/wpa_supplicant: ERR_load_ENGINE_strings: symbol not found
> Error relocating /bin/wpa_supplicant: X509_LOOKUP_file: symbol not found
> Error relocating /bin/wpa_supplicant: SSL_state: symbol not found
> Error relocating /bin/wpa_supplicant: SSL_get_error: symbol not found
> Error relocating /bin/wpa_supplicant: __cmsg_nxthdr: symbol not found
> Error relocating /bin/wpa_supplicant: OCSP_RESPONSE_print: symbol not found
> Error relocating /bin/wpa_supplicant: SSL_use_certificate_file: symbol not found
> Error relocating /bin/wpa_supplicant: PKCS12_free: symbol not found
> Error relocating /bin/wpa_supplicant: BN_add: symbol not found
These symbol errors continue for 200 lines.

I am using openipc.t20-nor-lite, I just updated to today's build and the same error occurs.

moosehadley, 2022-12-24:
The image appears to have two copies of wpa_supplicant.

The one in /bin/ produces this error, the one in /usr/sbin/ works correctly.

> root@openipc-t20:~# which wpa_supplicant 
> /bin/wpa_supplicant

> root@openipc-t20:~# /bin/wpa_supplicant
> Error loading shared library libssl.so.1.0.0: No such file or directory (needed by /bin/wpa_supplicant)
> Error loading shared library libcrypto.so.1.0.0: No such file or directory (needed by /bin/wpa_supplicant)

> root@openipc-t20:~# /usr/sbin/wpa_supplicant 
> Successfully initialized wpa_supplicant
> wpa_supplicant v2.9

moosehadley, 2022-12-25:
It appears that my camera had some files leftover from the original firmware, this was one of them.

closed 2022-12-25 by moosehadley