OpenIPC/firmware #602: Please build a T20 Ultimate binary

felipecrs, 2023-01-04:
I flashed the lite variant in my camera, and it fails with the logs below. My next step would be to try the ultimate variant, but it's not being built for this SoC. So, can you please add it to the build matrix?

```
U-Boot 2013.07-gc03939f-dirty (Nov 14 2022 - 19:40:07)

Board: ISVP (Ingenic XBurst T20 SoC)
DRAM:  64 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 440k for U-Boot at: 83f90000
Reserving 32832k for malloc() at: 81f80000
Reserving 32 Bytes for Board Info at: 81f7ffe0
Reserving 124 Bytes for Global Data at: 81f7ff64
Reserving 128k for boot params() at: 81f5ff64
Stack Pointer at: 81f5ff48
Now running in RAM - U-Boot at: 83f90000
MMC:   msc: 0
the manufacturer ef
SF: Detected W25Q128

In:    serial
Out:   serial
Err:   serial
Net:   Jz4775-9161
Hit any key to stop autoboot:  0
the manufacturer ef
SF: Detected W25Q128

--->probe spend 4 ms
SF: 3145728 bytes @ 0x50000 Read: OK
--->read spend 456 ms
## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   Linux-3.10.14-t20
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1607541 Bytes = 1.5 MiB
   Load Address: 80010000
   Entry Point:  80366e70
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14 (runner@fv-az572-198) (gcc version 8.4.0 (Buildroot -gff5aa3b6-dirty) ) #2 PREEMPT Mon Dec 26 12:30:51 UTC 2022
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 RESET ERROR PC:801BA6E4
[    0.000000] [<801ba6e4>] __delay+0x4/0x10
[    0.000000] CPU0 revision is: 00d00101 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] CCLK:860MHz L2CLK:430Mhz H0CLK:200MHz H2CLK:200Mhz PCLK:100Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00482000 @ 00010000 (usable)
[    0.000000]  memory: 0003e000 @ 00492000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 02700000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x026fffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x026fffff]
[    0.000000] Primary instruction cache 32kB, 8-way, VIPT, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] pls check processor_id[0x00d00101],sc_jz not support!
[    0.000000] MIPS secondary cache 128kB, 8-way, linesize 32 bytes.
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 9906
[    0.000000] Kernel command line: mem=39M@0x0 ispmem=5M@0x2700000 rmem=20M@0x2c00000 console=ttyS1,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=jz_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[... 82 lines cut ...]
[    0.602515] libphy: jz_mii_bus: probed
[    0.606990] =======>gmacdev = 0x81a45a80<================
[    0.612602] =========>gmacdev->MacBase = 0xb34b0000 DmaBase = 0xb34b1000
[    0.619752] Bus Mode Reg after reset: 0x00020101, cnt=0
[    0.627160] jz_mac jz_mac.0: JZ on-chip Ethernet MAC driver, Version 1.0
[    0.634501] i2c /dev entries driver
[    0.640140] jzmmc_v1.2 jzmmc_v1.2.0: vmmc regulator missing
[    0.646472] jzmmc_v1.2 jzmmc_v1.2.0: register success!
[    0.652504] ipip: IPv4 over IPv4 tunneling driver
[    0.658456] TCP: cubic registered
[    0.663438] NET: Registered protocol family 10
[    0.669418] NET: Registered protocol family 17
[    0.675674] soc_vpu probe success,version:1.0.0-03203fd46d
[    0.681914] input: gpio-keys as /devices/platform/gpio-keys/input/input0
[    0.689797] drivers/rtc/hctosys.c: unable to open rtc device (rtc0)
[    0.699401] List of all partitions:
[    0.703056] 1f00             256 mtdblock0  (driver?)
[    0.708494] 1f01              64 mtdblock1  (driver?)
[    0.713737] 1f02            2048 mtdblock2  (driver?)
[    0.719033] 1f03            5120 mtdblock3  (driver?)
[    0.724278] 1f04            8896 mtdblock4  (driver?)
[    0.729555] No filesystem could mount root, tried:  squashfs
[    0.735467] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
[    0.744134] Rebooting in 20 seconds..Restarting after 4 ms
```

jayfan0, 2023-04-02:
The problem isn't between lite and ultimate. It seems your rootfs didnt properly flash.
Try flashing rootfs again, and post the logs of the process here

felipecrs, 2023-04-02:
Oh, I'm sorry. I fixed that issue already. Thanks.

closed 2023-05-07 by themactep