OpenIPC/firmware #632: Erro rwhen upgrading firmware with web UI

darkfuncat, 2023-02-01:
When I try to upgrade firmware, the process end with error :
`umount: can't unmount /mnt/mmcblk0p1: Resource busy
`
It is busy because of recording.
When I try to stop recording, same error.
Need to stop recording, reboot, upgrade, reenable recording.

themactep, 2023-05-07:
is this still an issue?

darkfuncat, 2023-05-08:
Hello,
I just tried and yes from 2.3.03.21 to 2.2.05.08 (tehre is also a bug with the second digit version)
Here are the logs :
`[root@openipc-hi3518ev300 /tmp]# /usr/sbin/sysupgrade -k -r -x

OpenIPC System Updater v1.0.16

Vendor	hisilicon
SoC	hi3518ev300
Kernel	14:20:19 2023-03-21
RootFS	master+a89791ec, 2023-03-21

Synchronizing time
ntpd: setting time to 2023-05-08 12:01:49.753504 (offset +4135289.616669s)
Mon May  8 12:01:49 CEST 2023

Checking for sysupgrade update...
A new version is available, trying to activate updated script...
Done. Restarting...

OpenIPC System Updater v1.0.17

Vendor	hisilicon
SoC	hi3518ev300
Kernel	14:20:19 2023-03-21
RootFS	master+a89791ec, 2023-03-21

Synchronizing time
ntpd: setting time to 2023-05-08 12:01:51.438579 (offset +4135291.340286s)
Mon May  8 12:01:51 CEST 2023

Checking for sysupgrade update...
Same version. No update required.

Unmounting SD card
/mnt/mmcblk0p1
- checking for /mnt/mmcblk0p1/autoupdate-kernel.img
- checking for /mnt/mmcblk0p1/autoupdate-rootfs.img
- checking for /mnt/mmcblk0p1/autoupdate-uboot.img
umount: can't unmount /mnt/mmcblk0p1: Resource busy

EXIT CODE: 1

--- finished ---`

cronyx, 2023-05-08:
> is this still an issue?

Perhaps the potential problem [here](https://github.com/OpenIPC/firmware/blob/master/general/overlay/usr/sbin/sysupgrade#L350), the check of the sd card is performed before free_resources. If recording is in progress at this time, the card cannot be unmounted.

cronyx, 2023-05-27:
must be fixed in #827

closed 2023-06-05 by cronyx

darkfuncat, 2023-07-07:
Hello;
Cant test since I am stuck with the wifi bug (https://github.com/OpenIPC/firmware/blob/master/ATTENTION.md)
I cannot upgrade to test this, so cannot say if closed or not.