OpenIPC/firmware #638: video rx: unable to decrypt packet #0x47101

zhuwei, 2023-02-10:
drone: WFB-ng version 23.1.36.29031-b984dd7b
gs:WFB-ng version 23.1.16.78051-4886033f

gs: video rx: unable to decrypt packet #0x47101
Has anyone made this mistake?

The hardware is IVG-G2S

cronyx, 2023-02-10:
Possible wrong keys issue. Download gs.key from drone. (/etc/gs.key)

zhuwei, 2023-02-10:
@cronyx Yes, I did. I compared the gs.key file of drone with the gs.key file of gs. They are identical. I have rewritten the firmware several times. It's all this error

cronyx, 2023-02-10:
This is weird, i will try to reproduce the problem...

cronyx, 2023-02-10:
So, I wasn't able to reproduce the issue.

1. Cloned the wfb-ng repo, switched to the stable branch, compiled the wfb_rx wfb_tx binaries.
2. Run a firmware update on the camera and erased all settings.
```
sysupgrade -r -k -n
```
As soon as the camera started, I copied the gs.key:
```
scp root@192.168.0.101:/etc/gs.key .
```
and wrote nearby to wfb_rx then run:
```
./wfb_rx -p 0 -u 5600 -K gs.key -i wlan0
```
packets are successfully decrypted:

```cronyx@cronyx:~/openipc$ ./wfb_rx -p 0 -u 5600 -K gs.key -i wlan0

Setting wlan0 to channel 14
3169880693	PKT	0:0:0:0:0:0
3169881694	PKT	0:0:0:0:0:0
3169882695	PKT	0:0:0:0:0:0
3169883696	PKT	0:0:0:0:0:0
3169884697	PKT	0:0:0:0:0:0
3169884947	SESSION	0:1:8:12
3169885697	ANT	1	331:-46:-45:-42
3169885697	ANT	0	331:-48:-46:-44
3169885697	PKT	332:0:332:1:0:0
3169886697	ANT	1	448:-48:-45:-42
3169886697	ANT	0	448:-48:-45:-44
3169886697	PKT	449:0:449:0:0:0
3169887697	ANT	1	477:-46:-45:-42
3169887697	ANT	0	477:-48:-45:-44
3169887697	PKT	478:0:478:8:0:0
3169888697	ANT	1	417:-48:-45:-42
3169888697	ANT	0	417:-48:-45:-42
3169888697	PKT	418:0:418:2:0:0
3169889697	ANT	1	450:-46:-45:-42
3169889697	ANT	0	450:-48:-43:-42
3169889697	PKT	451:0:451:0:0:0
3169890697	ANT	1	516:-46:-45:-42
3169890697	ANT	0	516:-44:-42:-42
3169890697	PKT	517:0:517:0:0:0
3169891697	ANT	1	406:-46:-45:-44
3169891697	ANT	0	406:-46:-43:-40
```

zhuwei, 2023-02-11:
Can I send the drone.key of gs to drone?

zhuwei, 2023-02-11:
How to view the drone log?

cronyx, 2023-02-11:
> Can I send the drone.key of gs to drone?
On the GS:
cd /etc/
scp drony.key root@cam_ipaddress:/etc/

> How to view the drone log?
logread command

zhuwei, 2023-02-18:
@cronyx thank you very much!

closed 2023-02-18 by zhuwei