OpenIPC/firmware #640: Sensor request for AR0237

sergshadrap, 2023-02-12:
Hi there!
Trying to setup the AR0237 sensor for HI3516vc200 board.
What have I done : took the driver from vc300 board , put in usr\lib\sensors, and than put in /etc/sensors ar0237_i2c_dc_1080p_line.ini
Runned firstboot  and fw_setenv sensor ar0237.
Writes in the boot log :
hisilicon: Get data from environment and set SENSOR as ar0237
xxxx Invalid sensor type ar0237 xxxx
******* Error: There's something wrong, please check! *****
Starting majestic: OK
Is there the way for converting AR0237 for my board? Could you advise - how? Thanks in advance. Serg

FlyRouter, 2023-02-12:
Good afternoon
You can't do that, the drivers are incompatible.

sergshadrap, 2023-02-12:
Thanks for support, so it won't work , the new drivers no chance to be developed?

widgetii, 2023-02-12:
This is open source project, PRs are always welcome

FlyRouter, 2023-02-12:
> Thanks for support, so it won't work , the new drivers no chance to be developed?

You need to study what kind of device and firmware you have.
In some firmware it is possible to extract the sensor driver for use in OpenIPC.

sergshadrap, 2023-02-12:
Yah... - Unfortunately I have no skills in drivers building) Could you point me out on similar driver source code?

No, unfortunately there is no chance to get driver out of firmware, it's heavy crypted. I even can't locate "crypt" section .
The configuration is simple - HI3516vc200 + AR0237 sensor. Some private vendor.
I checked for XM's firmware for similar configuration but there is no trace of something that looks like ar0237 driver...

FlyRouter, 2023-02-12:
You need to remove the dump image from the flash drive by the programmer as well as send a photo of the board on both sides. This is the minimum you can start with.

sergshadrap, 2023-02-12:
Photo of the both board sides and old (original firmware) 
[image]
[image]
[HI316CV200+AR0237_old_firmware.zip](https://github.com/OpenIPC/firmware/files/10716768/HI316CV200%2BAR0237_old_firmware.zip)

FlyRouter, 2023-02-12:
Unfortunately in your firmware the drivers are compiled into the body of the main application.

sergshadrap, 2023-02-12:
Thanks for explanation , unfortunately that board seems to be not for OpenIpc)  . Btw,  I've got several XM's firmware for such configs. I wonder if I could check the drivers there.

dimerr, 2023-02-12:
try with 0230 + init from stock fw

widgetii, 2023-02-13:
You cannot [because of the answer](https://github.com/OpenIPC/firmware/issues/640#issuecomment-1427053008)

sergshadrap, 2023-02-13:
> try with 0230 + init from stock fw
Confused .....
What you suppose me to do with your post? There is no options for choosing ar0230 sensor for HI3516cv200 board common way..
Pocking around AR0237 sensor I've got a clue that it's HiSpi interfaced sensor and definitely it wouldn't work with I2C drivers.. 
 - sorry for my first questions.   Next I've found libsns_ar0230.so on your site in hi3516cv200 package as you advised...  There is any remark about interface ..

FlyRouter, 2023-02-13:
The AR0230 sensor is supported by the system and works fine. You need to specify it in the ENV of the system and change the configuration in the file. Also check the configuration of all your other boards and see if you can find something compatible.

sergshadrap, 2023-02-13:
Thanks . The driver seems to be fine for system:
hisilicon: Loading of kernel modules and initialization of the video system has started
mmz_start: 0x82000000, mmz_size: 96M
hisilicon: Get data from environment and set SENSOR as ar0230
insert audio
==== Your input Sensor type is ar0230 ====
Starting majestic: OK
No any error messages with AR driver , but still no image on preview page. 
Printenv says: 
ARCH=armv5tejl
USER=root
DEBUG=0
SHLVL=1
HOME=/root
ARC=
SENSOR=ar0230
PS1=\u@\h:\w\#
LOGNAME=root
TERM=vt100
PATH=/bin:/sbin:/usr/bin:/usr/sbin:/tmp
SHELL=/bin/sh
PWD=/root
HISTFILE=/tmp/.ash_history
TZ=GMT0
EDITOR=/bin/vi

FlyRouter, 2023-02-13:
Please run as example
```
fw_setenv sensor ar0130
fw_printenv
```

sergshadrap, 2023-02-13:
bootdelay=1
baudrate=115200
ethaddr=00:00:23:34:45:66
gatewayip=192.168.1.1
netmask=255.255.255.0
bootfile="uImage"
baseaddr=0x82000000
uknor8m=mw.b ${baseaddr} ff 1000000; tftpboot ${baseaddr} uImage.${soc} && sf pr                               obe 0; sf erase 0x50000 0x200000; sf write ${baseaddr} 0x50000 ${filesize}
uknor16m=mw.b ${baseaddr} ff 1000000; tftpboot ${baseaddr} uImage.${soc} && sf p                               robe 0; sf erase 0x50000 0x300000; sf write ${baseaddr} 0x50000 ${filesize}
urnor8m=mw.b ${baseaddr} ff 1000000; tftpboot ${baseaddr} rootfs.squashfs.${soc}                                && sf probe 0; sf erase 0x250000 0x500000; sf write ${baseaddr} 0x250000 ${file                               size}
urnor16m=mw.b ${baseaddr} ff 1000000; tftpboot ${baseaddr} rootfs.squashfs.${soc                               } && sf probe 0; sf erase 0x350000 0xa00000; sf write ${baseaddr} 0x350000 ${fil                               esize}
uknand=mw.b ${baseaddr} ff 1000000; tftpboot ${baseaddr} uImage.${soc} && nand e                               rase 0x100000 0x300000; nand write ${baseaddr} 0x100000 0x300000
urnand=mw.b ${baseaddr} ff 1000000; tftpboot ${baseaddr} rootfs.ubi.${soc} && na                               nd erase 0x400000 0x7c00000; nand write ${baseaddr} 0x400000 ${filesize}
mtdpartsubi=setenv mtdparts hinand:256k(boot),768k(wtf),3072k(kernel),-(ubi)
mtdpartsnand=setenv mtdparts hinand:256k(boot),768k(wtf),3072k(kernel),10240k(ro                               otfs),-(rootfs_data)
mtdpartsnor8m=setenv mtdparts hi_sfc:256k(boot),64k(env),2048k(kernel),5120k(roo                               tfs),-(rootfs_data)
mtdpartsnor16m=setenv mtdparts hi_sfc:256k(boot),64k(env),3072k(kernel),10240k(r                               ootfs),-(rootfs_data)
nfsroot=/srv/nfs/hi3516cv200
bootargsnfs=mem=${osmem} console=ttyAMA0,115200 panic=20 root=/dev/nfs rootfstyp                               e=nfs ip=${ipaddr}:::255.255.255.0::eth0 nfsroot=${serverip}:${nfsroot},v3,noloc                               k rw ${extras}
bootargsubi=mem=${osmem} console=ttyAMA0,115200 panic=20 init=/init root=ubi0:ro                               otfs rootfstype=ubifs ubi.mtd=3,2048 mtdparts=${mtdparts} ${extras}
bootnfs=setenv setargs setenv bootargs ${bootargsnfs}; run setargs; tftpboot ${b                               aseaddr} uImage.${soc}; bootm ${baseaddr}
bootcmdnand=setenv setargs setenv bootargs ${bootargs}; run setargs; nand read $                               {baseaddr} 0x100000 0x300000; bootm ${baseaddr}
bootcmdubi=setenv setargs setenv bootargs ${bootargsubi}; run setargs; nand read                                ${baseaddr} 0x100000 0x300000; bootm ${baseaddr}
bootcmdnor=setenv setargs setenv bootargs ${bootargs}; run setargs; sf probe 0;                                sf read ${baseaddr} 0x50000 0x300000; bootm ${baseaddr}
setnand=run mtdpartsubi; setenv bootcmd ${bootcmdubi}; saveenv; reset
setnor8m=run mtdpartsnor8m; setenv bootcmd ${bootcmdnor}; saveenv; reset
setnor16m=run mtdpartsnor16m; setenv bootcmd ${bootcmdnor}; saveenv; reset
restore=n
osmem=32M
soc=hi3516cv200
mtdparts=hi_sfc:256k(boot),64k(env),3072k(kernel),10240k(rootfs),-(rootfs_data)
bootcmd=setenv setargs setenv bootargs ${bootargs}; run setargs; sf probe 0; sf                                read ${baseaddr} 0x50000 0x300000; bootm ${baseaddr}
stdin=serial
stdout=serial
stderr=serial
totalmem=128M
verify=n
ver=U-Boot 2010.06 (Nov 14 2022 - 19:24:15)
setargs=setenv bootargs mem=32M console=ttyAMA0,115200 panic=20 root=/dev/mtdblo                               ck3 rootfstype=squashfs init=/init mtdparts=hi_sfc:256k(boot),64k(env),3072k(ker                               nel),10240k(rootfs),-(rootfs_data)
bootargs=mem=32M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=                               squashfs init=/init mtdparts=hi_sfc:256k(boot),64k(env),3072k(kernel),10240k(roo                               tfs),-(rootfs_data)
ipaddr=192.168.1.10
serverip=192.168.1.44
sensor=ar0130
rebooted - no changes, the sensor changed for ar0130 ,but no image still. Shall I  do firstboot at the begin ?

FlyRouter, 2023-02-13:
Please connect camera to Inet and run
`ipctool`

FlyRouter, 2023-02-13:
Then make the settings and follow up

`fw_setenv sensor ar0130`

Reboot device via power

Run again

`ipctool`

FlyRouter, 2023-02-13:
For a faster dialogue, I recommend that you install the messenger Telegram and communicate through it.
Worse options are Discord and Gitter, in the latter you can even communicate over the web.

https://openipc.org/our-channels

sergshadrap, 2023-02-13:
Ok, guys, anyway thanks for trying to help! 
ipctool  shows the same config - that is surprisingly for me - Ar0237 sits on I2C bus : 

chip:
  vendor: HiSilicon
  model: 3516CV200
board:
  vendor: OpenIPC
  version: 2.3.02.11
ethernet:
  mac: "00:00:23:34:45:66"
  u-mdio-phyaddr: 1
  phy-id: 0x02430c54
  d-mdio-phyaddr: 0
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: e11288e2
      - name: env
        size: 0x10000
        sha1: 0dfc3081
        contains:
          - name: uboot-env
            offset: 0x0
      - name: kernel
        size: 0x300000
        sha1: 11822238
      - name: rootfs
        size: 0xa00000
        path: /,squashfs
        sha1: b417afdd
      - name: rootfs_data
        size: 0x2b0000
        path: /overlay,jffs2,rw
    size: 16M
    addr-mode: 3-byte
ram:
  total: 128M
  media: 96M
firmware:
  u-boot: "2010.06 (Nov 14 2022 - 19:24:15)"
  kernel: "4.9.37 (Sat Feb 11 22:08:08 UTC 2023)"
  toolchain: gcc version 8.4.0 (Buildroot -gff5aa3b6-dirty)
  sdk: "Hi3518EV200_MPP_V1.0.5.0 B060 Release (Mar 21 2018, 08:56:43)"
  main-app: /usr/bin/majestic
sensors:
- vendor: ON Semiconductor
  model: AR0237
  control:
    bus: 0
    type: i2c
    addr: 0x20
  data:
    type: DC
  clock: 27MHz
Will look for you in Telegram

FlyRouter, 2023-02-13:
Install fw_setenv sensor ar0230 and create a configuration file, as a base it can be taken from another type of processor and experiment a lot.

sergshadrap, 2023-02-13:
I tried as you said - doesn't bring a success ..  but what about input mode in ini ? Shall I set as INPUT_MODE_CMOS = 4 ?

dimerr, 2023-02-13:
if driver is DC, then yes, otherwise you'll have to adjust/recompile the driver with DC init

sergshadrap, 2023-02-13:
No , it's not. Ok, thanks.

pianist, 2026-08-15:
Any update for ar0230? I have device with it and can do research.

widgetii, 2026-08-16:
**Re-opening the direction here — this is doable now.** The "drivers are compiled into the app" note above was about extracting from the *stock* firmware; it's not a limitation of OpenIPC.

OpenIPC sensor drivers are open source and built from source in **[OpenIPC/openhisilicon](https://github.com/OpenIPC/openhisilicon)**, under `libraries/sensor/<soc>/<vendor>_<sensor>/`. AR0237 already exists there for **cv300 / ev200 / av100** — it just hasn't been ported to **cv200** yet. AR0237 is a HiSPi sensor, and cv200 already ships the HiSPi **AR0230** (`libraries/sensor/hi3516cv200/aptina_ar0230`), so the bus path is proven — this is a missing-driver gap, not a hardware one.

cv200 and cv300 are the same driver API tier (`docs/sensor-driver-api.md §3.2`), so the clean way to add it is a **dual-platform, single-source port** — exactly like PRs #79 (jxf22) and #80 (sc2235): keep the canonical cv300 `.c`, add a cv200 `Makefile` that reuses it with `-Dhi3516cv200`, and `#if`-gate the V2/V3 differences (use the cv200 AR0230 driver as the reference for the V2 side). Then add an `/etc/sensors/ar0237_…_1080p.ini`.

PRs welcome — happy to review one against openhisilicon.

BlueMax, 2026-08-16:
@flyrouter 
> For a faster dialogue, I recommend that you install the messenger Telegram and communicate through it.
> Worse options are Discord and Gitter, in the latter you can even communicate over the web.
> 
> https://openipc.org/our-channels

Are Gitter/Discord still valid? If so, what are the channels to connect to? Are there any other chats beside Telegram? What about opening Github Discussions?