OpenIPC/firmware #641: Campcam products

Sshepard06, 2023-02-12:
Hi there, I'm brand new to this. Please be patient if I ask something stupid or violate a cultural norm.

I reached out to the email address dev@openipc.org and Igor Zalatov said I should posty inquiry here.  The company Campcam has cameras with solid hardware, but I haven't been able to access it via a browser/IP address.  For this reason, I'm suspicious of these cameras and who else is watching the feeds.  They require you to use their app it's seems.

What I'd like to do is convert these to cameras to IP access with some third party software and get rid of their firmware all together.  I can open these off and provide pictures of the inside.

2K Security Camera Wireless Outdoor with 360° PTZ, Battery Powered WiFi Camera for Home Security, 4MP Color Night Vision, Spotlight & Siren, PIR Human Detection, 2-Way Talk, IP66 Waterproof https://a.co/d/b250IPk

Light Bul6 Security Camera, Upgrade 10X Hybrid Zoom Light Bul6 Camera,Dual Lens,Color Night Vision,Siren & Light Alarm,Tracking,Indoor Pan/Tilt Security Camera,Works with Alexa https://a.co/d/6RCtOoi

Camcamp Wireless WiFi Light Bulb... https://www.amazon.com/dp/B0B49QRRS5?ref=ppx_pop_mob_ap_share

2.4G WiFi Battery Powered Home Security Camera Indoor, 180 Days Standby, PIR Motion Detection, SD/Cloud Storage, No Glow Night Vision and APP Alert, Wireless Portable Surveillance Nanny Cam (2 Pack) https://a.co/d/8fvmJ9k

[image]

themactep, 2023-02-12:
Can you please provide more photos from other sides of these boards?

Sshepard06, 2023-02-12:
[image]
[image]
[image]
[image]

Sshepard06, 2023-02-12:
Didn't mean to close this.

Sshepard06, 2023-02-12:
[image]
[image]
[image]
[image]

> Can you please provide more photos from other sides of these boards?

[image]
[image]
[image]
[image]

Sshepard06, 2023-02-13:
>

themactep, 2023-02-13:
I think we are missing the most important view here - the bottom side of the large board. Where SoC is supposedly resides.

Sshepard06, 2023-02-13:
> I think we are missing the most important view here - the bottom side of the large board. Where SoC is supposedly resides.

[image]
[image]
[image]

FlyRouter, 2023-02-13:
@Sshepard06 

Why do you keep closing and opening this Issue ?

themactep, 2023-02-13:
Ingenic T31 is supported. But you need to understand that while you get basic functionality like streaming video an taking pictures, you will most likey lose battery optimized perfor.mance, PIR sensor support and some other features available in stock firmware. You will have to work closely with us to get any missing features implemented as we do not have such a hardware on hands. Also, if you decide to go with flashing OpenIPC on this one, __do make a full firmware backup__. Consider it a madatory condition.

Your camera has exposed, marked and easily accessible UART, so flashing it should not be a problem.

Sshepard06, 2023-02-13:
> @Sshepard06
> 
> Why do you keep closing and opening this Issue ?

A general unfamiliarity with the system.

Sshepard06, 2023-02-13:
Camcamp Wireless WiFi Light Bulb... https://www.amazon.com/dp/B0B49QRRS5?ref=ppx_pop_mob_ap_share

[image]
[image]
[image]
[image]
[image]

Sshepard06, 2023-02-13:
https://www.amazon.com/dp/B0BB9RNDW3?ref_=cm_sw_r_apan_dp_945XEMVRKC2BCJ6CNZNW

[image]
[image]
[image]

FlyRouter, 2023-02-13:
The Fullhan processor is now almost unsupported
https://openipc.org/cameras/vendors/fullhan

FlyRouter, 2023-02-13:
The Ingenic T31 processor is supported but we need to know the type of WiFi chip. 
And in any case now there will be no availability of motor control.

Sshepard06, 2023-02-13:
2K Security Camera Wireless Outdoor with 360° PTZ, Battery Powered WiFi Camera for Home Security, 4MP Color Night Vision, Spotlight & Siren, PIR Human Detection, 2-Way Talk, IP66 Waterproof https://a.co/d/b250IPk

[image]
[image]
[image]
[image]
[image]

FlyRouter, 2023-02-13:
The Anyka processor is now almost unsupported
https://openipc.org/cameras/vendors/anyka

Looks like you've collected the maximum zoo in one place. ;)

Sshepard06, 2023-02-13:
> The Ingenic T31 processor is supported but we need to know the type of WiFi chip. And in any case now there will be no availability of motor control.

Is there anyway to bypass the PTZ functions or use them in a different way?  I really wish they'd just let me access these cameras via IP/Browser.

Sshepard06, 2023-02-13:
> Ingenic T31 is supported. But you need to understand that while you get basic functionality like streaming video an taking pictures, you will most likey lose battery optimized perfor.mance, PIR [...]

@themactep how do I back this up?  Is there a tutorial in terms of backing up/retaining original firmware?

themactep, 2023-02-13:
Install Telegram client and join our English channel. It will be easier to converse there than in this thread. Ping me there for details.

Sshepard06, 2023-02-14:
> The Ingenic T31 processor is supported but we need to know the type of WiFi chip. And in any case now there will be no availability of motor control.

@FlyRouter 

https://www.amazon.com/dp/B0BB9RNDW3?ref_=cm_sw_r_apan_dp_945XEMVRKC2BCJ6CNZNW

[image]
[image]

FlyRouter, 2023-02-14:
RTL8188FU supported, all OK