OpenIPC/firmware #642: 2 things...

CKlymko, 2023-02-15:
Is IPCtool missing from the recent builds of 3518ev100? 

And also I noticed that there seems to be support for the TI DM36x chips, but what build are they packaged with? I don't see anything for TI or davinci chips in the list of SoCs.

FlyRouter, 2023-02-15:
The ipctool is a cloud utility, it is pulled from GitHub.
Just run the command on a camera with internet and it will be available.
Its smaller brother, ipcinfo, is available locally.

The DM365 and DM368 are still under development. 
If you have such devices, please provide a photo of them.

CKlymko, 2023-02-15:
Ok. Unfortunately the cameras I need ipctool on are wireless only and I
need to check and set some registers to try an enable the wireless adapter.

The dm365 are grandstream cameras models gvx3610_hd. I'll grab some
pictures of the boards a little later.

FlyRouter, 2023-02-15:
You can download the utility to an SD card and use launching from it.
https://github.com/OpenIPC/ipctool/releases/tag/latest

Also pay attention to the GPIO table
https://wiki.openipc.org/en/gpio-settings.html

CKlymko, 2023-02-18:
[image]
[image]
[image]
This is the board from the back half of the camera. The one that has the SoC on it as well as all the connections.
These cameras were those grandstream sip enabled ones. I just have a few that are EOL here and was thinking of trying to give them some new life.

VictorDrijkoningen, 2026-11-05:
I believe this issue can be closed. Is that so? @CKlymko