OpenIPC/firmware #646: U-boot does not boot uImage from MX25L6436F / MX25L6406E

Piorun2002, 2023-02-20:
I'm trying to install openipc-hi3518ev200-lite-8mb.bin, after flashing the error is:

```
Wrong Image Format for bootm command
ERROR: can't get kernel image!
resetting ...

```
Flashig procedure
```
xmtech # tftp 0x82000000 openipc-hi3518ev200-lite-8mb.bin
Hisilicon ETH net controler
MAC:   00-12-14-3A-5C-21
eth0 : phy status change : LINK=UP : DUPLEX=FULL : SPEED=100M
TFTP from server 192.168.1.77; our IP address is 192.168.1.99
Download Filename 'openipc-hi3518ev200-lite-8mb.bin'.
Download to address: 0x82000000
Downloading: #  [ Connected ]
################################        [ 1.000 MB]
################################        [ 2.000 MB]
################################        [ 3.000 MB]
################################        [ 4.000 MB]
################################        [ 5.000 MB]
################################        [ 6.000 MB]
################################        [ 7.000 MB]
################################        [ 8.000 MB]

done
Bytes transferred = 8388608 (800000 hex)
xmtech # sf probe 0; sf lock 0; sf erase 0x0 0x800000; sf write 0x82000000 0x0 0x800000
8192 KiB hi_fmc at 0:0 is now current device
unlock all block.
Erasing at 0x800000 -- 100% complete.
Writing at 0x800000 -- 100% complete.
xmtech # reset
resetting ...

```

Old U-Boot shows that Nor is "MX25L6406E"  in new one one is "MX25L6436F"

New:
```
U-Boot 2010.06 (Nov 14 2022 - 19:23:46)

Check Flash Memory Controller v100 ... Found
SPI Nor(cs 0) ID: 0xc2 0x20 0x17
Block:64KB Chip:8MB Name:"MX25L6436F"
SPI Nor total size: 8MB
Cannot found a valid SPI Nand Device
MMC:
EMMC/MMC/SD controller initialization.
Card did not respond to voltage select!
No EMMC/MMC/SD device found !
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
RAM size: 64MB
*No SD card found!
USB:   scanning bus for devices... 1 USB Device(s) found
0 Storage Device(s) found
Hit any key to stop autoboot:  0
8192 KiB hi_fmc at 0:0 is now current device

Wrong Image Format for bootm command
ERROR: can't get kernel image!
resetting ...
```

Old:
```

U-Boot 2010.06 (Mar 29 2018 - 11:00:43)

Check Flash Memory Controller v100 ... Found
SPI Nor(cs 0) ID: 0xc2 0x20 0x17
Block:64KB Chip:8MB Name:"MX25L6406E"
CONFIG_CLOSE_SPI_8PIN_4IO = y.
at hifmc100_setTB() mid:0xc2,chipsize:0x800000 <no>.
at xm_get_macronix_type() cr:0x0.
lk[7 => 0x800000]
SPI Nor total size: 8MB
In:    serial
Out:   serial
Err:   serial
Press Ctrl+C to stop autoboot
CFG_BOOT_ADDR:0x58050000
8192 KiB hi_fmc at 0:0 is now current device

### boot load complete: 1890528 bytes loaded to 0x82000000
### SAVE TO 80008000 !
## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   linux
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1890464 Bytes = 1.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

```
Looks that new u-Boot can WRITE to  Nor (I can install backup) but it can't READ from Nor to Ram 
```
OpenIPC # sf probe 0
8192 KiB hi_fmc at 0:0 is now current device
OpenIPC # sf read 0x82000000 0x0 0x800000
OpenIPC # md.b 0x82000000
82000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
82000010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
82000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
82000030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
OpenIPC # 

```

I'm sure that openipc-hi3518ev200-lite-8mb.bin,  is correct because I see uImage_data in the hexview  ... but U-Boot read only 0 0 0 from Nor

dimerr, 2023-02-20:
[u-boot-hi3518ev200-universal.zip](https://github.com/OpenIPC/firmware/files/10787362/u-boot-hi3518ev200-universal.zip)
try this

Piorun2002, 2023-02-20:
Looks that this version can read Nor
```
OpenIPC # <INTERRUPT>
OpenIPC # sf probe 0
8192 KiB hi_fmc at 0:0 is now current device
OpenIPC # sf read 0x82000000 0x0 0x100

OpenIPC # md.b 0x82000000
82000000: 17 04 00 ea 14 f0 9f e5 14 f0 9f e5 14 f0 9f e5    ................
82000010: 14 f0 9f e5 14 f0 9f e5 14 f0 9f e5 14 f0 9f e5    ................
82000020: a0 12 70 80 00 13 70 80 60 13 70 80 c0 13 70 80    ..p...p.`.p...p.
82000030: 20 14 70 80 80 14 70 80 e0 14 70 80 78 56 34 12     .p...p...p.xV4.

```

Piorun2002, 2023-02-20:
I was able to boot the kernel.. but then I got 'kernel panic'

```
himci_control_cclk:255: disable or enable clk is timeout!

List of all partitions:
0100           65536 ram0  (driver?)
0101           65536 ram1  (driver?)
0102           65536 ram2  (driver?)
0103           65536 ram3  (driver?)
0104           65536 ram4  (driver?)
0105           65536 ram5  (driver?)
0106           65536 ram6  (driver?)
0107           65536 ram7  (driver?)
0108           65536 ram8  (driver?)
0109           65536 ram9  (driver?)
010a           65536 ram10  (driver?)
010b           65536 ram11  (driver?)
010c           65536 ram12  (driver?)
010d           65536 ram13  (driver?)
010e           65536 ram14  (driver?)
010f           65536 ram15  (driver?)
1f00             256 mtdblock0  (driver?)
1f01              64 mtdblock1  (driver?)
1f02            2048 mtdblock2  (driver?)
1f03            5120 mtdblock3  (driver?)
1f04             704 mtdblock4  (driver?)
No filesystem could mount root, tried:  squashfs

Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.9.37 #1
Hardware name: Hisilicon Hi3518EV20X (Flattened Device Tree)
[<c000e9d0>] (unwind_backtrace) from [<c000cc80>] (show_stack+0x10/0x14)
[<c000cc80>] (show_stack) from [<c00677fc>] (panic+0xb8/0x230)
[<c00677fc>] (panic) from [<c04eb260>] (mount_block_root+0x238/0x274)
[<c04eb260>] (mount_block_root) from [<c04eb470>] (mount_root+0xd8/0x100)
[<c04eb470>] (mount_root) from [<c04eb5f0>] (prepare_namespace+0x158/0x1b8)
[<c04eb5f0>] (prepare_namespace) from [<c04eada8>] (kernel_init_freeable+0x17c/0x1c4)
[<c04eada8>] (kernel_init_freeable) from [<c03e53e8>] (kernel_init+0x8/0xec)
[<c03e53e8>] (kernel_init) from [<c000a3e0>] (ret_from_fork+0x14/0x34)
Rebooting in 20 seconds..

```

Full log

```
U-Boot 2010.06-dirty (Feb 20 2023 - 22:34:50)

Check Flash Memory Controller v100 ... Found
SPI Nor(cs 0) ID: 0xc2 0x20 0x17
Block:64KB Chip:8MB Name:"MX25L6436F"
SPI Nor total size: 8MB
Cannot found a valid SPI Nand Device
MMC:
EMMC/MMC/SD controller initialization.
Card did not respond to voltage select!
No EMMC/MMC/SD device found !
In:    serial
Out:   serial
Err:   serial
RAM size: 64MB
*No SD card found!
USB:   scanning bus for devices... 1 USB Device(s) found
0 Storage Device(s) found
Hit any key to stop autoboot:  0
8192 KiB hi_fmc at 0:0 is now current device

## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   Linux-4.9.37-hi3518ev200
[... 227 lines cut ...]
010c           65536 ram12  (driver?)
010d           65536 ram13  (driver?)
010e           65536 ram14  (driver?)
010f           65536 ram15  (driver?)
1f00             256 mtdblock0  (driver?)
1f01              64 mtdblock1  (driver?)
1f02            2048 mtdblock2  (driver?)
1f03            5120 mtdblock3  (driver?)
1f04             704 mtdblock4  (driver?)
No filesystem could mount root, tried:  squashfs

Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.9.37 #1
Hardware name: Hisilicon Hi3518EV20X (Flattened Device Tree)
[<c000e9d0>] (unwind_backtrace) from [<c000cc80>] (show_stack+0x10/0x14)
[<c000cc80>] (show_stack) from [<c00677fc>] (panic+0xb8/0x230)
[<c00677fc>] (panic) from [<c04eb260>] (mount_block_root+0x238/0x274)
[<c04eb260>] (mount_block_root) from [<c04eb470>] (mount_root+0xd8/0x100)
[<c04eb470>] (mount_root) from [<c04eb5f0>] (prepare_namespace+0x158/0x1b8)
[<c04eb5f0>] (prepare_namespace) from [<c04eada8>] (kernel_init_freeable+0x17c/0x1c4)
[<c04eada8>] (kernel_init_freeable) from [<c03e53e8>] (kernel_init+0x8/0xec)
[<c03e53e8>] (kernel_init) from [<c000a3e0>] (ret_from_fork+0x14/0x34)
Rebooting in 20 seconds..

```

dimerr, 2023-02-21:
I'll post a patch for kernel later. Quite strange, this is a widely used flash, was sure its working fine, maybe its something about your board schematics though

Piorun2002, 2023-02-24:
>maybe its something about your board schematics though
Can I help in any way ?  I have modified the original firmware with telnet access, so I can collect additional logs if something is needed.

dimerr, 2023-02-25:
[uImage.zip](https://github.com/OpenIPC/firmware/files/10830448/uImage.zip)
try this kernel

Piorun2002, 2023-02-25:
Unfortunately doesn't work

```
U-Boot 2010.06-dirty (Feb 20 2023- 22:34:50)
.
## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   Linux-4.9.37-hi3518ev200
.
Booting Linux on physical CPU 0x0
Linux version 4.9.37 (pk@ps) (gcc version 8.4.0 (Buildroot -gbc9837fe-dirty) ) #2 Sat Feb 25 09:32:27 MSK 2023
.
himci_control_cclk:255: disable or enable clk is timeout!

List of all partitions:
0100           65536 ram0  (driver?)
..
1f03            5120 mtdblock3  (driver?)
1f04             704 mtdblock4  (driver?)
No filesystem could mount root, tried:  squashfs

Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.9.37 #2
Hardware name: Hisilicon Hi3518EV20X (Flattened Device Tree)
[<c000e9d0>] (unwind_backtrace) from [<c000cc80>] (show_stack+0x10/0x14)
..
[<c03e53e8>] (kernel_init) from [<c000a3e0>] (ret_from_fork+0x14/0x34)
Rebooting in 20 seconds..

```

full log
```
System startup

U-Boot 2010.06-dirty (Feb 20 2023 - 22:34:50)

Check Flash Memory Controller v100 ... Found
SPI Nor(cs 0) ID: 0xc2 0x20 0x17
Block:64KB Chip:8MB Name:"MX25L6436F"
Spi is locked. lock address[0 => 0x700000]
SPI Nor total size: 8MB
Cannot found a valid SPI Nand Device
MMC:
EMMC/MMC/SD controller initialization.
Card did not respond to voltage select!
No EMMC/MMC/SD device found !
In:    serial
Out:   serial
Err:   serial
RAM size: 64MB
*No SD card found!
USB:   scanning bus for devices... 1 USB Device(s) found
0 Storage Device(s) found
Hit any key to stop autoboot:  0
8192 KiB hi_fmc at 0:0 is now current device

## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   Linux-4.9.37-hi3518ev200
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2048923 Bytes = 2 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.9.37 (pk@ps) (gcc version 8.4.0 (Buildroot -gbc9837fe-dirty) ) #2 Sat Feb 25 09:32:27 MSK 2023
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[... 215 lines cut ...]
010c           65536 ram12  (driver?)
010d           65536 ram13  (driver?)
010e           65536 ram14  (driver?)
010f           65536 ram15  (driver?)
1f00             256 mtdblock0  (driver?)
1f01              64 mtdblock1  (driver?)
1f02            2048 mtdblock2  (driver?)
1f03            5120 mtdblock3  (driver?)
1f04             704 mtdblock4  (driver?)
No filesystem could mount root, tried:  squashfs

Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.9.37 #2
Hardware name: Hisilicon Hi3518EV20X (Flattened Device Tree)
[<c000e9d0>] (unwind_backtrace) from [<c000cc80>] (show_stack+0x10/0x14)
[<c000cc80>] (show_stack) from [<c00677fc>] (panic+0xb8/0x230)
[<c00677fc>] (panic) from [<c04eb260>] (mount_block_root+0x238/0x274)
[<c04eb260>] (mount_block_root) from [<c04eb470>] (mount_root+0xd8/0x100)
[<c04eb470>] (mount_root) from [<c04eb5f0>] (prepare_namespace+0x158/0x1b8)
[<c04eb5f0>] (prepare_namespace) from [<c04eada8>] (kernel_init_freeable+0x17c/0x1c4)
[<c04eada8>] (kernel_init_freeable) from [<c03e53e8>] (kernel_init+0x8/0xec)
[<c03e53e8>] (kernel_init) from [<c000a3e0>] (ret_from_fork+0x14/0x34)
Rebooting in 20 seconds..

```

dimerr, 2023-02-25:
it says flash is locked, maybe rootfs isnt there even, please check that, unlock from uboot or something

Piorun2002, 2023-02-25:
flash was locked  - but only u-boot and kernel
`OpenIPC # sf probe 0; sf lock 3
`

after full unlock the error still occurs
```
OpenIPC # sf probe 0; sf lock 0
8192 KiB hi_fmc at 0:0 is now current device
unlock all block.
OpenIPC # reset
..
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
```

dimerr, 2023-02-25:
try reading rootfs offset from uboot and md'ing it, to see hsqs header at least

Piorun2002, 2023-02-25:
looks fine
```

OpenIPC # sf probe 0
8192 KiB hi_fmc at 0:0 is now current device
OpenIPC # sf read 0x82000000 0x250000 0x500000

OpenIPC # md.b 0x82000000
82000000: 68 73 71 73 27 03 00 00 2e 19 e8 63 00 00 02 00    hsqs'......c....
82000010: 35 00 00 00 04 00 11 00 c0 00 02 00 04 00 00 00    5...............
82000020: 57 0f d6 12 00 00 00 00 02 08 4f 00 00 00 00 00    W.........O.....
82000030: fa 07 4f 00 00 00 00 00 ff ff ff ff ff ff ff ff    ..O.............
OpenIPC #

```

dimerr, 2023-02-26:
We've tested 18ev200 with same spi flash and it works even without patching...maybe it's a hardware fault ?

Piorun2002, 2023-02-27:
The hardware rather works properly, because I can upload the original software and then the system starts.
I think the problem is related to the NOR reading - in the boot loader you corrected it perfectly.
Could you add some diagnostic data to the kernel to see if it can read data from the nor memory ?

dimerr, 2023-02-27:
I can suggest booting from NFS, then you will be able to experiment

Piorun2002, 2023-02-27:
I can load kernel from NFS - but to start roots (hsqs) and overlay (jffs) system needs mtd - or I'm wrong ?

dimerr, 2023-02-27:
kernel from tftp, rootfs on nfs. check bootnfs cmd in uboot

Piorun2002, 2023-03-02:
I was able to boot from NFS, what next :)

logs

```
Linux 192.168.1.219 4.9.37 #1 Sat Feb 18 23:13:41 UTC 2023 armv5tejl GNU/Linux
/dev # dmesg
Booting Linux on physical CPU 0x0
Linux version 4.9.37 (runner@fv-az434-124) (gcc version 8.4.0 (Buildroot -gff5aa3b6-dirty) ) #1 Sat Feb 18 23:13:41 UTC 2023
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
OF: fdt:Machine model: Hisilicon HI3518EV20X DEMO Board
Memory policy: Data cache writeback
On node 0 totalpages: 8192
free_area_init_node: node 0, pgdat c0537b60, node_mem_map c1fb9000
  Normal zone: 64 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 8192 pages, LIFO batch:0
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Kernel command line: mem=32M console=ttyAMA0,115200 panic=20 init=/bin/sh root=/dev/nfs rootfstype=nfs ip=192.168.1.219:::255.255.255.0::eth0 nfsroot=192.168.1.2:/mnt/cam_rootfs,v3,nolock,tcp rw
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 26824K/32768K available (3969K kernel code, 144K rwdata, 908K rodata, 168K init, 217K bss, 5944K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc2800000 - 0xff800000   ( 976 MB)
    lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03e89d8   (3971 kB)
      .init : 0xc04ea000 - 0xc0514000   ( 168 kB)
      .data : 0xc0514000 - 0xc05381c0   ( 145 kB)
       .bss : 0xc05381c0 - 0xc056e920   ( 218 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:16 nr_irqs:16 16
VIC @fe0d0000: id 0x00641190, vendor 0x41
clocksource: arm,sp804: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 637086815595 ns
sched_clock: 32 bits at 3000kHz, resolution 333ns, wraps every 715827882841ns
Console: colour dummy device 80x30
Calibrating delay loop... 269.10 BogoMIPS (lpj=1345536)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x80008400 - 0x80008458
devtmpfs: initialized
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 256 (order: -1, 3072 bytes)
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
Serial: AMBA PL011 UART driver
20080000.uart: ttyAMA0 at MMIO 0x20080000 (irq = 21, base_baud = 0) is a PL011 rev2
console [ttyAMA0] enabled
ssp-pl022 200c0000.spi: ARM PL022 driver, device ID: 0x00800022
ssp-pl022 200c0000.spi: mapped registers from 0x200c0000 to fe1c0000
ssp-pl022 200e0000.spi: ARM PL022 driver, device ID: 0x00800022
ssp-pl022 200e0000.spi: mapped registers from 0x200e0000 to fe1e0000
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
clocksource: Switched to clocksource arm,sp804
NET: Registered protocol family 2
TCP established hash table entries: 1024 (order: 0, 4096 bytes)
TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
TCP: Hash tables configured (established 1024 bind 1024)
UDP hash table entries: 256 (order: 0, 4096 bytes)
[... 113 lines cut ...]
himci_wait_cmd:209:
Other CMD is running,please operate cmd again!

himci_control_cclk:255:
disable or enable clk is timeout!

himci_wait_cmd:209:
Other CMD is running,please operate cmd again!

himci_set_cclk:297:
set card MD is running,please operate cmd again!

himci_control_cclk:255:
disable or enable clk is timeout!

himci_wait_cmd:209:
Other CMD is running,please operate cmd again!

himci_control_cclk:255:
disable or enable clk is timeout!

/dev # uname -a
Linux 192.168.1.219 4.9.37 #1 Sat Feb 18 23:13:41 UTC 2023 armv5tejl GNU/Linux

```

Piorun2002, 2023-03-03:
I see that the nor memory is detected by kernel - how to read it ?

```
hisi-sfc hisi_spi_nor.0: SPI Nor ID Table Version 1.0
hisi-sfc hisi_spi_nor.0: all blocks is unlocked.
hisi-sfc hisi_spi_nor.0: mx25l6436f (Chipsize 8 Mbytes, Blocksize 64KiB)

```

Piorun2002, 2023-03-03:
I forgot to add  mtdparts=${mtdparts} to cmdparameters, now I see /dev/mtd*
```
hisi-sfc hisi_spi_nor.0: SPI Nor ID Table Version 1.0
hisi-sfc hisi_spi_nor.0: all blocks is unlocked.
hisi-sfc hisi_spi_nor.0: mx25l6436f (Chipsize 8 Mbytes, Blocksize 64KiB)
5 cmdlinepart partitions found on MTD device hi_sfc
5 cmdlinepart partitions found on MTD device hi_sfc
Creating 5 MTD partitions on "hi_sfc":
0x000000000000-0x000000040000 : "boot"
0x000000040000-0x000000050000 : "env"
0x000000050000-0x000000250000 : "kernel"
0x000000250000-0x000000750000 : "rootfs"
0x000000750000-0x000000800000 : "rootfs_data"
SPI Nand ID Table Version 2.6

```
however all partition are empty/null - the same issue had u-boot (before it was fixed)
```

root@openipc-hi3518ev200:~# cat /dev/mtdblock0 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00040000
root@openipc-hi3518ev200:~# cat /dev/mtdblock1 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00010000
root@openipc-hi3518ev200:~# cat /dev/mtdblock2 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00200000
root@openipc-hi3518ev200:~# cat /dev/mtdblock3 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*

```

dimerr, 2023-03-03:
[uImage.zip](https://github.com/OpenIPC/firmware/files/10884324/uImage.zip)

Piorun2002, 2023-03-03:
Doesn't work:
```
root@openipc-hi3518ev200:~# uname -a
Linux openipc-hi3518ev200 4.9.37 #2 Fri Mar 3 19:41:40 MSK 2023 armv5tejl GNU/Linux
root@openipc-hi3518ev200:~# cat /dev/mtd
mtd0       mtd1ro     mtd3       mtd4ro     mtdblock2
mtd0ro     mtd2       mtd3ro     mtdblock0  mtdblock3
mtd1       mtd2ro     mtd4       mtdblock1  mtdblock4
root@openipc-hi3518ev200:~# cat /dev/mtdblock1 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00010000
root@openipc-hi3518ev200:~# cat /dev/mtdblock2 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00200000
root@openipc-hi3518ev200:~# cat /dev/mtdblock3 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00500000

```

dimerr, 2023-03-03:
[uImage.zip](https://github.com/OpenIPC/firmware/files/10885621/uImage.zip)

Piorun2002, 2023-03-03:
the same

```
root@openipc-hi3518ev200:~# uname -a
Linux openipc-hi3518ev200 4.9.37 #3 Fri Mar 3 22:49:14 MSK 2023 armv5tejl GNU/Linux
root@openipc-hi3518ev200:~# cat /dev/mtdblock1 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00010000
root@openipc-hi3518ev200:~# cat /dev/mtdblock0 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00040000
root@openipc-hi3518ev200:~# cat /dev/mtdblock2 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00200000
root@openipc-hi3518ev200:~#

```

dimerr, 2023-03-03:
[uImage.zip](https://github.com/OpenIPC/firmware/files/10885985/uImage.zip)

Piorun2002, 2023-03-03:
the same, let me know if you need additional dumps/logs ... thank you for your support!
```
root@openipc-hi3518ev200:~# uname -a
Linux openipc-hi3518ev200 4.9.37 #4 Fri Mar 3 23:48:47 MSK 2023 armv5tejl GNU/Linux
root@openipc-hi3518ev200:~# cat /dev/mtdblock1 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00010000
root@openipc-hi3518ev200:~# cat /dev/mtdblock0 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00040000
root@openipc-hi3518ev200:~# cat /dev/mtdblock2 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00200000
root@openipc-hi3518ev200:~#

```

dimerr, 2023-03-04:
In uboot 
`setenv extras "hisi_mci.disabled=1";saveenv`

[uImage.zip](https://github.com/OpenIPC/firmware/files/10887501/uImage.zip)

Piorun2002, 2023-03-04:
Kernel starts much faster (no timeout), cam works but mtd partisions are 00 00 ..
btw; the kernel version is the same as yesterday - is it ok ?
```
root@openipc-hi3518ev200:~# uname -a
Linux openipc-hi3518ev200 4.9.37 #4 Fri Mar 3 23:48:47 MSK 2023 armv5tejl GNU/Linux

root@openipc-hi3518ev200:~# cat /proc/cmdline
mem=32M console=ttyAMA0,115200 panic=20 root=/dev/nfs rootfstype=nfs ip=192.168.1.219:::255.255.255.0::eth0 nfsroot=192.168.10.2:/mnt/cam_rootfs,v3,nolock,tcp rw mtdparts=hi_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data) hisi_mci.disabled=1

root@openipc-hi3518ev200:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00040000 00010000 "boot"
mtd1: 00010000 00010000 "env"
mtd2: 00200000 00010000 "kernel"
mtd3: 00500000 00010000 "rootfs"
mtd4: 000b0000 00010000 "rootfs_data"

root@openipc-hi3518ev200:~# cat /dev/mtdblock0 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00040000
root@openipc-hi3518ev200:~# cat /dev/mtdblock1 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00010000
root@openipc-hi3518ev200:~# cat /dev/mtdblock2 | hexdump -C | more
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00200000

```
Full log;

Piorun2002, 2023-03-06:
Maybe it it is the same issue like here "Kernel does not boot from KH25L6433F flash drive #30"
  Flash Name: XM_MX25L6433F{**0xC22017**), 0x800000.

I have the same chip: SPI Nor(cs 0) **ID: 0xc2 0x20 0x17**

dimerr, 2023-03-06:
It's not similar with anything else, because we are 100% sure this flash works on 18ev200. In your case there are some probs with board schematics, I've tried everything i could,  even at 50MHz it wouldnt work. If even it will somehow start at 33MHz or with some further spi driver hacks i don't see this going to master.

Piorun2002, 2023-03-06:
>It's not similar with anything else, because we are 100% sure this flash works on 18ev200
Ok, I thought it would be done the same way as you did for u-boot.
The strange thing is that I can run a standard build without any problem:
General_HZXM_IPC_HI3518EV200_50H20L_S38_V4.02.R12.Nat.OnvifS.20171010_ALL.bin

dimerr, 2023-03-06:
I did it the same way, i did even much lower speed , than in u-boot. So thats even a XM board. I even have such board myself... maybe you are missing some resistors around flash chip?

Piorun2002, 2023-03-06:
I can run the original firmware, with access via telnet.
How can I read / dump configurations for spi_nor to verify speed ?

dimerr, 2023-03-06:
I dont think it's possible to get such infos. Again, you'd better check your board, compare with the reference pics, there must be something wrong with the board.

Piorun2002, 2023-03-06:
How to use (register #)
`ipctool spidump / spiset `
to change SPI speed rate ?

Piorun2002, 2023-04-09:
I started the camera on the original firmware, there are only few events in the log related to SPI Nor:
```
Check Flash Memory Controller v100 ...  Found.
SPI Nor(cs 0) ID: 0xc2 0x20 0x17
Block:64KB Chip:8MB Name:"MX25L6406E"
CONFIG_CLOSE_SPI_8PIN_4IO = y.
at xm_get_macronix_type() cr:0x0.
at xm_select_FlashProtMgr() index:0x3,name:Macronix,chipSize:0x800000.<<INFO>>
at xm_setTB() g_SpiNorMFR:0xc2,chipsize:0x800000 <No>.
lk=>4, 0x780000.
SPI Nor total size: 8MB
6 cmdlinepart partitions found on MTD device hi_sfc
6 cmdlinepart partitions found on MTD device hi_sfc
at XMMtd_RegisterProtFreeRgn() index:0,offset:0x7b0000,size:0x50000.
at xm_flash_release_from() offset:0x7b0000, level:4, lockLevelMax:4.
Creating 6 MTD partitions on "hi_sfc":
0x000000000000-0x000000050000 : "boot"
0x000000050000-0x0000003c0000 : "romfs"
0x0000003c0000-0x000000640000 : "user"
0x000000640000-0x000000760000 : "web"
0x000000760000-0x0000007b0000 : "custom"
0x0000007b0000-0x000000800000 : "mtd"
himii: probed

```
does the openipc kernel also set this switch ?
CONFIG_CLOSE_SPI_8PIN_4IO = y.

widgetii, 2026-10-04:
@Piorun2002 — sorry this sat so long. We think we found it. On your board the flash returns nothing to dual-I/O reads; only the plain `0x03` read works. dimerr's universal U-Boot fixed U-Boot by allowing only that read, but the kernel kept using dual-I/O `0xBB` reads, which is why every test kernel still read zeros whatever the clock. Both U-Boot and the kernel now compare the two reads at startup and drop back to `0x03` when they differ (OpenIPC/linux#72, OpenIPC/u-boot-hi3516cv200#11).

If you still have the camera, could you flash this test image the same way as in your first post (`tftp` + `sf write` of the whole 8 MB) and paste the boot log? It's [`openipc-hi3518ev200-lite-8mb-test646.bin`](https://github.com/widgetii/linux/releases/download/test-646-hi3518ev200-single-io/openipc-hi3518ev200-lite-8mb-test646.bin), sha256 `871efd178489c002028fd2b5bef039b1f7550eb13b484903151734320cd41a64` (release page: https://github.com/widgetii/linux/releases/tag/test-646-hi3518ev200-single-io). We're looking for `Read cmd 0xbb returns other data than 0x03` from U-Boot and `falling back to 0x03` from the kernel, followed by a normal boot. Keep your UART connected so you can go back to your backup if anything goes wrong.

closed 2026-11-09 by openipc-ai