OpenIPC/firmware #665: Request for  hi3518ev100 wlan drivers in kernel

CKlymko, 2023-03-12:
I've got some zmodo cameras that ONLY have wireless. There's no ethernet or sd card so my only interaction has been through a serial connection. It has 2 different model numbers Zmodo ZM-SH75D0001 is on the packaging and ZH-IXY1D is on body of the camera.
The wireless adapter in it is a RTL8188EUS and every time I try and bring it up I get errors saying it isn't enabled or there are missing files. The builds from the previous week also seem to reboot the camera over and over because the network isn't up and it looks like majestic is calling for it.

So I did a little experimenting, and flashed uboot, and uImage for 3518ev100 and the rootfs.squashfs from the 3518ev200 which actually attempts to bring up the interface on its own after detecting it but throws an error during modprobe saying that the files are missing (3.0.8)because they would be in the uImage for v200, which doesn't load on my camera. (so I've got v100 of uimage flashed) I had read somewhere else in here that support WAS added, but I think it skipped over in the v100 builds but it looks like it is in v200.

I could try to build it myself, but my concern is that if I get it working and the thing takes an update I'd loose the ability to communicate with the camera again without opening it back up and connecting via serial.

I've got 2 of these things, one is left with the original firmware for reference for the gpio stuff and the one in the pics below is my test unit. I know they aren't that expensive, but I've got them collecting dust because they are only usable through a cloud service since one of their last forced updates. I really just want something I can throw a battery pack on so I can keep an eye on my dog while I'm at work. So I dunno if I'm wasting my time on this model or not. Hopefully someone has some insight or can update the v100 image with the wireless stuff.

Thanks in advance for any help.

The original factory firmware is a very minimal system that starts an app (App3518) where everything seems to happen. Killing the app drops the wireless and it restarts the app. Here is some of the wireless related output from the console on the default factory firmware:
# cfg80211: Calling CRDA to update world regulatory domain
RTL871X: module init start
RTL871X: rtl8188eu v4.3.0.8_13968.20150417
RTL871X: build time: Jun 20 2015 15:24:38
RTL871X:
usb_endpoint_descriptor(0):
RTL871X: bLength=7
RTL871X: bDescriptorType=5
RTL871X: bEndpointAddress=81
RTL871X: wMaxPacketSize=512
RTL871X: bInterval=0
RTL871X: RT_usb_endpoint_is_bulk_in = 1
RTL871X:
usb_endpoint_descriptor(1):
RTL871X: bLength=7
RTL871X: bDescriptorType=5
RTL871X: bEndpointAddress=2
RTL871X: wMaxPacketSize=512
RTL871X: bInterval=0
RTL871X: RT_usb_endpoint_is_bulk_out = 2
RTL871X:
usb_endpoint_descriptor(2):
RTL871X: bLength=7
RTL871X: bDescriptorType=5
RTL871X: bEndpointAddress=3
RTL871X: wMaxPacketSize=512
RTL871X: bInterval=0
RTL871X: RT_usb_endpoint_is_bulk_out = 3
RTL871X: nr_endpoint=3, in_num=1, out_num=2

RTL871X: USB_SPEED_HIGH
RTL871X: CHIP TYPE: RTL8188E
RTL871X: register rtw_netdev_ops to netdev_ops
RTL871X: Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
RTL871X: RF_Type is 3!!
RTL871X: _ConfigNormalChipOutEP_8188E OutEpQueueSel(0x05), OutEpNumber(2)
RTL871X: EEPROM type is E-FUSE
RTL871X: ====> _ReadAdapterInfo8188EU
RTL871X: Boot from EFUSE, Autoload OK !
RTL871X: SetHwReg8188EU: bMacPwrCtrlOn=1
bFWReady == _FALSE call reset 8051...
RTL871X: =====> _8051Reset88E(): 8051 reset success .
RTL871X: efuse_read_phymap_from_txpktbuf bcnhead:0
RTL871X: efuse_read_phymap_from_txpktbuf len:111, lenbak:111, aaa:111, aaabak:111
RTL871X: efuse_read_phymap_from_txpktbuf read count:109
RTL871X: EEPROM ID=0x8129
RTL871X: VID = 0x0BDA, PID = 0x8179
RTL871X: Customer ID: 0x00, SubCustomer ID: 0xCD
RTL871X: Hal_ReadPowerSavingMode88E...bHWPwrPindetect(0)-bHWPowerdown(0) ,bSupportRemoteWakeup(1)
RTL871X: ### PS params=>  power_mgnt(0),usbss_enable(0) ###
RTL871X: ======= Path 0, Channel 1 =======
RTL871X: Index24G_CCK_Base[0][1] = 0x2a
RTL871X: Index24G_BW40_Base[0][1] = 0x2c
RTL871X: ======= Path 0, Channel 2 =======
RTL871X: Index24G_CCK_Base[0][2] = 0x2a
RTL871X: Index24G_BW40_Base[0][2] = 0x2c
RTL871X: ======= Path 0, Channel 3 =======
RTL871X: Index24G_CCK_Base[0][3] = 0x2a
RTL871X: Index24G_BW40_Base[0][3] = 0x2c
RTL871X: ======= Path 0, Channel 4 =======
RTL871X: Index24G_CCK_Base[0][4] = 0x2a
RTL871X: Index24G_BW40_Base[0][4] = 0x2c
RTL871X: ======= Path 0, Channel 5 =======
RTL871X: Index24G_CCK_Base[0][5] = 0x2a
RTL871X: Index24G_BW40_Base[0][5] = 0x2c
RTL871X: ======= Path 0, Channel 6 =======
RTL871X: Index24G_CCK_Base[0][6] = 0x2a
RTL871X: Index24G_BW40_Base[0][6] = 0x2c
RTL871X: ======= Path 0, Channel 7 =======
RTL871X: Index24G_CCK_Base[0][7] = 0x2a
RTL871X: Index24G_BW40_Base[0][7] = 0x2c
RTL871X: ======= Path 0, Channel 8 =======
RTL871X: Index24G_CCK_Base[0][8] = 0x2a
RTL871X: Index24G_BW40_Base[0][8] = 0x2c
RTL871X: ======= Path 0, Channel 9 =======
RTL871X: Index24G_CCK_Base[0][9] = 0x2a
RTL871X: Index24G_BW40_Base[0][9] = 0x2c
RTL871X: ======= Path 0, Channel 10 =======
RTL871X: Index24G_CCK_Base[0][10] = 0x2a
RTL871X: Index24G_BW40_Base[0][10] = 0x2c
RTL871X: ======= Path 0, Channel 11 =======
RTL871X: Index24G_CCK_Base[0][11] = 0x2a
RTL871X: Index24G_BW40_Base[0][11] = 0x2c
RTL871X: ======= Path 0, Channel 12 =======
RTL871X: Index24G_CCK_Base[0][12] = 0x29
RTL871X: Index24G_BW40_Base[0][12] = 0x2b
RTL871X: ======= Path 0, Channel 13 =======
RTL871X: Index24G_CCK_Base[0][13] = 0x29
RTL871X: Index24G_BW40_Base[0][13] = 0x2b
RTL871X: ======= Path 0, Channel 14 =======
RTL871X: Index24G_CCK_Base[0][14] = 0x29
RTL871X: Index24G_BW40_Base[0][14] = 0x2b
RTL871X: ======= TxCount 0 =======
RTL871X: CCK_24G_Diff[0][0]= 0
RTL871X: OFDM_24G_Diff[0][0]= 0
RTL871X: BW20_24G_Diff[0][0]= -2
RTL871X: BW40_24G_Diff[0][0]= 0
RTL871X: EEPROMRegulatory = 0x0
RTL871X: mlmepriv.ChannelPlan = 0x20
RTL871X: CrystalCap: 0x26
RTL871X: EEPROM Customer ID: 0x 0
RTL871X: EEPROM : AntDivCfg = 0, TRxAntDivType = 3
RTL871X: Board Type: 0x 0
RTL871X: ThermalMeter = 0x1a
RTL871X: <==== _ReadAdapterInfo8188EU in 280 ms
RTL871X: init_channel_set ChannelPlan ID 20 Chan num:13
RTL871X: rtw_macaddr_cfg MAC Address  = 8c:18:d9:83:7b:79
RTL871X: bDriverStopped:1, bSurpriseRemoved:0, bup:0, hw_init_completed:0
RTL871X: rtw_ndev_init(wlan0)
RTL871X: _rtw_drv_register_netdev, MAC Address (if1) = 8c:18:d9:83:7b:79
usbcore: registered new interface driver rtl8188eu
RTL871X: module init ret=0

Here's some pics of the camera inside and out.

on board usb wifi
[image]

The front of the camera for reference
[image]

The rear of this camera for reference
[image]

Front side of board (don't mind the mess around the flash chip)
[image]

SoC side of board
[image]

zerog2k, 2023-11-08:
+1 I'm also facing this with a handful of old cameras I'd love to repurpose.
I was so excited to get them to boot up into OpenIPC, but then saddened to learn there is no RTL8188EU support :(

I have dug for a few hours, and found that if I try to enable either of these options (not sure what is the difference here), the build will fail for various reasons.
BR2_PACKAGE_RTL8188EU=y
or
BR2_PACKAGE_RTL8188EUS_OPENIPC=y

looking at these errors, I'm making a guess that it's failing on changed kernel apis, because these configs (br-ext-chip-hisilicon/configs/hi3518ev100_lite_defconfig) are using really ancient 3.0.8 kernel (probably same as original oem builds)

```
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_sta_assoc’:
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:4149:32: error: ‘STATION_INFO_ASSOC_REQ_IES’ undeclared (first use in this function); did you mean ‘STATION_INFO_TX_RETRIES’?
 4149 |                 sinfo.filled = STATION_INFO_ASSOC_REQ_IES;
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                STATION_INFO_TX_RETRIES
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:4149:32: note: each undeclared identifier is reported only once for each function it appears in
```
and
```
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8926:16: error: invalid use of undefined type ‘struct ieee80211_sta_vht_cap’
 8926 |         vht_cap->vht_supported = 1;
      |                ^~
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8927:16: error: invalid use of undefined type ‘struct ieee80211_sta_vht_cap’
 8927 |         vht_cap->cap = IEEE80211_VHT_CAP_RXLDPC;
      |                ^~
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8927:24: error: ‘IEEE80211_VHT_CAP_RXLDPC’ undeclared (first use in this function); did you mean ‘IEEE80211_HT_CAP_RX_STBC’?
 8927 |         vht_cap->cap = IEEE80211_VHT_CAP_RXLDPC;
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~
      |                        IEEE80211_HT_CAP_RX_STBC
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8931:28: error: ‘IEEE80211_VHT_MCS_SUPPORT_0_9’ undeclared (first use in this function)
 8931 |                 mcs_map |= IEEE80211_VHT_MCS_SUPPORT_0_9 << (i*2);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8934:16: error: invalid use of undefined type ‘struct ieee80211_sta_vht_cap’
 8934 |         vht_cap->vht_mcs.rx_mcs_map = cpu_to_le16(mcs_map);
      |                ^~
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8935:16: error: invalid use of undefined type ‘struct ieee80211_sta_vht_cap’
 8935 |         vht_cap->vht_mcs.tx_mcs_map = cpu_to_le16(mcs_map);
      |                ^~
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_init_wiphy’:
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8964:60: error: ‘struct ieee80211_supported_band’ has no member named ‘vht_cap’; did you mean ‘ht_cap’?
 8964 |                         rtw_cfg80211_create_vht_cap(&band->vht_cap);
      |                                                            ^~~~~~~
      |                                                            ht_cap
scripts/Makefile.build:305: recipe for target '/home/jens/build/openipc-firmware/output/build/rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/./os_dep/linux/ioctl_cfg80211.o' failed
```

and so on

I can think of at least 3 possible routes:
- try to use newer kernel, e.g. 4.x? for hi3518ev100. Not sure how hard this would be, or if there are reasons (such as resource limitations, etc) or if no one has attempted to port this
- try to revert to older rtl8188eu or rtl8188eus-openipc, assuming there is some older version compatible w/ 3.0.x kernel. Not sure how to do this with buildroot, or if it's possible
- try to hack/patch/revert the rtl8188eu or rtl8188eus-openipc driver code backwards. I dont like this one.

I wonder if OpenIPC folks know if there are other ways to overcome this?

zerog2k, 2023-11-09:
ok after a bit of a struggle was able to get an older variation of the rtl8188eu drivers to build
here is a build from my branch, with instructions for anyone who would like to test this out.
https://github.com/zerog2k/openipc-firmware/releases/tag/hi3518ev100-rtl8188eu
if I can get some confirmation it's working, I'll open the PR back to here

CKlymko, 2023-11-15:
I'm trying out your fix tonight. But from your instructions it looks like I need to have this thing somehow already on a network. Until wireless (the only network interface) is up I can only send stuff to the board over the serial connection. So the sysupgrade option doesn't work. Are these uImage and squashfs images drop in replacements for the originals? If so I'll just push them over serial and flash them and see how it shakes out.

zerog2k, 2023-11-15:
I actually was able to get a newer rtl8188eu driver built with a kernel patch I submitted to openipc/linux
I’m hoping to push a newer release to my repo fork later today.
If you have sdcard access in your booted os, you should be able to run sysupgrade to provide a local path to the kernel (uImage) and the rootfs (rootfs.squashfs), but if you dont have network you will have to use the -z option to prevent sysupgrade from trying to upgrade itself.

Also to make the wireless come up at boot, you’ll need to set wlandev, wlanssid, & wlanpass uboot env vars (either through setenv/saveenv on uboot, or fw_saveenv in os.)
I’ll put complete instructions on the release notes once I upload the latest.

zerog2k, 2023-11-15:
ok I updated release binaries and instructions
https://github.com/zerog2k/openipc-firmware/releases/tag/hi3518ev100-rtl8188eu-v5.2.2.4

CKlymko, 2023-11-15:
Ohhhh progress. OK, so I had to edit the file /etc/network/interfaces to change wlan0 from manual to auto and also this line :pre-up modprobe 8188eu
I've got my network information in the env that I can see from the running os and from uboot. But now it looks like it isn't pulling that data. The camera is just probing for the network name of "SSID" and is trying password as the psk. I'm going to tinker around with it a little.

Oh and the camera doesn't even have an SD card which is why I asked about loading this. In my case everything has to transfer over kermit on the serial connection. If I get this working I'll turn all my notes into a writeup for this specific camera so you can add it to your fork. There's a lot of people that have this model and would like to use it outside of the zmodo app jail.

zerog2k, 2023-11-15:
if you had previously tweaked around on /etc/network/... directory, you may need to clear your overlayfs (`sysupgrade -n`)
it _should_ automatically `modprobe 8188eu` and enable `wlan0` if you have uboot env `wlandev` set to `rtl8188eu-generic`, and then the default /etc/network/interfaces.d/wlan0 should setup and run wpa_supplicant w/ values from `wlanssid` & `wlanpass`

ref:
https://github.com/OpenIPC/firmware/blob/master/general/overlay/etc/init.d/S40network#L5-L19
https://github.com/OpenIPC/firmware/blob/master/general/overlay/etc/wireless/usb#L23-L27
https://github.com/OpenIPC/firmware/blob/master/general/overlay/etc/network/interfaces.d/wlan0#L2

CKlymko, 2023-11-15:
That did it. It connected and pulled all the network information from my router and I can connect to the web interface. Your fork seems to work. Now I'll just tinker around a little bit and get a full writeup for you to add to your instructions for anyone with this camera that lacks an SD card, or ethernet and only has wifi and console access via the serial connection.

zerog2k, 2023-11-15:
nice - fyi there are wiki docs for:
uboot (including serial uploading): https://github.com/OpenIPC/wiki/blob/master/en/help-uboot.md
wireless settings: https://github.com/OpenIPC/wiki/blob/master/en/wireless-settings.md

closed 2026-10-16 by openipc-ai