OpenIPC/firmware #666: GK7205V300 Not boot

TalusL, 2023-03-12:
[image]

[image]

https://user-images.githubusercontent.com/9034778/224540474-1f18cfc5-32b1-4020-9870-bde406945587.mp4

I flashed openIPC firmware to the spi nor flash.But console output a moving cursor when booting,not any ascii. I am sure the hardware is working good with sock firmware. maybe  GK7205V300 has bootloader lock？

dimerr, 2023-03-12:
Flashed which file? Flashed how?

TalusL, 2023-03-12:
> Flashed which file? Flashed how?

openipc-gk7205v300-lite-8mb(both 16).bin，
flashed and verifyed now. I work while i restore the stock firmware

TalusL, 2023-03-12:
stock firmware boot log
```
System startup

Uncomr▒Y▒.▒Ok!

U-Boot 2016.11 (Jul 22 2022 - 21:21:59 +0800)gk7205v300

Relocation Offset is: 0771e000
Relocating to 47f1e080, new gd at 47eddef0, sp at 47edded0
SPI Nor:  Check Flash Memory Controller v100 ... Found
SPI Nor ID Vable Version 1.0
SPI Nor(cs 0) ID: 0x20 0x40 0x18
Block:64KB Chip:16MB Name:"XM25QH128CHIQ"
SPI Nor tot`l size: 16MB
NAND*  0 MiB
MMC:
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   eth0
Warning: eth0 (eth0) using random MAC address - ce:ab:40:85:3b:22

gpio key muxconfig.
MV038V01 gpio wled_config muxconfig.
MV038V01 gpio redled_config muxconfig.
sensor default clk 27M
[GK_RWA_SIGN_VERIFY]RSA Verify Start...
[GK_RSA_SIGN_VERIFY]: run successfully!
[sw_load_kY.▒▒u▒RSA▒W▒▒▒▒successfully!
magic DEVINFO!,count 11,checksum a38c
Hit any key to stop autoboot:  0
device 0 offset 0x80000, size 0x1f0000

SF: 2031616 bytes @ 0x80000 Read: OK
## Booting kernel from Legacy Image at 42000800 ...
   Image Name:   Linux-4.9.#7
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1987718 Bytes = 1.9 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Loading Kernel Image ... OK

Startiog kernel ...

NR_IRQS:16 nr_irqs:16 16
GZ,"▒▒сinit...
Failed to initialize '/soc/amba▒Յ▒}ѥ▒▒▒▒2001000': -22100.00 BogoMIPS (lpj=50000)
ok
devtmpfs: initialized
implementor 41 arch▒tecture 2 part 30 variant 7 rev 5
Serial: AMBA PL011 UART driver
12040000.uart: ttyAMA0 at MMIO 0x12040000 (irq = 21, base_baud = 0) is a PL011 rev2
ssp-pl022 12070000.spi: ARM PL022 driver, device ID: 0x00041022
ssp-pl022 12071000.spi: ARM PL022 driver, device ID: 0x00041022
exFAT: Version 1.2.9
                    Block layer SCSI generic (bsg) driver version 0.4 loaded (major 252)
io scheduler noop registered
io scheduler deadline registered (`efault)
io scheduler cfq registered
pl061_gpio 120b0000.gpio_chip: PL061 GPIO chip @0x120b0000 registered
pl061_gpio 120b1000.gpio_chip: PL061 GPIO chip @0x120b1000 registered
pl061_gpio 120b2000.gpio_chip: PL061 GPIO chip @0x120b2000 registered
pl061_gpio 120b3000.gpio_chip: PL061 GPIO chip @0x120b3000 registered
pl061_gpio 120b4000.gpio_chip: PL063 GPIO chip @0x120b4000 registered
pl061_gpio 120b5000.gpio_chip: PL061 GPIO chip @0x120b5000 registered
pl061_gpio 120b6000.gpio_chip: PL061 GPIO chip @0x120b6000 registered
pl061_gpio 120b7000▒gpio_chip: PL061 GPIO chip @0x120b7000 registered
[... 586 lines cut ...]
(01-01  0:00:17.842) =CMCC-RTC-V1.5.3=SetDeviceInfo(), chip_module:GK7205V300,chip_uuid:,chi`_brand:gkw
======== cmcc_rtc_setopt ========
(01-01  0:00:17>843) VoipWper::SetOptionValue(),opt:36,value:0
[12900][01-01 00:00:17][INFO][SWOVD led_o▒Y▒5х▒▒́622][882] turn on led
[13900][01-01 00:00:18][INFO][AVS bitrate_auto_proc 81][882] chn:1 kbps:543
***********func:sw_moto_ioctl line:264 type=0,direct=0,angle=0,m_pulse_nums:0
[14660][01-01 00:00:19][WARN][GPIO sw_moto_stop 86][882] sw_moto_s▒
                                                                   5
                                                                    [14660][01-01***********func:sw_moto_ioctl line:264 type=1,direct=0,angle=100,m_pulse_nums81142
 00:00:19][INFO][swapp camera_moto_reset_proc 287][882] start v_dir moto cleck!!!C▒[14930][01-01 00:00:19][INFO][AVS bitrate_auto_proc 81][882] chn:1 kbps:543
[17270][01-01 00:00:20][INFO][network swwifimanager.c 468][731] recv wifi msg:CTRL-EVENT-SCAN-RESULTS
***********func:sw_moto_ioctl line:264 type=1,direct=1,angle=100,m_pulse_nums:1142
[16960][01-01 00:00:21][INFO][swapp sw_versioninfo_proc 128][882] wait network success
*******"***func:sw_moto_ioctl line:264 type=1,direct=0,angle=40,m_pulse_nums:457
***********func:sw_moto_ioctl nine:264 type=1,direct=1,angle=0,m_pulse_nums:0
[1▒LL▒m▒1-01 00:00:23][WARN][GPIO sw_moto_stop 86][882] sw_moto_stop
[18110][01-01 00:00:23][INFO][swapp camera_moto_resume_position 257][882] get cur coordinate :0:0
[18110][01-01 00:00:23][INFO][SWOVD swovd_moto_check 69][882] The device is in a wake-up state, and there is no need to set the motor to sleep
random: crng init done
[21960][01-01 00:00:26][INFO][swapp sw_versioninfo_proc 128][882] wait network success
[24290][01-01 00:00:27][INFO][network swwifimanager.c 468][731] recv wifi msg:CTRL-EVENT-SCAN-RESULTS
[26960][01-01 00:00:31][INFO][swapp sw_versioninfo_proc 128][882] wait network success
[29230][01-01 00:00:34][INFO][AVS bitraTe_auto_proc 81][882] chn:1 kbps:653
[31330][01-01 00:00:34][INFO][network swwifimanager.c 468][731] recv wifi msg:CTRL-EVENT-SCAN-RES
```

dimerr, 2023-03-12:
Could be secureboot, yes, but usually it prints some letters instead spaces... try Burn utility

closed 2023-03-12 by TalusL

vinibali, 2023-05-15:
@TalusL