OpenIPC/firmware #669: HI3516CV100 IMX291 support

DaxonBaggus, 2023-03-13:
XiongMai made an IPC module with HI3516CV100 SoC and IMX291 sensor.

Strangely the IMX291 is supported on many other HI3516 chips, but not on this one. Could something be added or changed to allow this combination to work?

XM product ID: IPG-54H20PL-S
PCB Marking: BLK18C-0090-38X38_S
Sensor: IMX291
FM version: V4.02.R11.00002532.10010.242800.00000
SoC: HI3516CV100

I have around 20 cameras I'm trying to refurbish with OpenIPC so maybe I can help with a donation?

[image]
[image]

widgetii, 2023-03-14:
IMX291 sensor driver should be written from scratch for this model. It will take 4-6 weeks of full-time reverse-engineer work with close interaction with hardware.

I doubt that it's even reasonable judging you can replace all boards to modern version

DaxonBaggus, 2023-03-14:
Thanks for your honest answer. Then sadly I will have to look for a different solution until someone feels the need to go and make this happen.

FlyRouter, 2023-03-14:
I wrote you some questions in PM Telegram, please post as you get a chance.

FlyRouter, 2023-03-16:
I made a small crutch with imx138 since we don't have a driver.
Not sure if it will work well but it breathes nonetheless... ;)
https://openipc.org/snapshots/368807

FlyRouter, 2023-03-16:
Solution

One commit for [sensor init](https://github.com/OpenIPC/firmware/commit/b3b507242557b545a67e0cd0134112e99fd1a3f4)

Set old integrated sensor from Linux
`fw_setenv sensor imx138_spi`

DaxonBaggus, 2023-03-19:
[IPG-54H20PL-S stock firmwares 2532.zip](https://github.com/OpenIPC/firmware/files/11011215/IPG-54H20PL-S.stock.firmwares.2532.zip)

Attached all the versions of the stock firmware for this board. Maybe this contains a driver somewhere? 

The camera briefly worked with the imx138 change, but after changing a couple of settings in the web interface it is now broken, and I have not been able to get it do display an image again.

widgetii, 2023-03-19:
Sensor driver is compiled into Sofia binary statically. There is no way to extract and use it

DaxonBaggus, 2023-03-19:
Ah ok, I think I understand now, you are looking for something like this?
https://github.com/EliasKotlyar/Xiaomi-Dafang-Software/blob/master/drivers/sensors/imx291/imx291.c

widgetii, 2023-03-20:
No, this is driver for Ingenic platform (the code is absolutely useless for HiSilicon)

mpissarello, 2023-04-10:
Same issue as #204 
At November 2022, my camera was half-working, see [my comment](https://github.com/OpenIPC/firmware/issues/204#issuecomment-1305774560), I need to try with the latest OpenIPC firmware.

closed 2026-09-04 by openipc-ai