OpenIPC/firmware #675: [FR] gk7205v200 /v300 u-boot breaking only ctrl+c

whoim2, 2023-03-16:
This is very important for FPV topics. If the camera restarts in flight, for example, by watchdog, then the UART telemetry will stop loading the y-boot and the camera will freeze until the power is reset. Interrupting on e.g. Сtrl+С should solve this problem.

FlyRouter, 2023-03-16:
You need to collect at least 10 signatures for this feature. 
Your colleagues need to sign here and then it will be implemented.

kobzzon, 2023-03-16:
yes, I really need this project!

Haker-che, 2023-03-16:
Implement it, wait!

Ed1975I, 2023-03-16:
Yes, it is very necessary

FlyRouter, 2023-03-16:
I forgot to add...
Accounts created no later than March 15 can participate in the voting...

themactep, 2023-03-16:
I think limiting key combination to Ctrl-C would be beneficial for any platform, regardless FPV or not. My 2 cents.

Ra3eR63, 2023-03-17:
Очень нужный проект!

Platidors, 2023-03-17:
I really need it

AlexPotatoKun, 2023-03-19:
Nice idea, really need it not only for FPV

vbond-app, 2023-03-19:
it is important feature for FPV mode

zhuwei, 2023-03-20:
I really need it!
gk7205v300, What is the minimum latency for 1080P FPV?

whoim2, 2023-03-20:
> I really need it! gk7205v300, What is the minimum latency for 1080P FPV?
At v200 about 200-250ms. Go to "sandbox-fpv" repo and join to us tg group.

dimerr, 2023-03-20:
[u-boot-gk7205v200-universal.zip](https://github.com/OpenIPC/firmware/files/11014202/u-boot-gk7205v200-universal.zip)

whoim2, 2023-03-20:
im sorry, but
```
 flashcp -v /tmp/u-boot-gk7205v200-universal.bin /dev/
mtd0
Erasing block: 4/4 (100%)
Writing kb: 248/248 (100%)
Verifying kb: 248/248 (100%)
```
```
Hit any key to stop autoboot:  0
OpenIPC #
```

whoim2, 2023-03-20:
thanks! this worked u-boot at https://github.com/OpenIPC/sandbox-fpv/tree/master/gk7205v200

zhuwei, 2023-03-21:
> 

"fast 720p@50 "What is it? I don't understand. Can you explain it?

zhuwei, 2023-03-21:
@whoim2 "fast 720p@50 "What is it? I don't understand. Can you explain it?

whoim2, 2023-03-21:
> @whoim2 "fast 720p@50 "What is it? I don't understand. Can you explain it?

50fps > 30fps )

closed 2023-03-21 by whoim2

whoim2, 2023-05-17:
> > > I really need it! gk7205v300, What is the minimum latency for 1080P FPV?
> > 
> > 
> > At v200 about 180-230ms. Go to "sandbox-fpv" repo and join to us tg group.
> 
> Is the group name OpenIPC (EN)?
https://openipc.org/our-channels
Try fpv (ru)