OpenIPC/firmware #676: HI3518EV200 with GC2033 support

balb0x, 2023-03-16:
Hi!
I have a camera board labeled as C26S-V2_Main_V1.1 with a hi3518ev200 SoC. 
I replaced the original firmware with OpenIPC but the camera does not work. The majestic binary gives an unknown sensor error.  Searching in the original firmware logs, i've found what i suppose is the camera sensor:

```
linear mode
GalaxyCore GC2033 1080p MIPI init success!
```
Searching in the OpenIPC WIKI i've found that the hi3516cv200 [supports it](https://github.com/OpenIPC/wiki/blob/561d47304cfdd68e95b9c98d373898016a92cd9e/en/guide-supported-sensors.md)

I've tried building for that SoC so it generates the `libsns_gc2033.so` and tried with different configurations generating a `gc2033_i2c_1080p.ini` and seems to load but i'm not getting any image from the camera.

```
19:19:57  <       majestic> [    main] main@125                      Majestic Lite for HiSilicon, version master+70636c1, built on 2023-03-11
19:19:57  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
19:19:57  <       majestic> [watchdog] watchdog_start@101            Detected Hisilicon Watchdog
19:19:57  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
19:19:57  <       majestic> [     sdk] try_sensor_config@3286        SENSOR=gc2033
19:19:57  <       majestic> [     sdk] find_sensor_config@3249       matched sensor config: gc2033_i2c_1080p.ini
19:19:57  <       majestic> [     sdk] start_sdk@415                 Using /etc/sensors/gc2033_i2c_1080p.ini sensor configuration file
19:19:57  <       majestic> [     sdk] start_sdk@424                 App was built with MPP version: Hi3518EV200_MPP_V1.0.5.0.B060 Release
19:19:57  <       majestic> [     sdk] start_sdk@427                 Current MPP version: HI_VERSION=Hi3518EV200_MPP_V1.0.5.0 B060 Release
19:19:57  <       majestic> [     sdk] start_sdk@447                 sensor gc2033
19:19:57  <       majestic> [     sdk] start_sdk@452                   input_mode CMOS_18V, WDR NONE
19:19:57  <       majestic> [     sdk] start_sdk@459                   dev [1920x1080]@0x0 30fps, BGGR
19:19:57  <       majestic> [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_gc2033.so
19:19:57  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem before allocation: 32580KB
19:19:57  <       majestic> [     sdk] dump_vb_configuration@2164    VB configuration:
19:19:57  <       majestic> [     sdk] dump_vb_configuration@2174      [0]: 3110400 x 3
19:19:57  <       majestic> [     sdk] dump_vb_configuration@2174      [1]: 4147200 x 1
19:19:57  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem after allocation: 19404KB
19:19:57  <       majestic> [     sdk] init_sensor@2360              Sensor driver loaded
19:19:57  <       majestic> [     sdk] init_vi@2807                  SDK is in 'online' mode
19:19:57  <       majestic> [    puts]                               linear mode
19:19:57  <       majestic> [    puts]                               GalaxyCore GC2033 1080p MIPI init success!
19:19:57  <       majestic> [     sdk] log_venc_chn@1255             H.264 1920x1080 20fps 4096Kbit  20 GOP
19:19:57  <     iq_monitor> [     sdk] monitor_thread@344            Loading IQ profile /etc/sensors/iq/default.ini
19:19:57  <       majestic> [     sdk] create_vpss_chn@1002            new venc: 0   vpss_grp: 0,   vpss_chn: 0
19:19:57  <       majestic> [     sdk] start_video@1914                venc took 7704KB of MMZ [1518]
19:19:57  <       majestic> [image_tu] start_image_params_tuning@63  Image tuning task started
19:19:57  <       majestic> [     sdk] start_sdk@559                 HiSilicon SDK started
19:19:57  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem finally: 7620KB
19:19:57  <       majestic> [   httpd] new_http_server@368           HTTP server started on :::80
19:19:57  <       majestic> [    rtsp] rtsp_init@31                  RTSP server started on port 554
19:19:57  <       majestic> [   netip] netip_start@2055              NETIP server started on port 34567
[Func]:HI_MPI_ISP_QueryExposureInfo [Line]:730 [Info]:Get Active Stat Buffer Err
19:19:58  <     iq_monitor> [     sdk] monitor_thread@363            Query exposure info
19:19:58  <     iq_monitor> [     sdk] monitor_thread@363            Unknown error code A01C8046
19:19:59  <       majestic> [     sdk] venc_read@926                 Timeout get any data from venc channel 0
19:19:59  <       majestic> [     sdk] venc_read@946                 NOTE: Current frame is NULL!
19:20:01  <       majestic> [     sdk] venc_read@926                 Timeout get any data from venc channel 0
19:20:01  <       majestic> [     sdk] venc_read@946                 NOTE: Current frame is NULL!
``` 

Is it possible to get that sensor working on the hi3518ev200?

FlyRouter, 2023-03-16:
Do this in the console and reboot the camera first

```
    cli -s .image.contrast 50
    cli -s .image.luminance 50
    cli -s .video0.codec h264
    cli -s .hls.enabled false
```

dimerr, 2023-03-16:
From stock fw, save the following:
/proc/umap/* 
ipctool reginfo
ipctool reginfo --script
ipctool gpio scan

balb0x, 2023-03-16:
I tried the cli commands but it seemed they didn't work. 

Then, i copied the ipctool binary to the stock firmware and theese are the results:

```
/proc/umap/*

acodec  ai      h264e   jpege   rc      tde     vgs
adec    ao      isp     logmpp  rgn     vb      vi
aenc    chnl    ive     md      sys     venc    vpss
```

```
ipctool reginfo

muxctrl_reg0 0x200f0000 0x1 GPIO0_4 [SENSOR_CLK]
muxctrl_reg1 0x200f0004 0 [SENSOR_RSTN] GPIO0_5
muxctrl_reg2 0x200f0008 0x1 GPIO0_6 [FLASH_TRIG] SFC_EMMC_BOOT_MODE SPI1_CSN1 VI_VS_BT1120
muxctrl_reg3 0x200f000c 0x1 GPIO0_7 [SHUTTER_TRIG] SFC_DEVICE_MODE VI_HS_BT1120
muxctrl_reg4 0x200f0010 0x1 GPIO2_0 [RMII_CLK] VO_CLK SDIO1_CCLK_OUT
muxctrl_reg5 0x200f0014 0x1 GPIO2_1 [RMII_TX_EN] VO_VS SDIO1_CARD_DETECT
muxctrl_reg6 0x200f0018 0x1 GPIO2_2 [RMII_TXD0] VO_DATA5 SDIO1_CWPR
muxctrl_reg7 0x200f001c 0x1 GPIO2_3 [RMII_TXD1] VO_DE SDIO1_CDATA1
muxctrl_reg8 0x200f0020 0x1 GPIO2_4 [RMII_RX_DV] VO_DATA7 SDIO1_CDATA0
muxctrl_reg9 0x200f0024 0x1 GPIO2_5 [RMII_RXD0] VO_DATA2 SDIO1_CDATA3
muxctrl_reg10 0x200f0028 0x1 GPIO2_6 [RMII_RXD1] VO_DATA3 SDIO1_CCMD
muxctrl_reg11 0x200f002c 0x1 GPIO2_7 [EPHY_RST] BOOT_SEL VO_HS SDIO1_CARD_POWER_EN
muxctrl_reg12 0x200f0030 0 [GPIO0_3] SPI1_CSN1 VO_DATA0
muxctrl_reg13 0x200f0034 0x1 GPIO3_0 [EPHY_CLK] VO_DATA1 SDIO1_CDATA2
muxctrl_reg14 0x200f0038 0x1 GPIO3_1 [MDCK] BOOTROM_SEL VO_DATA6
muxctrl_reg15 0x200f003c 0x1 GPIO3_2 [MDIO] VO_DATA4
muxctrl_reg16 0x200f0040 0x2 GPIO3_3 SPI0_SCLK [I2C0_SCL]
muxctrl_reg17 0x200f0044 0x2 GPIO3_4 SPI0_SDO [I2C0_SDA]
muxctrl_reg18 0x200f0048 0 [GPIO3_5] SPI0_SDI
muxctrl_reg19 0x200f004c 0 [GPIO3_6] SPI0_CSN
muxctrl_reg20 0x200f0050 0 [GPIO3_7] SPI1_SCLK I2C1_SCL
muxctrl_reg21 0x200f0054 0 [GPIO4_0] SPI1_SDO I2C1_SDA
muxctrl_reg22 0x200f0058 0 [GPIO4_1] SPI1_SDI
muxctrl_reg23 0x200f005c 0 [GPIO4_2] SPI1_CSN0
muxctrl_reg24 0x200f0060 0 [GPIO4_3] I2C2_SDA
muxctrl_reg25 0x200f0064 0 [GPIO4_4] I2C2_SCL
muxctrl_reg26 0x200f0068 0x1 GPIO4_5 [USB_OVRCUR]
muxctrl_reg27 0x200f006c 0x1 GPIO4_6 [USB_PWREN]
muxctrl_reg28 0x200f0070 0 [GPIO0_0] IR_IN TEMPER_DQ
muxctrl_reg29 0x200f0074 0 [GPIO0_1] TEMPER_DQ
muxctrl_reg30 0x200f0078 0 [GPIO0_2] TEMPER_DQ
muxctrl_reg31 0x200f007c 0 [GPIO1_0] VI_DATA13 I2S_BCLK_TX PWM0
muxctrl_reg32 0x200f0080 0 [GPIO1_1] VI_DATA10 I2S_SD_TX UART1_TXD
muxctrl_reg33 0x200f0084 0 [GPIO1_2] VI_DATA12 I2S_MCLK UART1_CTSN
muxctrl_reg34 0x200f0088 0 [GPIO1_3] VI_DATA11 I2S_WS_TX UART2_RXD
muxctrl_reg35 0x200f008c 0 [GPIO1_4] VI_DATA15 VI_VS_SEN I2S_WS_RX UART1_RXD
muxctrl_reg36 0x200f0090 0 [GPIO1_5] VI_DATA14 VI_HS_SEN I2S_BCLK_RX UART1_RTSN
muxctrl_reg37 0x200f0094 0 [GPIO1_6] VI_DATA9 I2S_SD_RX UART2_TXD
muxctrl_reg38 0x200f0098 0x1 GPIO1_7 [SDIO0_CARD_POWER_EN]
muxctrl_reg39 0x200f009c 0x1 GPIO4_7 [SDIO0_CARD_DETECT]
muxctrl_reg40 0x200f00a0 0x1 GPIO5_0 [SDIO0_CWPR]
muxctrl_reg41 0x200f00a4 0x1 GPIO5_1 [SDIO0_CCLK_OUT]
muxctrl_reg42 0x200f00a8 0x1 GPIO5_2 [SDIO0_CCMD]
muxctrl_reg43 0x200f00ac 0x1 GPIO5_3 [SDIO0_CDATA0]
muxctrl_reg44 0x200f00b0 0x1 TEST_CLK [SDIO0_CDATA1] GPIO5_4
muxctrl_reg45 0x200f00b4 0x1 GPIO5_5 [SDIO0_CDATA2]
muxctrl_reg46 0x200f00b8 0x1 GPIO5_6 [SDIO0_CDATA3]
muxctrl_reg47 0x200f00bc 0 [GPIO5_7] EMMC_DATA6 I2S_SD_TX UART1_RTSN
muxctrl_reg48 0x200f00c0 0 [GPIO6_0] EMMC_DATA5 I2S_WS_TX UART1_RXD
muxctrl_reg49 0x200f00c4 0 [GPIO6_1] EMMC_DATA7 I2S_MCLK UART1_CTSN
muxctrl_reg50 0x200f00c8 0 [GPIO6_2] EMMC_DS I2S_SD_RX UART1_TXD
muxctrl_reg51 0x200f00cc 0 [GPIO6_3] EMMC_DATA1 UART2_RXD
muxctrl_reg52 0x200f00d0 0 [GPIO6_4] EMMC_DATA2 I2S_BCLK_TX UART2_TXD
muxctrl_reg53 0x200f00d4 0 [GPIO6_5] JTAG_TRSTN SPI1_CSN1 I2S_MCLK
muxctrl_reg54 0x200f00d8 0 [GPIO6_6] JTAG_TCK SPI1_SCLK I2S_WS_TX I2C1_SCL
[... 76 lines cut ...]
devmem 0x200f00ec 32 0x1
devmem 0x200f00f0 32 0x1
devmem 0x200f00f4 32 0x1
devmem 0x200f00f8 32 0
devmem 0x200f00fc 32 0
devmem 0x200f0100 32 0x1
devmem 0x200f0104 32 0x1
```

```
ipctool gpio scan

Gr: 0, Addr:0x2014003C, Data:0x03 = 0bxxxx0011, Addr:0x20140400, Dir:0x03 = 0bxxxx0011
Gr: 1, Addr:0x201501FC, Data:0x04 = 0bx0000100, Addr:0x20150400, Dir:0x04 = 0bx0000100
Gr: 2, Addr:0x20160000, Data:0x00 = 0bxxxxxxxx, Addr:0x20160400, Dir:0x00 = 0bxxxxxxxx
Gr: 3, Addr:0x20170380, Data:0x00 = 0b000xxxxx, Addr:0x20170400, Dir:0x80 = 0b100xxxxx
Gr: 4, Addr:0x2018007C, Data:0x00 = 0bxxx00000, Addr:0x20180400, Dir:0x47 = 0bxxx00111
Gr: 5, Addr:0x20190200, Data:0x00 = 0b0xxxxxxx, Addr:0x20190400, Dir:0x00 = 0b0xxxxxxx
Gr: 6, Addr:0x201A03FC, Data:0x00 = 0b00000000, Addr:0x201A0400, Dir:0x00 = 0b00000000
Gr: 7, Addr:0x201B00FC, Data:0x00 = 0bxx000000, Addr:0x201B0400, Dir:0x3C = 0bxx111100
Gr: 8, Addr:0x201C000C, Data:0x00 = 0bxxxxxx00, Addr:0x201C0400, Dir:0x03 = 0bxxxxxx11

```

I'll try to adjust to theese parameters in the OpenIPC firmware

dimerr, 2023-03-16:
You should save every file under /proc/umap

balb0x, 2023-03-16:
Theese are the /proc/umap files:
[umap.zip](https://github.com/OpenIPC/firmware/files/10995888/umap.zip)

dimerr, 2023-03-16:
use config from 1034, adjust resolution and mask according to umap/vi and try .so from stock

balb0x, 2023-03-17:
I'm still getting erros:

```
13:59:41  <       majestic> [    main] main@125                      Majestic Lite for HiSilicon, version master+70636c1, built on 2023-03-11
13:59:41  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
13:59:42  <       majestic> [watchdog] watchdog_start@101            Detected Hisilicon Watchdog
13:59:42  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
13:59:42  <       majestic> [     sdk] try_sensor_config@3286        SENSOR=gc2033
13:59:42  <       majestic> [     sdk] find_sensor_config@3249       matched sensor config: gc2033_i2c_1080p.ini
13:59:42  <       majestic> [     sdk] start_sdk@415                 Using /etc/sensors/gc2033_i2c_1080p.ini sensor configuration file
13:59:42  <       majestic> [     sdk] start_sdk@424                 App was built with MPP version: Hi3518EV200_MPP_V1.0.5.0.B060 Release
13:59:42  <       majestic> [     sdk] start_sdk@427                 Current MPP version: HI_VERSION=Hi3518EV200_MPP_V1.0.5.0 B060 Release
13:59:42  <       majestic> [     sdk] start_sdk@447                 sensor gc2033
13:59:42  <       majestic> [     sdk] start_sdk@452                   input_mode CMOS_18V, WDR NONE
13:59:42  <       majestic> [     sdk] start_sdk@459                   dev [1920x1080]@0x0 15fps, GRBG
13:59:42  <       majestic> [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_gc2033.so
13:59:42  <      hi_logger> [    vpss] VPSS_DestroyGrp@4348          [  vpss] !! Disable VpssChn:0 timeout 120ms!!!
13:59:42  <      hi_logger> [     viu] VIU_DRV_DisableChn@1309       [   viu] !! Disable ViChn:0 timeout 120ms!!!
13:59:42  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem before allocation: 32580KB
13:59:42  <       majestic> [     sdk] dump_vb_configuration@2164    VB configuration:
13:59:42  <       majestic> [     sdk] dump_vb_configuration@2174      [0]: 3110400 x 3
13:59:42  <       majestic> [     sdk] dump_vb_configuration@2174      [1]: 4147200 x 1
13:59:42  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem after allocation: 19404KB
13:59:42  <       majestic> [     sdk] init_sensor@2360              Sensor driver loaded
13:59:42  <       majestic> [     sdk] init_vi@2807                  SDK is in 'online' mode
13:59:42  <      hi_logger> [     isp] VREG_DRV_Exit@103             [   isp] The vreg of u32BaseAddr 0x10000 has not registerd!
13:59:42  <       majestic> [    puts]                               linear mode
13:59:42  <       majestic> [    puts]                               GalaxyCore GC2033 1080p MIPI init success!
13:59:42  <       majestic> [     sdk] log_venc_chn@1255             H.264 1920x1080 15fps 4096Kbit  15 GOP
13:59:42  <     iq_monitor> [     sdk] monitor_thread@344            Loading IQ profile /etc/sensors/iq/default.ini
13:59:42  <       majestic> [     sdk] create_vpss_chn@1002            new venc: 0   vpss_grp: 0,   vpss_chn: 0
13:59:42  <       majestic> [     sdk] start_video@1914                venc took 7704KB of MMZ [1518]
13:59:42  <       majestic> [     sdk] start_sdk@559                 HiSilicon SDK started
13:59:42  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem finally: 7620KB
13:59:42  <       majestic> [   httpd] new_http_server@368           HTTP server started on :::80
13:59:42  <       majestic> [    rtsp] rtsp_init@31                  RTSP server started on port 554
13:59:42  <       majestic> [   netip] netip_start@2055              NETIP server started on port 34567
[Func]:HI_MPI_ISP_QueryExposureInfo [Line]:730 [Info]:Get Active Stat Buffer Err
13:59:43  <     iq_monitor> [     sdk] monitor_thread@363            Query exposure info
13:59:43  <     iq_monitor> [     sdk] monitor_thread@363            Unknown error code A01C8046
13:59:44  <       majestic> [     sdk] venc_read@926                 Timeout get any data from venc channel 0
13:59:44  <       majestic> [     sdk] venc_read@946                 NOTE: Current frame is NULL!
```

This is the .ini file i used:
[gc2033_i2c_1080p.zip](https://github.com/OpenIPC/firmware/files/11002127/gc2033_i2c_1080p.zip)

I'll keep trying!

balb0x, 2023-03-18:
[image]

dimerr, 2023-03-18:
Oh, its MIPI on stock, omg.

So check Input_mode and Input_mod in config, should be like in MIPI configs

balb0x, 2023-03-18:
We are getting closer!
Now majestic does not give any error, but when i try to get the preview image or the rtsp stream, i get a black image. 

I copied the contents of the `ov2735_i2c_1080p.in` config file for the hi3516cv300 and modified it with the resolution and mask. 

This is the vimdiff of the vi files (stock and current)

[image]

dimerr, 2023-03-18:
Interrupt count is growing, so VI works properly. Probably current driver cant set sensor exposition