OpenIPC/firmware #679: [XM530] Not support this sensor

vinibali, 2023-03-18:


widgetii, 2023-03-18:
The best way to continue R&D on all XM SDK sensors is to integrate [openxiongmai](https://github.com/openIPC/openxiongmai/) builds straight into firmware CI (as currently [openhisilicon](https://github.com/openIPC/openhisilicon/) does). Then using sample_venc provided by XM SDK try to get any video stream for the board and debug sensor library.

But as for me is more reasonable is to find fresh XM SDK version with all new sensor drivers enabled

vinibali, 2023-03-18:
I'm not really sure, I did the building correctly.
I used this command after the build for the XM530 board was done.
```
make CROSS_COMPILE=arm-openipc-linux-musleabi- PATH=/media/ramdisk/firmware/output/host/bin:$PATH CFLAGS="-I../isp/include/isp -I../isp/include/xm -Werror -std=c99 -fPIC -DCHIPID_XM530 -DSOC_SYSTEM -DAWB_ALGO_V2" V=s -j1
```
I assume this way I might used musl libC instead of uClibC.
I haven't found any other SDK at that moment.
Is there a makefile option to build only the cross GCC package in the firmware repo?

dimerr, 2023-03-18:
Have you set soc to xm550 ? Check sc3335 commit

vinibali, 2023-03-18:
No, I set the XM530 in the firmware build.
Do you want me to check this?
https://github.com/OpenIPC/openxiongmai/commit/8c9d37612a01aec3f88127e53366794e5edb8e00#diff-c2e4c23db9af79f8119d7a92a76c4b67841ddb9e22b15d5bd6420644aecb788cL1105

dimerr, 2023-03-18:
Since you have XM550, you have to set it there, otherwise lib will crash

vinibali, 2023-03-18:
I only used the firmware build to have the cross GCC, which I've used on the libsns_X50's makefile. Was that wrong?

dimerr, 2023-03-18:
Dunno, our toolchain prolly works too. Just set that line to xm550

vinibali, 2023-04-09:
I tried that one too, the compiled binary is exactly the same. Then I recompiled the libispapp_XM530.so as well, but it still gives the segmentation fault.
As I saw, the X530 and X550 are so much identical, many parts are combined in the documentation as well. I assume some other lib is still needs some care.
Can I help somehow to get the XM libraries to build with the CI?

vinibali, 2023-04-09:
However I haven't find the source for libisp_XM530.so.
Do we have somebody how has a very competitive knowledge in the XM series?

widgetii, 2023-04-09:
There is no source code for `libisp_XM530`, this library should be decompiled and reverse engineered first

vinibali, 2023-04-10:
I replaced only the custom built libispapp library now, while original libsns library remained the same. I receive the segmentation fault again.
I think the problem is with the compiled library and the only thing which has been changed in the makefile is the buildchain.
Do you think if can I find the original buildchain called arm-xm-linux- somewhere?
A somewhat better debugging would be great, but ipctool's strace is not so verbose.

vinibali, 2023-04-15:
I've built a static trace and here is the strace output of running Majestic with the new libsns library:
```
root@openipc-xm530:/usr/lib# ln -s /usr/lib/sensors/libsns_X50_XM530.so libsns.so
root@openipc-xm530:/usr/lib# /tmp/strace /usr/bin/majestic 
execve("/usr/bin/majestic", ["/usr/bin/majestic"], 0xbeaa1d70 /* 20 vars */) = 0
set_tls(0xb6f8d410)                     = 0
set_tid_address(0xb6f8dfc4)             = 672
brk(NULL)                               = 0x15e8000
brk(0x15ea000)                          = 0x15ea000
mmap2(0x15e8000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x15e8000
open("/etc/ld-musl-arm.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libispapp_XM530.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libispapp_XM530.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libispapp_XM530.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, 0xbecd7698) = -1 ENOSYS (Function not implemented)
fstat64(3, {st_mode=S_IFREG|0644, st_size=75908, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\20Y\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 110592, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6ee3000
mmap2(0xb6efd000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x12000) = 0xb6efd000
mprotect(0xb6ee3000, 110592, PROT_READ|PROT_WRITE|PROT_EXEC) = 0
close(3)                                = 0
open("/lib/libisp_XM530.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libisp_XM530.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libisp_XM530.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, 0xbecd7698) = -1 ENOSYS (Function not implemented)
fstat64(3, {st_mode=S_IFREG|0644, st_size=63288, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\3108\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 98304, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6ecb000
mmap2(0xb6ee2000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xf000) = 0xb6ee2000
mprotect(0xb6ecb000, 98304, PROT_READ|PROT_WRITE|PROT_EXEC) = 0
close(3)                                = 0
open("/lib/libsns.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libsns.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libsns.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, 0xbecd7698) = -1 ENOSYS (Function not implemented)
fstat64(3, {st_mode=S_IFREG|0644, st_size=38644, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\330\27\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 106496, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6eb1000
mmap2(0xb6ec9000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x8000) = 0xb6ec9000
close(3)                                = 0
open("/lib/libjson-c.so.5", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libjson-c.so.5", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libjson-c.so.5", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, 0xbecd7698) = -1 ENOSYS (Function not implemented)
fstat64(3, {st_mode=S_IFREG|0755, st_size=42628, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0 -\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 110592, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6e96000
mmap2(0xb6eaf000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x9000) = 0xb6eaf000
close(3)                                = 0
open("/lib/libz.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libz.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libz.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, 0xbecd7698) = -1 ENOSYS (Function not implemented)
fstat64(3, {st_mode=S_IFREG|0755, st_size=54744, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\24\36\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 122880, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6e78000
mmap2(0xb6e94000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xc000) = 0xb6e94000
close(3)                                = 0
open("/lib/libopus.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libopus.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libopus.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, 0xbecd7698) = -1 ENOSYS (Function not implemented)
fstat64(3, {st_mode=S_IFREG|0755, st_size=218852, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0( \0\0004\0\0\0"..., 936) = 936
[... 1281 lines cut ...]
[... cut ...]
ioctl(23, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd25dc) = 0
ioctl(23, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd25dc) = 0
ioctl(23, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd25dc) = 0
ioctl(23, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd25dc) = 0
ioctl(23, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd25dc) = 0
ioctl(23, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd25dc) = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x16} ---
+++ killed by SIGSEGV +++
Segmentation fault
root@openipc-xm530:/usr/lib# client_loop: send disconnect: Broken pipe
```

vinibali, 2023-04-15:
According to a small portion of the output from line 1165, assume the /dev/isp interface and the libisp files are very much connected to the libsns.
```
open("/dev/isp", O_RDWR|O_LARGEFILE)    = 22
ioctl(22, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd5354) = 0
ioctl(22, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd5354) = 0
ioctl(22, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd5354) = 0
ioctl(22, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd5354) = 0
ioctl(22, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd5354) = 0
ioctl(22, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd5354) = 0
ioctl(22, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd5354) = 0
ioctl(22, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd5354) = 0
ioctl(22, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd5354) = 0
ioctl(22, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd5354) = 0
ioctl(22, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd5354) = 0
ioctl(22, _IOC(_IOC_WRITE, 0x49, 0x1, 0xc), 0xbecd5354) = 0
writev(2, [{iov_base="\33[0;32;32mDEBUG: DspChip:XM530\n\33"..., iov_len=34}, {iov_base=NULL, iov_len=0}], 2DEBUG: DspChip:XM530
) = 34
writev(2, [{iov_base="\33[0;32;32mDEBUG: SnsIF:DVP\n\33[m", iov_len=30}, {iov_base=NULL, iov_len=0}], 2DEBUG: SnsIF:DVP
) = 30
writev(2, [{iov_base="\33[0;32;32mDEBUG: Vstd:PAL[25fps]"..., iov_len=36}, {iov_base=NULL, iov_len=0}], 2DEBUG: Vstd:PAL[25fps]
) = 36
open("/dev/isp", O_RDWR|O_LARGEFILE)    = 23
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_SHARED, 23, 0) = 0xb6cd4000
writev(2, [{iov_base="[sensor_init]this sensor chip un"..., iov_len=45}, {iov_base=NULL, iov_len=0}], 2[sensor_init]this sensor chip unsupport 130!
) = 45
ioctl(23, _IOC(_IOC_WRITE, 0x49, 0x2, 0xc), 0xbecd535c) = 0
#LEADING TO SIGSEGV
```
Do you have any idea how could I find a newer SDK for the XM family?

dimerr, 2023-04-15:
taobao, ebaina, 16rd. but i think it's gonna be a challenge

widgetii, 2023-04-15:
Every whom I've connected with said that XM is a crap and don't waste your time on time

widgetii, 2023-04-15:
Actually to continue adding new sensor to existing SDK you should RE libfvideo library, add sensor specific code (including hardware specific registers like PLL configuration), compile it back and hope that everything goes smoothly. This also will take months or even years of your time spending on studying binaries

[image]

vinibali, 2023-05-01:
Thanks @widgetii, I thought we have all the sources to modify the firmware.
I wrote to the company in the meanwhile. For now they are refusing the give me the firmware directly.

vinibali, 2023-05-14:
I've downloaded a package with the help of BaiduDownloader, which contains some libs, the Linux kernel, busybox, U-boot and so on.
According to the libsns/snsX50 libraries content, it's somewhat older than the current openxiongmai-sdk.

dima333750, 2026-11-05:
Work https://github.com/OpenIPC/firmware/pull/2663