OpenIPC/firmware #681: The problem related to the lack of SD mount checks

FlyRouter, 2023-03-18:
The problem related to the lack of SD mount checks

```
#!/bin/sh

if grep -q loop /proc/devices; then
	tar -cf /tmp/settings.tar -C /overlay root
	mv -f /tmp/settings.tar /mnt/mmcblk0p1 && echo OpenIPC: Save settings to sdcard
fi
```

FlyRouter, 2023-03-18:
Also, until the problem is fixed, the same kind of code in /init is disabled, which breaks the current algorithms

```
elif grep -q loop /proc/devices; then
    mount -t devtmpfs devtmpfs /dev
    mount -t tmpfs tmpfs /overlay
    sleep 1
    mkdir -p /overlay/sdcard
    mount -r /dev/mmcblk0p1 /overlay/sdcard
    tar -xf /overlay/sdcard/settings.tar -C /overlay && echo OpenIPC: Extract settings from sdcard
    umount /overlay/sdcard
    rmdir /overlay/sdcard
```

viktorxda, 2023-03-18:
The savescript and the init file edit are not ideal, those concerns were noted on the PR #674 and on Telegram.
The script assumes that the system is booted from the sdcard, so /dev/mmcblk0p1 should be available.

FlyRouter, 2023-03-18:
At the moment the script thinks it is checking the SD, but it's not really there and mv won't go anywhere and there will be a lot of swearing. Also, if there is a lot of stuff in the overlay, it will run out of RAM as well.

We need to make the inittab and the script universal, because now if you run this on the usual cameras that we use - the output is a brick, I checked it ;) So that you understand I described the problem, a view from our side.

FlyRouter, 2023-03-18:
> The savescript and the init file edit are not ideal, those concerns were noted on the PR #674 and on Telegram. It is assumed that the system is booted from sdcard, so /dev/mmcblk0p1 should be available.

Thanks for the clarification.
We just tested the ideas in practice and encountered a problem, so part of the code is urgently disabled and I warned you, there is no claim for you to understand.

viktorxda, 2023-03-18:
It is fine, this is very experimental overall, so it was likely to encounter some issues.
The sdcard approach is maybe only sufficient for initial bringup/debugging, for a general usage it might be more appropriate to use the internal nand.

cronyx, 2023-03-18:
If you need to load settings from an SD card, you can use a flag in env (for example boot_from_sd=1) and check it in init (fw_printenv boot_from_sd), and also check if the sd card is mounted (for example: cat /proc/mounts | grep mmcblk0p1)

FlyRouter, 2023-03-18:
We can always apply this patch, there's no need to rush. It will be much more interesting to fix it, as your ideas are quite original both on the backup and on the mount. Just need more options to check and enable features.

viktorxda, 2023-03-18:
Something like this for example:
```
elif grep -q sdcardboot /proc/cmdline; then
if [ -e /dev/mmcblk0p1 ]
	mount -t devtmpfs devtmpfs /dev
	mount -t tmpfs tmpfs /overlay
	sleep 1
	mkdir -p /overlay/sdcard
	mount -r /dev/mmcblk0p1 /overlay/sdcard
	tar -xf /overlay/sdcard/settings.tar -C /overlay && echo OpenIPC: Extract settings from sdcard
	umount /overlay/sdcard
	rmdir /overlay/sdcard
else
	echo OpenIPC: SDCard not available
fi
```

```
if grep -q sdcardboot /proc/cmdline; then
	if [ -e /mnt/mmcblk0p1 ]
		tar -cf /tmp/settings.tar -C /overlay root
		mv -f /tmp/settings.tar /mnt/mmcblk0p1 && echo OpenIPC: Save settings to sdcard
	else
		echo OpenIPC: SDCard not available
	fi
fi
```

About the overlayfs, it is only intended to save the configuration files, that were only a few kilobytes from my testing.

FlyRouter, 2023-03-18:
All principles are clear and understandable, thank you.
It was just a question of practical implementation, so that this functionality does not break the current system.

Colleagues, please comment on Viktor's suggested script  @themactep @cronyx @dimerr

dimerr, 2023-03-18:
Isn't there some 100% way to check this without extra args? Like loopback mount or something

cronyx, 2023-03-18:
`elif grep -q sdcardboot /proc/cmdline; then
		mount -t devtmpfs devtmpfs /dev
		mount -t tmpfs tmpfs /overlay
		sleep 1
	if [ -e /dev/mmcblk0p1 ]
		mkdir -p /overlay/sdcard
		mount -r /dev/mmcblk0p1 /overlay/sdcard
		tar -xf /overlay/sdcard/settings.tar -C /overlay && echo OpenIPC: Extract settings from sdcard
		umount /overlay/sdcard
		rmdir /overlay/sdcard
	else
		echo OpenIPC: SDCard not available
	fi`

Looks like a good. Why do we need a delay of 1 second when working in memory?

viktorxda, 2023-03-18:
This was an issue that i (rarely) encountered, it seems like the sdcard isn't sometimes immediately available on the devtmpfs.

cronyx, 2023-03-18:
> This was an issue that i (rarely) encountered, it seems like the sdcard isn't sometimes immediately available on the devtmpfs.

Thank you, understand.

viktorxda, 2023-03-18:
I think it is the best solution to remove the scripts from the general overlay.
The idea was to merge the scripts to the sigmastar-initramfs package, which is exactly intended for this purpose.
This way it can be kept as niche solution without interfering with the other devices.

viktorxda, 2023-03-18:
Added the proposed solution in #684.

closed 2023-05-10 by FlyRouter