OpenIPC/firmware #691: hi3516cv300 00022520 IPG-80H20PT-S SOI F22

SvetlanaCh22, 2023-03-24:
Hello friends,

I have a camera Misecu IPC-BT621A-20H, based on chip hi3516cv300. But sensor is SOI F22.

After flashing firmware I see:
**Hardware**
Processor hi3516cv300
Family hi3516cv300
Sensor unknown
Flash 8 MB

**Firmware**
Version 2.3.03.21-lite
Build master+a89791ec, 2023-03-21
Majestic HEAD+f3370d8, 2023-03-21
Hostname openipc-hi3516cv300

and video is not available.

I try to use:
fw_setenv sensor jxf22_i2c_dc
reboot

Next load I see:
Sensor jxf22_i2c_dc

but video is not available.

**Log is here:**

```
Mar 25 00:19:26 openipc-hi3516cv300 user.info majestic[906]: Majestic Lite for HiSilicon, version HEAD+f3370d8, built on 2023-03-21
Mar 25 00:19:26 openipc-hi3516cv300 user.info majestic[906]: Using /etc/majestic.yaml as main configuration
Mar 25 00:19:26 openipc-hi3516cv300 user.debug majestic[906]: Main audio codec is opus (other options: aac, pcm, alaw, ulaw)
Mar 25 00:19:26 openipc-hi3516cv300 user.info majestic[906]: Using /etc/sensors/jxf22_i2c_dc_1080p_line.ini sensor configuration file
Mar 25 00:19:26 openipc-hi3516cv300 user.debug majestic[906]: App was built with MPP version: Hi3516CV300_MPP_V1.0.4.0.B050 Release
Mar 25 00:19:26 openipc-hi3516cv300 user.debug majestic[906]: Current MPP version: HI_VERSION=Hi3516CV300_MPP_V1.0.4.0 B050 Release
Mar 25 00:19:26 openipc-hi3516cv300 user.debug majestic[906]: sensor jxf22
Mar 25 00:19:26 openipc-hi3516cv300 user.debug majestic[906]:   input_mode CMOS, WDR NONE
Mar 25 00:19:26 openipc-hi3516cv300 user.debug majestic[906]:   dev [1920x1080]@0x0 25fps, BGGR
Mar 25 00:19:26 openipc-hi3516cv300 user.debug majestic[906]: trying to load /usr/lib/sensors/libsns_jxf22_i2c_dc.so
Mar 25 00:19:27 openipc-hi3516cv300 user.debug majestic[906]: Free MMZ mem before allocation: 32252KB
Mar 25 00:19:27 openipc-hi3516cv300 user.debug majestic[906]: VB configuration:
Mar 25 00:19:27 openipc-hi3516cv300 user.debug majestic[906]:   [0]: 3136320 x 9
Mar 25 00:19:27 openipc-hi3516cv300 user.debug majestic[906]:   [1]: 4147200 x 1
Mar 25 00:19:27 openipc-hi3516cv300 user.debug majestic[906]: Free MMZ mem after allocation: 624KB
Mar 25 00:19:27 openipc-hi3516cv300 user.debug majestic[906]: Sensor driver loaded
Mar 25 00:19:27 openipc-hi3516cv300 user.debug majestic[906]: SDK is in 'offline' mode
Mar 25 00:19:27 openipc-hi3516cv300 user.info majestic[906]: linear mode
Mar 25 00:19:27 openipc-hi3516cv300 user.info majestic[906]: Init dev i2c_0!
Mar 25 00:19:27 openipc-hi3516cv300 user.info majestic[906]: ------------- f22 1080p 30fps  init ok![16CV300] by gjj --------2016.12.19--------
Mar 25 00:19:27 openipc-hi3516cv300 user.err majestic[906]: Cannot create VPSS grp 0
Mar 25 00:19:27 openipc-hi3516cv300 user.err majestic[906]: ERR_VPSS_NOMEM: The memory fails to be allocated
Mar 25 00:19:27 openipc-hi3516cv300 user.info majestic[906]: Loading IQ profile /etc/sensors/iq/default.ini
Mar 25 00:19:27 openipc-hi3516cv300 user.debug majestic[906]: OSD initialized
Mar 25 00:19:27 openipc-hi3516cv300 user.debug majestic[906]:   OSD took 296KB of MMZ
Mar 25 00:19:27 openipc-hi3516cv300 user.err majestic[906]: Cannot enable AI dev 0
Mar 25 00:19:27 openipc-hi3516cv300 user.err majestic[906]: ERR_AI_NOMEM: The memory fails to be allocated
Mar 25 00:19:27 openipc-hi3516cv300 user.err majestic[906]: Cannot start Ai for AiDev=0
Mar 25 00:19:27 openipc-hi3516cv300 user.err majestic[906]: ERR_AI_NOMEM: The memory fails to be allocated
Mar 25 00:19:27 openipc-hi3516cv300 user.info majestic[906]: Image tuning task started
Mar 25 00:19:27 openipc-hi3516cv300 user.debug majestic[906]: HiSilicon SDK started
Mar 25 00:19:27 openipc-hi3516cv300 user.debug majestic[906]: Free MMZ mem finally: 36KB
Mar 25 00:19:28 openipc-hi3516cv300 user.info majestic[906]: HTTP server started on 0.0.0.0:80
Mar 25 00:19:28 openipc-hi3516cv300 user.info majestic[906]: RTSP server started on port 554
Mar 25 00:19:28 openipc-hi3516cv300 user.info majestic[906]: NETIP server started on port 34567
Mar 25 00:19:28 openipc-hi3516cv300 kern.err kernel: [   15.584505] mmz_userdev:mmz_userdev_release: MMB LEAK(pid=906): 0x83F87000, 77824 bytes, 'isp_stat_user'
Mar 25 00:19:28 openipc-hi3516cv300 kern.warn kernel: [   15.584550] mmz_userdev:mmz_userdev_release: mmz_userdev_release: mmb<0x83F87000> mapped to userspace 0xb6936000 will be force unmaped!
```

[... 67 lines cut ...]
hi3516cv300_sensor      6948  0 
hi3516cv300_ive       227043  0 
hi3516cv300_jpege      47391  0 
hi3516cv300_h265e     167365  0 
hi3516cv300_h264e     141205  0 
hi3516cv300_vedu       13623  0 
hi3516cv300_chnl       35395  0 
hi3516cv300_venc      241737  3 hi3516cv300_jpege,hi3516cv300_h265e,hi3516cv300_h264e
hi3516cv300_rc        112239  0 
hi3516cv300_vou       173835  0 
hi3516cv300_vpss      290600  0 
hi3516cv300_isp       152757  0 
hi3516cv300_viu       283135  2 hi3516cv300_vpss,hi3516cv300_isp
hi3516cv300_vgs       175171  0 
hi3516cv300_region     78729  0 
hi3516cv300_sys        97303  6 hi_acodec,hi3516cv300_aenc,hi3516cv300_aio,hi3516cv300_venc,hi3516cv300_vpss,hi3516cv300_viu
hi3516cv300_base       51198 22 hi3516cv300_adec,hi3516cv300_aenc,hi3516cv300_ao,hi3516cv300_ai,hi3516cv300_aio,hi_piris,hi3516cv300_sensor,hi3516cv300_ive,hi3516cv300_jpege,hi3516cv300_h265e,hi3516cv300_h264e,hi3516cv300_vedu,hi3516cv300_chnl,hi3516cv300_venc,hi3516cv300_rc,hi3516cv300_vou,hi3516cv300_vpss,hi3516cv300_isp,hi3516cv300_viu,hi3516cv300_vgs,hi3516cv300_region,hi3516cv300_sys
hi_osal                58728 26 hi3516cv300_wdt,hi_mipi,hi_acodec,hi3516cv300_adec,hi3516cv300_aenc,hi3516cv300_ao,hi3516cv300_ai,hi3516cv300_aio,hi3516cv300_pwm,hi3516cv300_sensor,hi3516cv300_ive,hi3516cv300_jpege,hi3516cv300_h265e,hi3516cv300_h264e,hi3516cv300_vedu,hi3516cv300_chnl,hi3516cv300_venc,hi3516cv300_rc,hi3516cv300_vou,hi3516cv300_vpss,hi3516cv300_isp,hi3516cv300_viu,hi3516cv300_vgs,hi3516cv300_region,hi3516cv300_sys,hi3516cv300_base
sys_config              2536  0 
vfat                    9127  0 
fat                    54439  1 vfat
f2fs                  138066  0 
```

Please help. Let me know if you need some other information for debug

widgetii, 2023-03-25:
Please include `free` output as well

SvetlanaCh22, 2023-03-25:
Dear, Dmitry, I don't clearly understand what you mean on "free output".

widgetii, 2023-03-25:
An output of `free` linux command

SvetlanaCh22, 2023-03-25:
Oh, understand, sorry.
```
[root@openipc-hi3516cv300 /tmp]# free

              total        used        free      shared  buff/cache   available
Mem:          27232        9464        1672         132       16096       14568
Swap:             0           0           0
```

widgetii, 2023-03-25:
The system is configured as 64 megabytes of total RAM (32 in osmem and 32 in MMZ area), but real board mem size is 128 megs. @ZigFisher @dimerr How do we fix it in such cases?

widgetii, 2023-03-25:
@SvetlanaCh22 try to run `fw_setenv totalmem 128M`, reboot camera and check what's changed

SvetlanaCh22, 2023-03-25:
@widgetii after changing to 128M ram camera shows picture in **Preview** mode and very slow in **mjpeg** mode.

But web interface works too slow.

**RSTP** doesn't work by link:
`rtsp://admin:mypass@192.168.1.10:554/stream=0`
same by:
`rtsp://192.168.1.10:554/stream=0`

**Onvif** stream founded on the port 34567, but Gray screen on main and extra stream.

**mp4** from:
`http://192.168.1.10/video.mp4`
works, but freezes each 2-3 seconds.

Camera settings is:

**Stream 0:**

```
.video0.enabled: true
.video0.codec: h265
.video0.size: 1920×1080
.video0.fps: 25
.video0.bitrate: 2048
.video0.gopSize: 1
.video0.gopMode: normal
.video0.rcMode: vbr
.video0.crop: 
.video0.sliceUnits: 4
```

**Stream 1:**

```
.video1.enabled: false
.video1.codec: h265
.video1.size: 704×576
.video1.fps: 25
.video1.bitrate: 512
.video1.gopSize: 1
.video1.gopMode: normal
.video1.rcMode: avbr
.video1.crop: 
.video1.sliceUnits: 4
```

**RSTP**:

```
.rtsp.enabled: true
.rtsp.port: 554
```

Complete log is:

```
Mar 25 13:00:24 openipc-hi3516cv300 daemon.info hisilicon: Get data from environment and set SENSOR as jxf22_i2c_dc
Mar 25 13:00:24 openipc-hi3516cv300 kern.emerg kernel: [   11.238422] Module himedia: init ok
Mar 25 13:00:24 openipc-hi3516cv300 kern.info kernel: [   11.238447] Hisilicon Media Memory Zone Manager
Mar 25 13:00:24 openipc-hi3516cv300 kern.emerg kernel: [   11.238459] The module param setup_allocator is hisi 
Mar 25 13:00:24 openipc-hi3516cv300 kern.emerg kernel: [   11.238978] hi_osal 1.0 init success!
Mar 25 13:00:24 openipc-hi3516cv300 kern.warn kernel: [   11.303978] hi3516cv300_base: module license 'Proprietary' taints kernel.
Mar 25 13:00:24 openipc-hi3516cv300 kern.warn kernel: [   11.304010] Disabling lock debugging due to kernel taint
Mar 25 13:00:24 openipc-hi3516cv300 kern.emerg kernel: [   11.395823] load sys.ko...OK!
Mar 25 13:00:24 openipc-hi3516cv300 kern.emerg kernel: [   11.471508] load region.ko...OK!
Mar 25 13:00:24 openipc-hi3516cv300 kern.emerg kernel: [   11.567438] load vgs.ko...OK!
Mar 25 13:00:24 openipc-hi3516cv300 kern.emerg kernel: [   11.710826] load viu.ko...OK!
Mar 25 13:00:24 openipc-hi3516cv300 kern.emerg kernel: [   11.781606] ISP Mod init!
Mar 25 13:00:25 openipc-hi3516cv300 kern.emerg kernel: [   11.924202] load vpss.ko...OK!
[... 44 lines cut ...]
Mar 25 13:00:28 openipc-hi3516cv300 user.info majestic[907]: Loading IQ profile /etc/sensors/iq/default.ini
Mar 25 13:00:28 openipc-hi3516cv300 user.debug majestic[907]:   venc took 8148KB of MMZ [1518]
Mar 25 13:00:29 openipc-hi3516cv300 user.debug majestic[907]: OSD initialized
Mar 25 13:00:29 openipc-hi3516cv300 user.debug majestic[907]:   OSD took 296KB of MMZ
Mar 25 13:00:29 openipc-hi3516cv300 user.debug majestic[907]: Ai(0,0) bind to AencChn:0 ok!
Mar 25 13:00:29 openipc-hi3516cv300 user.debug majestic[907]:   Audio took 340KB of MMZ
Mar 25 13:00:29 openipc-hi3516cv300 user.info majestic[907]: Image tuning task started
Mar 25 13:00:29 openipc-hi3516cv300 user.debug majestic[907]: HiSilicon SDK started
Mar 25 13:00:29 openipc-hi3516cv300 user.debug majestic[907]: Free MMZ mem finally: 53520KB
Mar 25 13:00:29 openipc-hi3516cv300 user.info majestic[907]: HTTP server started on 0.0.0.0:80
Mar 25 13:00:29 openipc-hi3516cv300 user.info majestic[907]: RTSP server started on port 554
Mar 25 13:00:29 openipc-hi3516cv300 user.info majestic[907]: NETIP server started on port 34567
Mar 25 13:00:29 openipc-hi3516cv300 user.debug majestic[907]: Using Opus as an audio codec for MP4.
Mar 25 13:00:29 openipc-hi3516cv300 user.err majestic[907]: error loading CA
Mar 25 13:00:29 openipc-hi3516cv300 user.err majestic[907]: internal error
Mar 25 13:00:29 openipc-hi3516cv300 user.err majestic[907]: internal error
Mar 25 13:00:29 openipc-hi3516cv300 user.info majestic[907]: Connecting to wss://cc.torturelabs.com:8443/v1
Mar 25 13:00:40 openipc-hi3516cv300 cron.info crond[678]: time disparity of 5577 minutes detected
Mar 25 13:00:42 openipc-hi3516cv300 user.info majestic[907]: New RTSP client from: IPEYE
Mar 25 13:00:42 openipc-hi3516cv300 user.info majestic[907]: New connection to IPEYE server
Mar 25 13:00:42 openipc-hi3516cv300 user.debug majestic[907]: Connected to IPEYE network as 00:00:23:34:45:66
Mar 25 13:01:06 openipc-hi3516cv300 user.debug majestic[907]: onvif discovery probe from 192.168.1.242:64836
```

Also I try to change IP and Mac address, but after reboot it rolling back to 192.168.1.10 and default mac during flashing 25Q64 memory. And also I can't find any way to use xmeye cloud ID, is it possible in openipc firmware?

SvetlanaCh22, 2023-03-25:
Also log files contain information about slow network, but camera connected with 20cm rj45 cable to router and connected external power 12V 3A.

```
Mar 25 13:16:55 openipc-hi3516cv300 user.debug majestic[907]: New local WebRTC joined 0xb6ae0dd0
Mar 25 13:16:56 openipc-hi3516cv300 user.warn majestic[907]: Slow network for MJPEG transmit, skipping data for 192.168.1.242:62917
Mar 25 13:16:56 openipc-hi3516cv300 user.warn majestic[907]: Slow network for MJPEG transmit, skipping data for 192.168.1.242:62917
Mar 25 13:27:06 openipc-hi3516cv300 kern.info kernel: [ 1612.995637] hisi-femac 10050000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
Mar 25 13:27:51 openipc-hi3516cv300 kern.info kernel: [ 1657.945478] hisi-femac 10050000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
```

And after 2-3 minutes onvif port is not accept connections at all and Preview page show test screen with colored lines.

FlyRouter, 2023-03-25:
Please run this if you device not have internet

```
ipcinfo -c
ipcinfo -l
```
Or this with Internet

```
ipctool
```

widgetii, 2023-03-25:
> Oh, understand, sorry.
> 
> ```
> [root@openipc-hi3516cv300 /tmp]# free
> 
>               total        used        free      shared  buff/cache   available
> Mem:          27232        9464        1672         132       16096       14568
> Swap:             0           0           0
> ```

Then increase osmem gradually because 32Mb is too low for everything you want (like WebRTC)

SvetlanaCh22, 2023-03-25:
@FlyRouter 

```
[root@openipc-hi3516cv300 /tmp]# ipctool

jrun.cgi: eval: line 1: ipctool: not found
-sh: ipctool: not found

[root@openipc-hi3516cv300 /tmp]# ipcinfo -c

hi3516cv300

[root@openipc-hi3516cv300 /tmp]# ipcinfo -l

jxf22_i2c

[root@openipc-hi3516cv300 /tmp]# cli -s .image.contrast 50

jrun.cgi: eval: line 1: cli: not found
-sh: cli -s .image.contrast 50: not found
```

At any way I changed **video0.codec** and video1.codec to **h264** via web panel and disable **HLS**. But at any way NETIP/ONVIF show pixelize and explosion at video.

Also I changed **MAC** using fw_setenv ethaddr, but can't change **IP** using fw_setenv ipaddr (it doesn't change after it). Web interface also can't change network settings (show changed ok, but display old settings).

@widgetii if I clearly understand, only changing memory on the camera board to 64M or 128M can help to use additional services in a memory with OpenIPC, right?

PS: Is any chance to restore original firmware from camera vendor using web interface of OpenIPC? The main reason why I unsolder and flash 25Q64 memory - is a bricking camera after online update and power disconnection (power station in our village works unstable), most parts of memory filled to FF include crypto block of the camera and I have no original backup, only clean firmware from vendor and backuped text data of Mac, Serial and xmeye Cloud ID.

widgetii, 2023-03-25:
As I remember, changing osmem from 32 to 40 should be enough for basic services on cv300

FlyRouter, 2023-03-25:
There are a lot of oddities in your device.
You don't have /etc/profile applied or you disabled it, so a number of commands don't work.

Although I understood the reason by reading, you must be using the Web. 
Go through SSH and repeat these commands please.

SvetlanaCh22, 2023-03-25:
@FlyRouter 

```
root@openipc-hi3516cv300:~# ipctool
The ipctool installed as remote GitHub plugin

---
chip:
  vendor: HiSilicon
  model: 3516CV300
board:
  vendor: OpenIPC
  version: 2.3.03.21
  possible-IR-cut-GPIO: 66
ethernet:
  mac: <skipped>
  u-mdio-phyaddr: 1
  phy-id: 0x001cc816
  d-mdio-phyaddr: 0
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: b9e93d92
      - name: env
        size: 0x10000
        sha1: 114d937b
        contains:
          - name: uboot-env
            offset: 0x0
      - name: kernel
        size: 0x200000
        sha1: 452566e0
      - name: rootfs
        size: 0x500000
        path: /,squashfs
        sha1: b2a49025
      - name: rootfs_data
        size: 0xb0000
        path: /overlay,jffs2,rw
    size: 8M
    addr-mode: 3-byte
ram:
  total: 128M
  media: 96M
firmware:
  kernel: "3.18.20 (Tue Mar 21 12:55:07 UTC 2023)"
  toolchain: gcc version 8.4.0 (Buildroot -gff5aa3b6-dirty)
  sdk: "Hi3516CV300_MPP_V1.0.4.0 B050 Release (Jun  3 2018, 21:42:04)"
sensors:
- vendor: Silicon Optronics
  model: JXF22
  control:
    bus: 0
    type: i2c
    addr: 0x80
  data:
    type: DC
  clock: 24MHz
```

SvetlanaCh22, 2023-03-25:
@widgetii can I change the osmem by software remapping or I need to change memory chip on the board?

widgetii, 2023-03-25:
Adjust Linux kernel boot params:

1) Check `fw_printenv bootargs`
2) If `mem` is set to `${osmem}` you need to change `osmem` U-Boot env param like `fw_setenv osmem 40M`
3) If `mem` is set to specific numeral value, then copy-paste it and change entirely like `fw_setenv bootargs 'mem=40M console=ttyAMA0,115200 panic=20 ....` (amend to your own values)

SvetlanaCh22, 2023-03-25:
@widgetii thanks, changed! But video is not works correctly via onvif/netip.

Is any chance to restore original firmware from camera vendor using web interface of OpenIPC? The main reason why I unsolder and flash 25Q64 memory - is a bricking camera after online update and power disconnection (power station in our village works unstable), most parts of memory filled to FF include crypto block of the camera and I have no original backup, only clean firmware from vendor and backuped text data of Mac, Serial and xmeye Cloud ID.

FlyRouter, 2023-03-25:
@SvetlanaCh22 

First, write me a private message (check your email and GitHub tasks) and we'll bring up a tunnel to your device for configuration. Then our specialist will be able to see how you can do the firmware on the original. Also I think we can try to do some OpenIPC firmware tuning for your device, because it's strange that it works unstable. We already have thousands of installations including your chip.

SvetlanaCh22, 2023-03-25:
@widgetii I just connect via Vesta CMS to 192.168.1.100 and port 34567. Same way as for another cameras.

> If you don't have a backup for crypto area for your camera, it's almost impossible to revert original firmware

Thank you.

@FlyRouter I'm so sorry, but I can't find any your contacts in github :( Also I don't see any private messaging here. How can I do it? Please sorry for the dumb question but as is.

FlyRouter, 2023-03-25:
I opened an Issue with a suggestion in your last repository... You can delete it.
Okay... email me at flyrouter@gmail.com or better yet, on Telegram @FlyRouter

SvetlanaCh22, 2023-03-25:
Thank you, @FlyRouter, onvif/netip now works fine.

closed 2023-03-25 by SvetlanaCh22

vinibali, 2023-06-24:
Sorry for bumping this, but what was this?
```
Mar 25 13:00:29 openipc-hi3516cv300 user.info majestic[907]: Connecting to wss://cc.torturelabs.com:8443/v1
```
This domain is not resolvable now.