OpenIPC/firmware #706: Ingenic T31X can't insert sinfo.ko

mariarti, 2023-04-02:
`insmod: can't insert '/lib/modules/3.10.14__isvp_swan_1.0__/ingenic/sinfo.ko': invalid module format
err: insmod sinfo`

```
[    8.831788] sinfo: version magic '3.10.14__isvp_**monkey**_1.0__ preempt mod_unload MIPS32_R1 32BIT ' should be '3.10.14__isvp_**swan**_1.0__ preempt mod_unload MIPS32_R1 32BIT '
[    8.832075] sinfo: version magic '3.10.14__isvp_**monkey**_1.0__ preempt mod_unload MIPS32_R1 32BIT ' should be '3.10.14__isvp_**swan**_1.0__ preempt mod_unload MIPS32_R1 32BIT '
```

```
root@openipc-t31:~# majestic
11:08:41  <       majestic> [    main] main@125                      Majestic Lite for Ingenic, version HEAD+b28acb7, built on 2023-04-02
11:08:41  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
11:08:41  <       majestic> [  config] parse_i64@467                 Can't parse param 'irSensorPin' value 'irSensorPinInvert: false'. Is not a integer (dec or hex) number.
11:08:41  <       majestic> [app_conf] parse_app_config@282          Main audio codec is alaw (other options: opus, aac, pcm, ulaw)
11:08:41  <       majestic> [watchdog] watchdog_start@101            Detected jz Watchdog
11:08:41  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 30 seconds
11:08:41  <       majestic> [sns_conf] try_sensor_config@13          SENSOR=gc2053
11:08:41  <       majestic> [sns_conf] load_sensor_config@44         Using /etc/sensor/gc2053.yaml as sensor config
11:08:41  <       majestic> [sns_conf] parse_sensor_config@128       VI: gc2053 @ 0x37 [1920x1080]
11:08:41  <       majestic> [     sdk] start_sdk@1125                CPU T31-ZX, SDK SYSUTILS-1.1.4, model Swan, devID 5614000a
11:08:41  <       majestic> [     sdk] imp_system_init@301           failed to open ISP
11:08:41  <       majestic> [     sdk] start_sdk@1218                IMP_System_Init() failed
11:08:41  <       majestic> [    main] main@209                      Shutdown main thread
```

@dimerr

mariarti, 2023-04-02:
The sinfo from rootfs march 28th is loadable.
but the sensor sensor_gc2053_t31.ko also can not (I guess not only my gc2053 and all in this family)
```
ingenic: Loading of kernel modules and initialization of the video system has started
ingenic: Get data from environment and set SENSOR as gc2053
--------------------
ISP_PARAM: isp_clk=125000000
SENSOR: gc2053
SENSOR_PARAM: sensor_max_fps=25 data_interface=1
--------------------
insmod: can't insert '/lib/modules/3.10.14__isvp_swan_1.0__/ingenic/sensor_gc2053_t31.ko': invalid module format
err: insmod sensor drv
exit

[   16.046553] sensor_gc2053_t31: version magic '3.10.14__isvp_monkey_1.0__ preempt mod_unload MIPS32_R1 32BIT ' should be '3.10.14__isvp_swan_1.0__ preempt mod_unload MIPS32_R1 3'
[   16.046832] sensor_gc2053_t31: version magic '3.10.14__isvp_monkey_1.0__ preempt mod_unload MIPS32_R1 32BIT ' should be '3.10.14__isvp_swan_1.0__ preempt mod_unload MIPS32_R1 3'
```
With sensor_gc2053_t31.ko from March 28th loaded everything, there is a picture from majestic.

dimerr, 2023-04-04:
Fixed in #713