OpenIPC/firmware #728: which Hi3516Ev300 board is best for OpenIPC?

catkira, 2023-04-16:
I want to by a Hi3516Ev300 board that works well with OpenIPC. So far my main candidates are
https://de.aliexpress.com/item/1005002294020685.html
https://de.aliexpress.com/item/1005002345471331.html
The PCBs look quite different on the front and back
first link PCB:
[image]
[image]
second link PCB:
[image]
[image]

which do You think is best? Or can You recommend even another one?

catkira, 2023-04-16:
The 2nd one has an additional connector (P4) with debug signals, that seems to be an advantage.

catkira, 2023-04-16:
https://www.xiongmaitech.com/en/index.php/product/product-detail/185/187/377
this is the manufacturer of the 2nd module

catkira, 2023-04-16:
This is another variant
https://pt.aliexpress.com/item/1005002304736662.html

cronyx, 2023-04-16:
> https://www.xiongmaitech.com/en/index.php/product/product-detail/185/187/377 this is the manufacturer of the 1st module

This is a good option, but the hi3516ev300 chips have been out of production for a long time, their counterpart is GK7205v300. https://www.xiongmaitech.com/en/index.php/product/product-detail/185/187/465

closed 2023-04-24 by cronyx