OpenIPC/firmware #761: T21n reboots about 10 seconds after login prompt

CapnRon, 2023-05-03:
I thought it may be a memory issue but I was able to run free before it reboot

```
root@openipc-t21:~# free
              total        used        free      shared  buff/cache   available
Mem:          35420        7340       16488           0       11592           0
-/+ buffers/cache:         7340       28080
Swap:             0           0           0
```

themactep, 2023-05-03:
`killall majestic; sleep 3; majestic`
Check the output for errors.

CapnRon, 2023-05-04:
no errors, I can kill majestic and bring it back up before it reboots again. 

```
root@openipc-t21:~# kill -9 770
root@openipc-t21:~# ps ax
PID   USER     COMMAND
    1 root     init
    2 root     [kthreadd]
    3 root     [ksoftirqd/0]
    4 root     [kworker/0:0]
    5 root     [kworker/0:0H]
    6 root     [kworker/u2:0]
    7 root     [rcu_preempt]
    8 root     [rcu_bh]
    9 root     [rcu_sched]
   10 root     [watchdog/0]
   11 root     [khelper]
   12 root     [kdevtmpfs]
   13 root     [kworker/u2:1]
  126 root     [writeback]
  128 root     [bioset]
  130 root     [kblockd]
  172 root     [khubd]
  203 root     [kworker/0:1]
  204 root     [rpciod]
  207 root     [kworker/0:2]
  212 root     [kswapd0]
  261 root     [fsnotify_mark]
  263 root     [nfsiod]
  271 root     [crypto]
  532 root     [deferwq]
  536 root     [kworker/0:1H]
  550 root     [jffs2_gcd_mtd4]
  582 root     /sbin/syslogd -n -C64 -t
  586 root     /sbin/klogd -n
  614 root     /usr/sbin/rngd -r /dev/urandom
  650 root     udhcpc -x hostname:openipc-t21 -T 1 -t 5 -R -b -O search -p /var
  657 root     /usr/sbin/ntpd -n
  662 root     /usr/sbin/dropbear -R -B -k -p 22 -K 300
  667 root     /usr/sbin/httpd httpd -p 85 -f -c /etc/httpd.conf -r Authenticat
  680 root     /usr/sbin/crond -f -c /etc/crontabs
  744 root     [irq/37-isp-m0]
  746 root     [irq/38-isp-w02]
  769 root     -sh
  787 root     ps ax
root@openipc-t21:~# majestic
16:26:03  <       majestic> [    main] main@125                      Majestic Li te for Ingenic, version HEAD+1db6327, built on 2023-05-02
16:26:03  <       majestic> [app_conf] load_config@89                Using /etc/ majestic.yaml as main configuration
16:26:03  <       majestic> [watchdog] watchdog_start@101            Detected jz  Watchdog
16:26:03  <       majestic> [watchdog] watchdog_start@121            Watchdog ti meout set to 10 seconds
16:26:03  <       majestic> [sns_conf] try_sensor_config@13          SENSOR=ov27 35b
16:26:03  <       majestic> [sns_conf] load_sensor_config@44         Using /etc/ sensor/ov2735b.yaml as sensor config
16:26:03  <       majestic> [sns_conf] parse_sensor_config@128       VI: ov2735b  @ 0x3c [1920x1080]
16:26:03  <       majestic> [     sdk] start_sdk@1243                CPU T21-N,  SDK SYSUTILS-1.0.29, model Turkey, devID
16:26:03  <       majestic> [     sdk] imp_system_init@334           failed to A ddSensor [ov2735b]
16:26:03  <       majestic> [     sdk] start_sdk@1342                IMP_System_ Init() failed
16:26:03  <       majestic> [    main] main@209                      Shutdown ma in thread
```
 i have also tried killing all these other processes but the system still reboots
 
 ```
 /usr/sbin/rngd -r /dev/urandom
udhcpc -x hostname:openipc-t21 -T 1 -t 5 -R -b -O search -p /var
/usr/sbin/ntpd -n
/usr/sbin/dropbear -R -B -k -p 22 -K 300
/usr/sbin/httpd httpd -p 85 -f -c /etc/httpd.conf -r Authenticat
/usr/sbin/crond -f -c /etc/crontabs
[irq/37-isp-m0]
[irq/38-isp-w02]
-sh
```

CapnRon, 2023-05-04:
I was able to get a full dmesg before it reboot

```
root@openipc-t21:~# dmesg
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14__isvp_turkey_1.0__ (runner@fv-az1107-468) (gcc version 8.4.0 (Buildroot -gff5aa3b6-dirty) ) #2 PREEMPT Tue May 2 16:17:23 UTC 2023
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 RESET ERROR PC:8001CC48
[    0.000000] [<8001cc48>] jz_wait_irqoff+0x68/0xa8
[    0.000000] CPU0 revision is: 00d00100 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] CCLK:864MHz L2CLK:432Mhz H0CLK:225MHz H2CLK:225Mhz PCLK:112Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 0045d000 @ 00010000 (usable)
[    0.000000]  memory: 00033000 @ 0046d000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 02800000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x027fffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x027fffff]
[    0.000000] On node 0 totalpages: 10240
[    0.000000] free_area_init_node: node 0, pgdat 8046af60, node_mem_map 81000000
[    0.000000]   Normal zone: 80 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 10240 pages, LIFO batch:1
[    0.000000] Primary instruction cache 16kB, 8-way, VIPT, linesize 32 bytes.
[    0.000000] Primary data cache 16kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] pls check processor_id[0x00d00100],sc_jz not support!
[    0.000000] MIPS secondary cache 64kB, 8-way, linesize 32 bytes.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 10160
[    0.000000] Kernel command line: mem=40M rmem=24M@0x2800000 console=ttyS1,115200n8 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=jz_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 35216k/40960k available (3372k kernel code, 5744k reserved, 1092k data, 204k init, 0k highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:422
[    0.000000] clockevents_config_and_register success.
[    0.000030] Calibrating delay loop... 858.52 BogoMIPS (lpj=4292608)
[    0.037695] pid_max: default: 32768 minimum: 301
[    0.042698] Mount-cache hash table entries: 512
[    0.047921] Initializing cgroup subsys debug
[    0.052186] Initializing cgroup subsys freezer
[    0.058584] devtmpfs: initialized
[    0.064561] regulator-dummy: no parameters
[    0.069192] NET: Registered protocol family 16
[    0.120503] bio: create slab <bio-0> at 0
[    0.137880] jz-dma jz-dma: JZ SoC DMA initialized
[    0.145500] usbcore: registered new interface driver usbfs
[    0.151380] usbcore: registered new interface driver hub
[    0.157137] usbcore: registered new device driver usb
[    0.162875]  (null): set:280  hold:281 dev=112500000 h=562 l=562
[    0.172712] Switching to clocksource jz_clocksource
[    0.183144] dwc2 otg probe start
[    0.183203] jz-dwc2 jz-dwc2: cgu clk gate get error
[    0.188430] DWC IN OTG MODE
[    0.192080] dwc2 dwc2: Keep PHY ON
[    0.195458] dwc2 dwc2: Using Buffer DMA mode
[    0.200130] dwc2 dwc2: Core Release: 3.00a
[    0.204251] dwc2 dwc2: DesignWare USB2.0 High-Speed Host Controller
[    0.210715] dwc2 dwc2: new USB bus registered, assigned bus number 1
[    0.218888] hub 1-0:1.0: USB hub found
[    0.222644] hub 1-0:1.0: 1 port detected
[... 366 lines cut ...]
[   14.230726] PrId  : 00d00100 (Ingenic Xburst)
[   14.230734] Modules linked in: sensor_ov2735b_t21(O) audio(O) tx_isp_t21(O) sinfo(O) vfat fat f2fs
[   14.230769] Process majestic (pid: 770, threadinfo=81944000, task=820b9440, tls=77837090)
[   14.230778] Stack : ff0a0000 ffffffff c01f6224 0000041f 00000438 804a0000 c0573548 00000780
[   14.230778]    00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   14.230778]    00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   14.230778]    00000000 00000000 fffffdfd 824bde00 01000000 c01f1d68 00000000 80445ca0
[   14.230778]    00000000 80353574 00000000 81945d20 01000000 821e9e00 01000000 fffffdfd
[   14.230778]    ...
[   14.230904] Call Trace:
[   14.231036] [<c01f3d10>] ispcore_core_ops_init+0x51c/0x6e8 [tx_isp_t21]
[   14.231284] [<c01f1d68>] ispcore_core_ops_ioctl+0xa8/0x1e8 [tx_isp_t21]
[   14.231523] [<c01cb750>] tx_isp_notify+0xb4/0x114 [tx_isp_t21]
[   14.231750] [<c01c482c>] subdev_sensor_ops_set_input+0x1b0/0x234 [tx_isp_t21]
[   14.231971] [<c01c538c>] subdev_sensor_ops_ioctl+0xb0/0x5d4 [tx_isp_t21]
[   14.232203] [<c01f1c64>] ispcore_sensor_ops_ioctl+0x74/0xc8 [tx_isp_t21]
[   14.232440] [<c01cc218>] tx_isp_unlocked_ioctl+0x438/0xea4 [tx_isp_t21]
[   14.232568] [<800e2870>] vfs_ioctl+0x30/0x48
[   14.232578]
[   14.232587]
[   14.232587] Code: 8e020128  afa2001c  8e020120 <8c430000> 27a20024  90610000  24630001  a0410000  1420fffc
[   14.232630] ---[ end trace a447c768bef61b2c ]---
root@openipc-t21:~#

```

dimerr, 2023-05-04:
do sysupgrade

CapnRon, 2023-05-04:
idk how to do sysupgrade in 10 seconds before it reboots? 

I did try the nightly build from 10 hours ago and it also reboots after about 10 seconds.

[image]

dimerr, 2023-05-04:
something like chmod -x /usr/bin/majestic

CapnRon, 2023-05-04:
I was able to get it in but still rebooted after a couple seconds

[image]

themactep, 2023-05-04:
`mv /etc/init.d/S95* ~/`

I think tx-isp module. Had a similar symptoms on a t21 board. You need to prevent ingenic-specific drivers from loading.

CapnRon, 2023-05-04:
Thank you @themactep !

closed 2023-05-04 by CapnRon