OpenIPC/firmware #766: How to build nand rootfs for Hi3516CV500

TalusL, 2023-05-07:
How to build nand rootfs for Hi3516CV500？

widgetii, 2023-05-07:
Use other platforms with NAND support like 3516cv300 or 3516av200 to study how to enable ubifs in configs

TalusL, 2023-05-07:
> Use other platforms with NAND support like 3516cv300 or 3516av200 to study how to enable ubifs in configs

I modified the broad config and rebuilt it.Then I got the rootfs.ubi and rootfs.ubifs .But the  kernel does not find rootfs while booting and  reboot after 20 seconds.Something wrong？

widgetii, 2023-05-07:
I suggest reading books about this topic (`Chris Simmonds, Mastering Embedded Linux` is the best choice from my point of view), then continue reading project's documentation

closed 2023-05-07 by widgetii