OpenIPC/firmware #776: building.sh not creating same output as precompiled binary

cdg123, 2023-05-08:
When attempting to build the images for XM530-lite using building.sh, and following the wiki article [building](https://github.com/OpenIPC/wiki/blob/master/en/building.md), the output image created for rootfs.squashfs.xm530 is not the same as the precompiled image. The resulting image is unstable and causes the camera to reboot. The precompiled images are stable.

I have replicated this again today (8th May) by downloading the precompiled image and cloning this repo and found a number of differences however it appears the main issue is the majestic binary in the output from building.sh. Precompiled majestic binary file size is 1149852 bytes, my build creates a file 939444.

To try and prove the problem I unsquashed the rootfs.squashfs.xm530 from the precompiled binary download and copied the majestic bin (1149852) to the unsquashed rootfs.squashfs.xm530 of the output from building.sh. Then squashed the output of building.sh back into the squashfs.xm530 image and flashed it to the camera. It has been stable for a number of hours now.

The majestic version info is reporting LIBH264: Complied at Apr 30 2019 16:55:22, HEAD+3604e8a, 2023-05-08 for the precompiled version and LIBH264: Complied at Apr 30 2019 16:55:22, master+3604e8a, 2023-05-03 for the output from building.sh.

The other issues seem to be a very small number of symlinks are different but I will look at these again and report as a seperate issue as they don't seem to be affecting the current camera operation.

FlyRouter, 2023-05-09:
The size difference is due to the fact that different compilers and options are used when publishing releases and when building in CI, which uses partially precompiled components from majestic. You may want to reconsider Issues, this question was already asked about half a year ago. It takes time and resources to synchronize the binary dimensions, while the streamer authors doesn't have them, but they are aware of the task.

closed 2023-05-10 by cdg123

widgetii, 2023-05-10:
Please check latest build, we tried to use same options for CI builds as for binary builds