OpenIPC/firmware #800: Add additional MMC on HiSilicon for SDIO WiFi

ialexlog, 2023-05-17:
Camera: EZVIZ CS-CV206 C0-3B2WFR (Hi3518EV200) Mini O Plus
Issue: SDIO WiFi module is not detected on the bus:
WiFi module: Marvell 88w8801 SDIO
[image]

OpenIPC:

```
root@openipc-hi3518ev200:~# ls /sys/bus/mmc/devices/
mmc0:59b4
```
SD card only available `mmc0:59b4`

`/sys/devices/platform/soc/10020000.himciv200.SD/mmc_host/mmc0/mmc0:59b4/block/mmcblk0/mmcblk0p1`
`/sys/devices/platform/soc/10030000.himciv200.SD/mmc_host/mmc1`

On the stock:
```
# find /sys/bus/
/sys/bus/mmc/devices/mmc2:0001
.
.
/sys/bus/sdio/devices/mmc2:0001:1
/sys/bus/sdio/devices/mmc2:0001:2
/sys/bus/sdio/devices/mmc2:0001:3
```
Where `mmc2:0001` is WiFi module. (no SD card inserted)

```
# cat /sys/devices/platform/hi_mci.2/mmc_host/mmc2/mmc2:0001/mmc2:0001:1/class 0x00
# cat /sys/devices/platform/hi_mci.2/mmc_host/mmc2/mmc2:0001/mmc2:0001:1/modalias sdio:c00v02DFd9139
# cat /sys/devices/platform/hi_mci.2/mmc_host/mmc2/mmc2:0001/mmc2:0001:1/device 0x9139
# cat /sys/devices/platform/hi_mci.2/mmc_host/mmc2/mmc2:0001/mmc2:0001:1/uevent 
SDIO_CLASS=00
SDIO_ID=02DF:9139
MODALIAS=sdio:c00v02DFd9139
# cat /sys/devices/platform/hi_mci.2/mmc_host/mmc2/mmc2:0001/mmc2:0001:1/vendor 0x02df

# ls /sys/bus/mmc/devices/mmc2:0001/mmc2\:0001\:1/
class      device     modalias   power      subsystem  uevent     vendor
```

stock dmesg

[dmesg_stock.log](https://github.com/OpenIPC/firmware/files/11502357/dmesg_stock.log)

stock boot log

[boot_orig.log](https://github.com/OpenIPC/firmware/files/11502359/boot_orig.log)

https://t.me/openipc_software/92731

Related issue: 
https://github.com/OpenIPC/firmware/issues/722

relghuar, 2025-05-07:
Hello, out of curiosity - did anyone manage to get this working?
I'm playing with old Litmor Battery Cam as their cloud system is now fully dead and the cameras useless.
It's Hi3518ev200 with Hi1131 SDIO wifi card on mmc1 bus, which I can't get to even detect in openipc kernel. I've traced SDIO_CLK and SDIO_CMD pins on the wifi module with oscilloscope, and I can't see any activity anywhere, which looks like SoC doesn't even try to talk to it, although according to logs it should:

`
[    1.837675] himci: mmc host probe
[    1.927644] mmc1: clock 0Hz busmode 2 powermode 1 cs 0 Vdd 21 width 1 timing 0
[    1.927666] himci_set_ios:1263: [1] ios->power_mode = 1 
[    2.141831] himci_set_ios:1276: [1] ios->clock = 0 
[    2.146724] himci_set_ios:1302: [1] ios->bus_width = 0 
[    2.152010] himci_do_voltage_switch:1011: [1] ios_v=0 ctrl=0x00000000
[    2.158494] himci_do_voltage_switch:1014: [1] switch voltage 330
[    2.183349] himci 10030000.himciv200.SD: [1] setting SD pad-drv caps
[    2.262846] mmc1: clock 400000Hz busmode 2 powermode 2 cs 0 Vdd 21 width 1 timing 0
[    2.262861] himci_set_ios:1263: [1] ios->power_mode = 2 
[    2.268231] himci_set_ios:1276: [1] ios->clock = 400000 
[    2.273567] himci_set_ios:1302: [1] ios->bus_width = 0 
[    2.704847] mmc1: mmc_rescan_try_freq: trying to init card at 400000 Hz
[    2.711617] mmc1: starting CMD52 arg 00000c00 flags 00000195
[    2.711645] himci_exec_cmd:517: [1] arg_reg 0x28, val 0xc00
[    2.717235] himci_exec_cmd:579: [1] cmd->opcode = 52 cmd->arg = 0xC00
[    2.723748] himci_exec_cmd:598: [1] cmd_reg 0x2c, val 0xa0002174
[    2.729812] himci_cmd_done:643: [1] irq cmd status stat = 0x104 is timeout error!
[    2.737317] mmc1: req done (CMD52): -110: 00000000 00000000 00000000 00000000
[    2.737384] mmc1: starting CMD52 arg 80000c08 flags 00000195
[    2.737402] himci_exec_cmd:517: [1] arg_reg 0x28, val 0x80000c08
[    2.743424] himci_exec_cmd:579: [1] cmd->opcode = 52 cmd->arg = 0x80000C08
[    2.750346] himci_exec_cmd:598: [1] cmd_reg 0x2c, val 0xa0002174
[    2.756365] himci_cmd_done:643: [1] irq cmd status stat = 0x104 is timeout error!
[    2.763897] mmc1: req done (CMD52): -110: 00000000 00000000 00000000 00000000
[    2.763926] mmc1: clock 400000Hz busmode 2 powermode 2 cs 1 Vdd 21 width 1 timing 0
[    2.763939] himci_set_ios:1263: [1] ios->power_mode = 2 
[    2.769296] himci_set_ios:1276: [1] ios->clock = 400000 
[    2.774635] himci_set_ios:1302: [1] ios->bus_width = 0 
[    2.793006] mmc1: starting CMD0 arg 00000000 flags 000000c0
[    2.793040] himci_exec_cmd:517: [1] arg_reg 0x28, val 0x0
[    2.798584] himci_exec_cmd:579: [1] cmd->opcode = 0 cmd->arg = 0x0
[    2.804783] himci_exec_cmd:598: [1] cmd_reg 0x2c, val 0xa000a000
[    2.810866] mmc1: req done (CMD0): 0: 00000000 00000000 00000000 00000000
[    2.811904] mmc1: clock 400000Hz busmode 2 powermode 2 cs 0 Vdd 21 width 1 timing 0
[    2.811919] himci_set_ios:1263: [1] ios->power_mode = 2 
[    2.817244] himci_set_ios:1276: [1] ios->clock = 400000 
[    2.822630] himci_set_ios:1302: [1] ios->bus_width = 0 
[    2.828922] mmc1: starting CMD8 arg 000001aa flags 000002f5
[    2.828943] himci_exec_cmd:517: [1] arg_reg 0x28, val 0x1aa
[    2.834529] himci_exec_cmd:579: [1] cmd->opcode = 8 cmd->arg = 0x1AA
[    2.840936] himci_exec_cmd:598: [1] cmd_reg 0x2c, val 0xa0002148
[    2.846957] himci_cmd_done:643: [1] irq cmd status stat = 0x104 is timeout error!
[    2.854491] mmc1: req done (CMD8): -110: 00000000 00000000 00000000 00000000
[    2.854515] mmc1: starting CMD5 arg 00000000 flags 000002e1
[    2.854532] himci_exec_cmd:517: [1] arg_reg 0x28, val 0x0
[    2.859984] himci_exec_cmd:579: [1] cmd->opcode = 5 cmd->arg = 0x0
[    2.866175] himci_exec_cmd:598: [1] cmd_reg 0x2c, val 0xa0002045
[    2.872223] himci_cmd_done:643: [1] irq cmd status stat = 0x104 is timeout error!
[    2.879753] mmc1: req failed (CMD5): -110, retrying...
[    2.879771] himci_exec_cmd:517: [1] arg_reg 0x28, val 0x0
[    2.885182] himci_exec_cmd:579: [1] cmd->opcode = 5 cmd->arg = 0x0
[    2.891405] himci_exec_cmd:598: [1] cmd_reg 0x2c, val 0xa0002045
[    2.897454] himci_cmd_done:643: [1] irq cmd status stat = 0x104 is timeout error!
[    2.904949] mmc1: req failed (CMD5): -110, retrying...
[    2.904966] himci_exec_cmd:517: [1] arg_reg 0x28, val 0x0
[    2.910409] himci_exec_cmd:579: [1] cmd->opcode = 5 cmd->arg = 0x0
[    2.916598] himci_exec_cmd:598: [1] cmd_reg 0x2c, val 0xa0002045
[    2.922644] himci_cmd_done:643: [1] irq cmd status stat = 0x104 is timeout error!
[    2.930169] mmc1: req failed (CMD5): -110, retrying...
[    2.930186] himci_exec_cmd:517: [1] arg_reg 0x28, val 0x0
[    2.935593] himci_exec_cmd:579: [1] cmd->opcode = 5 cmd->arg = 0x0
[    2.941879] himci_exec_cmd:598: [1] cmd_reg 0x2c, val 0xa0002045
[    2.947935] himci_cmd_done:643: [1] irq cmd status stat = 0x104 is timeout error!
[    2.955434] mmc1: req done (CMD5): -110: 00000000 00000000 00000000 00000000
`
From this I'd expect to see SoC try talking to the card, but SDIO_CLK stays on 1.8V and SDIO_CMD on GND all the time...
As far as I can tell clock is set correctly (CRG49:[0x200300c4] == 0x00000A02) which means mmc1_mux bit clk_mmc1 (bit 1) is set and clock-select is 0 (which is probably correct as the mmc1_mux_table has only one entry "0").
Is there anything else I'm missing in the MMC1 setup?
[... cut ...]
muxctrl_reg8 0x200f0020 0x3 GPIO2_4 RMII_RX_DV VO_DATA7 [SDIO1_CDATA0]
muxctrl_reg9 0x200f0024 0x3 GPIO2_5 RMII_RXD0 VO_DATA2 [SDIO1_CDATA3]
muxctrl_reg10 0x200f0028 0x3 GPIO2_6 RMII_RXD1 VO_DATA3 [SDIO1_CCMD]
muxctrl_reg11 0x200f002c 0x1 GPIO2_7 [EPHY_RST] BOOT_SEL VO_HS SDIO1_CARD_POWER_EN
muxctrl_reg13 0x200f0034 0x3 GPIO3_0 EPHY_CLK VO_DATA1 [SDIO1_CDATA2]
`
There's no detect of course, and power is handled independently (currently I'm setting wifi-pwr/rst up in uboot).
Does anyone have any ideas here?
Thanks a lot!

relghuar, 2025-05-09:
Hello again, just to elaborate a bit further - there's definitely a problem with PINMUX information for hi3518ev200 - after comparing with stock liteos firmware I've managed to get linux MMC subsystem to recognize the SDIO card.
Apparently all the SDIO1_xxx related registers in PINMUX must have a value of 0x4, NOT 0x3 as would seem from ipctool. I'll open some issue in that repository as well, although I do not have any definitive corrections beside the stock firmware memory dumps.
Now I have some basic connection to the hi1131 card, and apparently it identifies itself as vendor=0x0296 and product=0x5347, which DOES already have some kind of driver in the openipc kernel, specifically in drivers/staging/wilc1000/wilc_sdio.c there is this:
```
#define SDIO_VENDOR_ID_WILC 0x0296
#define SDIO_DEVICE_ID_WILC 0x5347
```
I've built and loaded that module, but it's not a smooth sail as it threw some CMD53 errors right in the probe. Now it's a question whether the problem is in staging state of the driver, or if this hi1131 is not actually compatible and only misidentifies itself as something totally different by accident :-/
I'll try to put together at least a rudimentary fix for the kernel repo to have a potentially working mmc1 there.