OpenIPC/firmware #828: Не стартует majestic при включении IPEYE

al2fil, 2023-05-27:
На gk7205v300 (gk7205v200 family), imx335, 16 MB nor flash, 2.3.05.27-ultimate
при включении IPEYE начинает перезагружаться по WDT

на gk7205v300 (gk7205v200 family), imx335, 16 MB nor flash, 2.3.04.23-lite нормально работает.

[consol.txt](https://github.com/OpenIPC/firmware/files/11583190/consol.txt)

/var/log/majestic.log не создаёт

widgetii, 2026-10-04:
**RU:** Проверили на текущей master-сборке: IPEYE включён, сессия с облаком (стенд, имитирующий облако IPEYE) держится 10 минут без падений и перезагрузок — и в lite, и в ultimate. На вашем логе majestic 2023 года падал с `Illegal instruction` сразу после подключения к IPEYE; перезагрузка по WDT — следствие этого. В текущей сборке ultimate защищённое (HTTPS) подключение к балансировщику IPEYE проходит, и сессия держится. Проверка была на hi3516av300, не на gk7205v300 — пожалуйста, обновитесь до текущей прошивки (`sysupgrade -k -r`), включите IPEYE в ultimate и напишите, воспроизводится ли. Если тишина — закроем.

**EN:** Re-tested on current master: with IPEYE enabled, the cloud session (against a bench that stands in for the IPEYE cloud) holds for 10 minutes with no crash or reboot, on both lite and ultimate. In your log the 2023 build died with `Illegal instruction` right after connecting to IPEYE; the WDT reboot followed from that. The current ultimate build completes the secure (HTTPS) connection to the IPEYE balancer and holds the session. The test ran on hi3516av300, not gk7205v300, so please update to current firmware (`sysupgrade -k -r`), enable IPEYE on ultimate, and tell us whether it still happens. We'll close this if we don't hear back.

al2fil, 2026-10-10:
на прошивке 2023 года после sysupgrade -k -r веб-интерфейс не поднялся, sysupgrade -w больше не поддерживается.

Прошил openipc-gk7205v300-nor-ultimate-16mb.bin

Device
SoC
gk7205v300 (gk7205v200)
Sensor
imx335
Firmware
2.6.10.09-ultimate
Build
master+6d5708b, 2026-10-09
Majestic
Ultimate HiSilicon (gk7205v200), master+9ce4cfd, 2026-10-09 17:34
WebUI
master+0eefc06, 2026-10-09 18:19

+  irCutPin1: 10 closing
+  irCutPin2: 11 opening
+  irSensorPin: 66
+  backlightPin: 52

сенсор 5M (h265 2592x1944)

IPEYE работает

Правда теперь если активировать доппоток на камере, то начинает перегружать, но к теме это не относиться.

closed 2026-10-10 by al2fil