OpenIPC/firmware #852: Include sunwait to support day/night mode on cameras without LDR

TimelessNL, 2023-06-06:
Some camera's like [this](https://aliexpress.com/item/33020810043.html) model do not have the LDR installed. I'm currently looking for alternatives to control the Day/Night. And one thing that caught my attentions is using [sunwait](https://github.com/risacher/sunwait) with cron in order to call the Majestic API endpoint to switch Day/Night. Luckily [sunwait is in buildroot](https://github.com/buildroot/buildroot/blob/master/package/sunwait/sunwait.mk).  Would it be possible to add sunwait to the lite image so camera's without LDR can have an alternative way of switching Day/Night mode?~~

cronyx, 2023-06-06:
[alternative auto day/night control](https://github.com/OpenIPC/sandbox/blob/main/scripts/backlight-control/wl_improve.sh) for cameras based on the hisilicon or goke chips without light sensors

TimelessNL, 2023-06-07:
That seems like a suitable solution indeed, I tried the script but there seems something odd.
When I put my finger on the lens (_simulating total darkness_) the camera switches between day/night mode continuously.
```
log output:
again_led_off: 1024
again_led_on: 0
Light is still OFF. Nothing changed. Continue
....................
again_led_off: 1024
again_led_on: 1024
Light is still OFF. Nothing changed. Continue
....................
again_led_off: 32381
again_led_on: 1024
LIGHT IS ON
....................
again_led_off: 0
again_led_on: 1024
LIGHT IS OFF
....................
again_led_off: 32381
again_led_on: 1024
LIGHT IS ON
....................
again_led_off: 0
again_led_on: 1024
LIGHT IS OFF
....................
again_led_off: 6387
again_led_on: 1024
Light is still OFF. Nothing changed. Continue
....................
again_led_off: 32381
again_led_on: 1024
LIGHT IS ON
....................
again_led_off: 0
again_led_on: 1024
LIGHT IS OFF
....................
again_led_off: 1024
again_led_on: 1024
Light is still OFF. Nothing changed. Continue
....................
```
True putting my finger on the lens is not a real world scenario but I prefer not to have it continuously switch between day/night when the lens is blocked somehow.

TimelessNL, 2023-06-11:
Ok after some try and error I got it to work properly.
Would it be possible to have it included in the lite firmware?

FlyRouter, 2023-06-11:
This is an external script, depends on your settings, you must install it yourself.
You are also welcome to send in your revisions for modernisation.

TimelessNL, 2023-06-11:
Thanks, will do when I'm completely comfortable with the script.

closed 2023-06-11 by TimelessNL