OpenIPC/firmware #859: MJSXJ02HL seems bricked after firmware upgrade

darkfuncat, 2023-06-07:
Info : hi3518ev300 
Hi, after a simple upgrade of firmware few days ago (from a previous formware of mid-may I would say), my camera MJSXJ02HL just rebooted in loop with orange led light.
No access at all.
I tried to re-flash it from scratch with new sdcard, the led became white a while, and reboot, but again loop reboot in orange.
I retry but now nothing work, no led, nothing.
I retry to access bootloader wia Zadig software but I cannot see it anymore now.
I think it is totally bricked.
Any suggestion ?
Thanks

cronyx, 2023-06-12:
https://github.com/OpenIPC/firmware/blob/master/ATTENTION.md

Stay tuned, we will be fix it soon.

eardunk, 2023-06-23:
I have the same problem, tried everything and can get it work.

Use fastboot(u-boot) from this topic

https://kasito.ru/proshivka-zagruzchika-ip-kamery-mjsxj02hl-s-pomoshhyu-usb/

And then drop this file to sd-card and power on with reset button

https://github.com/kasitoru/mjsxj02hl_firmware/releases/tag/1.5.0

If you cannot see bootloader in zadig try to reinstall drivers/reboot pc.

darkfuncat, 2023-06-28:
Hello
Thank you for your message @eardunk .
Knowing you were in the same situation I test again and it works now !
My solution was (I guess, not sure) to replace the cable. With the old one windows detected a change on usb, but not zadig.
Anyway, now I am back to the kasitoru firmware.
Is it safe now to reinstall OpenIPC latest version or not yet ?

FlyRouter, 2023-06-28:
The problem is in the wifi driver.
It needs to be rebuilt with the new toolchain and archived.
This should be done by someone who has such a camera in his hands and will be able to check the work immediately.

darkfuncat, 2023-08-10:
> The problem is in the wifi driver.
> It needs to be rebuilt with the new toolchain and archived.
> This should be done by someone who has such a camera in his hands and will be able to check the work immediately.

Hi @FlyRouter 
I own this  camera and i can help to test, but I dont know what to do.
Any information or manual or guide anywhere ? 
Thanks

kenkit, 2024-08-21:
You should remove the chip and flash it with other tools like raspi

kenkit, 2024-08-21:
you need to make sure the contents of the flash folder are at the root of the sdcard, e.g  autoconfig.sh at the root

VictorDrijkoningen, 2026-11-05:
Can this issue be closed, @darkfuncat?

darkfuncat, 2026-11-05:
yes sorry, as these devices are not sold anymore I gave up this model.
it's a shame because it was for me the only device that can be transform with openipc with only software (not soldering or else)
thanks for all the work anyway