OpenIPC/firmware #878: [GPIO] How to figure out GPIO functions?

vinibali, 2023-06-16:
Hi there!
What would be the way to figure out the GPIO pins of a GK7205V210 SoC with SC223a CMOS sensor?
I need to find the GPIO pins related to the IRCUT IRSTATUS and IRCTL.
It seems the factory setting are not working well. Unfortunately I cannot identify the PCB.
Thanks

widgetii, 2023-06-17:
You may use [ipctool gpio scanner](https://github.com/openIPC/ipctool/#usage) on stock firmware to determine GPIO pins

closed 2023-06-17 by widgetii

vinibali, 2023-06-17:
This doesn't really helped now. I used https://github.com/OpenIPC/firmware/issues/200 as a reference now. However I tried to trigger the GPIOs (0-23) with the following oneliner.
```
for i in $(seq 0 23) ; do gpio set $i ; sleep 5s ; gpio clear $i ; sleep 5s ; done
```
It it possible, that this device is using a photodiode to automatically turn on the IR?
I haven't see any change in the video, while running this script.

themactep, 2023-06-18:
1. Usually, IR LEDs are activated by solely an LDR on the array.

https://github.com/OpenIPC/firmware/assets/37488/54fcb62b-7666-481f-81d6-56fbc947b275

3. Range 0-23 may be not enough to find GPIO pins for IRCUT.

4. 5 sec delay may fry your IRCUT solenoid as its reaction time is only 100 milliseconds. Disconnect the IRCUT and use a LED instead.

https://github.com/OpenIPC/firmware/assets/37488/0ed5daaf-dc1c-439a-a41c-8b94a192f15f

vinibali, 2023-06-18:
As the gpio scan showed this higher than 23, I thought it has only 24 pins. I will continue the scan from that.
The main idea was to find the IRCTL pin with the oneliner.
```
sh: write error: Invalid argument
Error: No direction file!
```
This is the result from this morning:
```
ipctool gpio scan
Gr: 0, Addr:0x120B0044, Data:0x00 = 0bxxx0xxx0, Addr:0x120B0400, Dir:0xFF = 0bxxx1xxx1
Gr: 1, Addr:0x120B13CC, Data:0x00 = 0b0000xx00, Addr:0x120B1400, Dir:0x7E = 0b0111xx10
Gr: 2, Addr:0x120B2004, Data:0x00 = 0bxxxxxxx0, Addr:0x120B2400, Dir:0xFF = 0bxxxxxxx1
Gr: 3, Addr:0x120B3000, Data:0x00 = 0bxxxxxxxx, Addr:0x120B3400, Dir:0x00 = 0bxxxxxxxx
Gr: 4, Addr:0x120B4000, Data:0x00 = 0bxxxxxxxx, Addr:0x120B4400, Dir:0x00 = 0bxxxxxxxx
Gr: 5, Addr:0x120B5000, Data:0x00 = 0bxxxxxxxx, Addr:0x120B5400, Dir:0x00 = 0bxxxxxxxx
Gr: 6, Addr:0x120B63F0, Data:0x9C = 0b100111xx, Addr:0x120B6400, Dir:0x00 = 0b000000xx
Gr: 7, Addr:0x120B703C, Data:0x03 = 0bxxxx0011, Addr:0x120B7400, Dir:0x0C = 0bxxxx1100
Gr: 8, Addr:0x120B83C0, Data:0xE0 = 0b1110xxxx, Addr:0x120B8400, Dir:0x00 = 0b0000xxxx
======================================================================================
Waiting for while something changes...
======================================================================================
Gr:7, Addr:0x120B703C, Data:0x03 = 0bxxxx0011 --> 0x0B = 0bxxxx1011
Mask: "devmem 0x120B7020 32 0x08", GPIO7_3, GPIO59, Dir:Output, Level:1
======================================================================================
Gr:7, Addr:0x120B703C, Data:0x0B = 0bxxxx1011 --> 0x03 = 0bxxxx0011
Mask: "devmem 0x120B7020 32 0x00", GPIO7_3, GPIO59, Dir:Output, Level:0
======================================================================================
Gr:1, Addr:0x120B13CC, Data:0x00 = 0b0000xx00 --> 0x80 = 0b1000xx00
Mask: "devmem 0x120B1200 32 0x80", GPIO1_7, GPIO15, Dir:Input, Level:1
======================================================================================
Gr:1, Addr:0x120B13CC, Data:0x80 = 0b1000xx00 --> 0x00 = 0b0000xx00
Mask: "devmem 0x120B1200 32 0x00", GPIO1_7, GPIO15, Dir:Input, Level:0
======================================================================================
Gr:1, Addr:0x120B13CC, Data:0x00 = 0b0000xx00 --> 0x80 = 0b1000xx00
Mask: "devmem 0x120B1200 32 0x80", GPIO1_7, GPIO15, Dir:Input, Level:1
======================================================================================
Gr:1, Addr:0x120B13CC, Data:0x80 = 0b1000xx00 --> 0x00 = 0b0000xx00
Mask: "devmem 0x120B1200 32 0x00", GPIO1_7, GPIO15, Dir:Input, Level:0
======================================================================================
Gr:1, Addr:0x120B13CC, Data:0x00 = 0b0000xx00 --> 0x80 = 0b1000xx00
Mask: "devmem 0x120B1200 32 0x80", GPIO1_7, GPIO15, Dir:Input, Level:1
======================================================================================
Gr:7, Addr:0x120B703C, Data:0x03 = 0bxxxx0011 --> 0x0B = 0bxxxx1011
Mask: "devmem 0x120B7020 32 0x08", GPIO7_3, GPIO59, Dir:Output, Level:1
======================================================================================
Gr:7, Addr:0x120B703C, Data:0x0B = 0bxxxx1011 --> 0x03 = 0bxxxx0011
Mask: "devmem 0x120B7020 32 0x00", GPIO7_3, GPIO59, Dir:Output, Level:0
```

themactep, 2023-06-18:
Seems like GPIO 59, which is set as output, is your IRCUT trigger, and GPIO 15, set as input, is getting signal from LDR.

vinibali, 2023-06-18:
GPIO 15 as signal from IR sensor and GPIO 59 as IR cut filter, if I understand well there no need to set the "GPIO pin to turn on the night illumination". So this will just turn on the "Toggle night mode" switch, right?

vinibali, 2023-06-30:
I had a chance to test and these settings are fine. The LDR are controlling the IR LEDs. It just turned out, that the light of these infrared diodes are very weak. I can see the objects I'd those are close but not after a bigger distance.
What are the common distance, which can be light by the IR LEDs?

themactep, 2023-06-30:
It depends. Did you check that ircut is open? Otherwise it would be stripping the waverange that IR LEDs emmit.

blackdot, 2023-11-13:
I have a similar board (same chipset). With default firmware, I can hear, that IRcut is opening and closing based on light conditions (even when a board is booting). But after flashing openIPC (setting IRcut to 15/16) it doesn't work. So I tried to restore original firmware and now I cannot see/hear IRcut working again.
I ordered second camera and it happened again. Any ideas what could went wrong? Any chance to fix it?

It was this one : https://www.aliexpress.com/item/32885834218.html
I can provide more information...

vinibali, 2024-01-25:
> I have a similar board (same chipset). With default firmware, I can hear, that IRcut is opening and closing based on light conditions (even when a board is booting). But after flashing openIPC [...]

pretty strange. I have no idea. have you had time to work on that since this comment?

blackdot, 2024-01-27:
We already discussed this heavily over Telegram group chat and it turned out, that I burned ircut. On second camera, it got burnt right after first boot of openIPC.
I have a new ircut on table, so once I will have a time, I will try it and see.. I will report back...